betaflight/betaflight #5464: vtx_band has no 0 value to inhibit VTX setup at power on

TwoToneEddy, 2018-03-18:
Hi, 
In this article:

https://github.com/betaflight/betaflight/wiki/VTX-CLI-Settings

it states "if vtx_band==0 and vtx_freq==0 then the settings will not be sent out to the VTX". When I try and set vtx_band to 0 it appear to only have a range of 1-5:

vtx_band = 5
Allowed range: 1 - 5

With this being the case is there any other way to stop the flight controller trying to set up the VTX on startup. I have a suspicion that my VTX (atlatl) doesn't always like being told what to do immediately after power on, I would like to inhibit the flight controller from doing this but i'm not sure how to do it. Can we still do this?
Thanks
Lee

codecae, 2018-03-18:
Not sure why this would be... I'll address it.

codecae, 2018-03-18:
Why wouldn't you want it to set the vtx on startup?

codecae, 2018-03-18:
No matter the reason, you should be able to set the vtx_band to zero, as this enables custom frequency selection mode.    There's nothing special about 'startup' that makes the system behave differently, though.  The system will enforce consistency between the VTX and the settings you configure, whether or not it's during startup.   If the settings between the vtx and fc match, no changes will be requested of the VTX.

TwoToneEddy, 2018-03-18:
The reason is because the atlatl doesn't always like being configured as soon as it's powered up, perhaps is needs abit of time to initialise before it starts listening. 

When you say I should be able to set vtx_band to 0 how come it only states a range of 1-5? I'm running the latest 3.3.0 release, is there a pre-release version I need to use later than this?

TwoToneEddy, 2018-03-18:
Another odd thing I noticed, I'm not sure if its my confused atlatl on startup but only certain frequencies are able to be written to the VTX, for example, I can change between race and 1,4,5,6, and 7 but not the others. Very strange. The VTX has an led that indicates a setting change request, this led doesn't indicate a request on those channels that don't work.

codecae, 2018-03-18:
When I mentioned that you should be able to, I meant the option should be there.  Unfortunately, someone issued a PR that changed the range, so I issued one to repair this issue.  Which protocol does this vtx use?  Tramp or smartaudio?

TwoToneEddy, 2018-03-18:
IRC tramp.

PR #5465 (2018-04-03): Corrected CLI to include vtx_band=0 to enable freq selection mode.
For whatever reason, the vtx_band cli command has been changed, preventing users from supplying vtx_band=0.  This is required for the enabling of custom frequency selection mode.  This change will restore vtx_band=0 capability.

Introduced VTX_COMMON_NO_BAND to preserve array depth and logic with respect to VTX_COMMON_BAND_COUNT.

Fixes #5464
closed 2018-04-03 by mikeller