betaflight/betaflight #5468: bardwell f4 blackbox issue

Tbeal79, 2018-03-19:
hey guys i went to try the black box out on the jb f4 today, took out my micro sd from my bf3 fc erased all that was on it before i put it in the jb f4, and i got into the blackbox setting only to see it say fatal error????
tryed to hit the button to erase and flash, but get nothing from doing it.
tryed to search in the cli for blackbox and found nothing about it? kinda weird when this fc is supposed to support it.
so i did a dump for you guys to look at, tell me what im missing.

# Betaflight / CLRACINGF4 (CLR4) 3.3.0 Mar  2 2018 / 03:39:18 (177472b4f) MSP API: 1.37

# name
name pokeball2

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B08
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource INVERTER 1 C00
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
[... 571 lines cut ...]
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set thr_mid = 35
set thr_expo = 10
set rates_type = BETAFLIGHT
set roll_rc_rate = 115
set pitch_rc_rate = 115
set yaw_rc_rate = 100
set roll_expo = 15
set pitch_expo = 15
set yaw_expo = 10
set roll_srate = 55
set pitch_srate = 55
set yaw_srate = 71
set tpa_rate = 20
set tpa_breakpoint = 1650

wuhao19920619, 2018-03-19:
I guess it is the same with [https://github.com/betaflight/betaflight/issues/5303](url)

In this case sometimes the FC can be accessed but it shows fatal error.

Tbeal79, 2018-03-19:
Well your address didn't work when I clicked on it. Shows a 404 message.

wuhao19920619, 2018-03-19:
@Tbeal79 #5303