betaflight/betaflight #5470: 3.3.0 ff pikoblx problem with spektrum

garyjohnmac, 2018-03-19:
when flashing to 3.3.0 the bourd does not see the rx but the rx orange led is burning solid orange when the tx is on so it has bind with the tx . If i flash back to 3.2.5 all works normally and the bourd see the rx so it cant be a hardware problem it must be a problem with ff pikoblx  3.3 sofware same problem of a flash  butterflight  .I  have tried all the rx  ports with no success

etracer65, 2018-03-19:
Have you tried changing the value of serialrx_inverted in the CLI?

garyjohnmac, 2018-03-19:
checked there is no value all that stands is  serialrx_inverted = on  in CLI

etracer65, 2018-03-19:
Have you tried `set serialrx_inverted = OFF` ?  That's what I meant by changing the value.  May not solve it but a good place to check as the serial rx inversion logic changed in BF 3.3 - as documented in the release notes.

AndersHoglund, 2018-03-19:
Tested BF3.3.0 onSPRF3 and a Lemon diversity Rx on UART3. No problems. I have no PIKO_BLX to test with, but there should not be any difference. Please check all serialrx related settings, like: 
```
# get serial
serialrx_provider = SPEK2048
Allowed values: SPEK1024, SPEK2048, SBUS, SUMD, SUMH, XB-B, XB-B-RJ01, IBUS, JETIEXBUS, CRSF, SRXL, CUSTOM, FPORT

serialrx_inverted = OFF
Allowed values: OFF, ON

serialrx_halfduplex = OFF
Allowed values: OFF, ON
# map
map TAER1234
```
Also make sure you start configuring from a clean base, i.e. cli "defaults" or gui "Reset Settings". Then only select serlal port and serialrx settings as of above and test again. Add no other settings, yet.

garyjohnmac, 2018-03-19:
Thank you so much change to  , set serialrx_inverted = OFF and now the reciever is communicating with the bourd all is working now

closed 2018-03-19 by etracer65