betaflight/betaflight #5471: Betaflight 3.3.0 Omnibus F4 V5 magnetometer not recognised

BlueFishUK, 2018-03-19:
The external magnetometer in my Ublox M8N GPS puck does not show in Betaflight 3.3.0 the magnetometer option is greyed out.
# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.3.0 Mar  2 2018 / 03:56:35 (177472b4f) MSP API: 1.37

GPS puck working fine as I am getting Sats, longitude and latitude in Betaflight but Mag connected to SDA and SDL is not being recognised:

Here is my get mag config:
# get mag
align_mag = DEFAULT
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

mag_bustype = I2C
Allowed values: NONE, I2C, SPI, SLAVE

mag_i2c_device = 2
Allowed range: 0 - 3

mag_i2c_address = 0
Allowed range: 0 - 119

mag_spi_device = 0
Allowed range: 0 - 3

mag_hardware = AUTO
Allowed values: AUTO, NONE, HMC5883, AK8975, AK8963

mag_declination = 0
Allowed range: -18000 - 18000

magzero_x = 0
Allowed range: -32768 - 32767

magzero_y = 0
Allowed range: -32768 - 32767

magzero_z = 0
Allowed range: -32768 - 32767

I reassigned RX3 and TX3 to I2C, here is my resource list:
# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource PPM 1 B08
resource PWM 1 B08
resource PWM 2 B09
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource SONAR_TRIGGER 1 A01
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 6 C07
resource INVERTER 3 C09
resource INVERTER 6 C08
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B08
resource ADC_BATT 1 C02
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C01
resource BARO_CS 1 B03
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 B07

Wiring seems fine as GPS puck powers and GPS works fine, could this be an issue with the code for this FC target?

jflyper, 2018-03-20:
It has been working fine with supported and correctly configured hardware. Please use other media for support. Thank you in advance.

I'll be waiting for you in the RCG Omnibus F4 AIO thread.

closed 2018-03-20 by jflyper

BlueFishUK, 2018-03-23:
@jflyper I have spent the last 4 days trying everything I have even ordered another GPS Puck with HMC5883 chip have one of the guys on this forum with coding experience see  #4531 trying to help all to no avail. Both of my GPS and Mag units work perfectly in iNav with absolutely no hardware changes and the same settings as Betaflight so I assume my hardware is fine. 

If you are certain it is not an issue with Betaflight and as you say above it has been working for years with the correctly configured hardware could you give me a recommendation for hardware brand or type to buy as this may be my easiest option?

Any help appreciated. I have really tried and put a lot of time and effort into trying to solve this issue and would not waste your time if I had not exhausted all of my options.

jflyper, 2018-03-24:
Your remaining option is to make a posting in RCG Omnibus F4 AIO thread or ask for support in slack.