betaflight/betaflight #5473: Dterm transition of 0.01 causes 'hole' in D if stick crosses centre

ctzsnooze, 2018-03-19:
Currently there is a divide by zero in the D 'transition' calculation.

This means that the D transition parameter cannot ever go to 0.  The value entered in CLI is 1-100 and in the code is converted to a 0.01 - 1.0 factor, which is what we see in the configurator slider.

When transition is set to 0.01, the D weight value is applied in full while the stick is more than 1% out from centre.

But when fully centred, the D weight value becomes zero, and in that small 1% range either side of centre, the D weight value suddenly goes from zero to maximum.

This has the potential to cause a glitch while accelerating the stick smoothly across the centre stick position.

Could I suggest that the range be changed to 0 - 100 in CLI (and matching 0-1 in configurator), and then the code would *disable* transition entirely if set to 0.  Once disabled, the D weight effect would apply smoothly and consistently regardless of stick position.

ctzsnooze, 2018-03-20:
See https://github.com/betaflight/betaflight/pull/5484

fujin, 2018-05-05:
Fixed by #5484

closed 2018-05-05 by fujin