betaflight/betaflight #5475: Can we readd RX_MSP feature?

Tachiorz, 2018-03-20:
RX_MSP missing for SPRACINGF3EVO after this pull. I was using it to fly, can we readd it somehow?
https://github.com/betaflight/betaflight/pull/5175

mikeller, 2018-03-20:
@Tachiorz: RX_MSP was dropped in favour of other features and protocols that are actually designed for flying. RX_MSP is meant for debugging only, and it is lacking some essential safeguards (like failsafe detection when the connection to the TX is lost), so it should really not be used to operate any craft with propellers attached and a battery connected. Please change your setup to an RC protocol that is actually designed for flying.

Tachiorz, 2018-03-20:
Even so, it was handy to fly mini quad with a PC and gamepad without the need to buy expensive transmitter :( But I see you point, sacrificing rarely used feature for more frequently needed makes sense. I can build my own if I want that feature. Or rewrite my software

mikeller, 2018-03-21:
Exactly, building a custom firmware for your target is relatively simple, since features like RX_MSP are enabled by defines, and you can make space for it by disabling other RX protocols that you don't need.

https://github.com/cleanflight/cleanflight/blob/master/docs/Customized%20Version.md has some hints.

jflyper, 2018-03-21:
And this may help, too, to setup an build environment.

https://github.com/betaflight/betaflight/wiki/Building-Betaflight-on-Windows,-the-Linux-subsystem-way

Tachiorz, 2018-03-21:
SBUS protocol looks pretty simple, so I'll close this issue.

closed 2018-03-21 by Tachiorz

mikeller, 2018-03-21:
@Tachiorz: There are libraries that implement the SBus protocol (like https://github.com/bolderflight/SBUS) that you could use. I think if you were to start a project to use a joystick device to generate an SBus stream on serial / bluetooth, you'd get a lot of interest.