betaflight/betaflight #5479: SPF3EVO - Softserial at BF 3.3.0 not working

JBlocks, 2018-03-20:
With the new betaflight version 3.3.0 it is not possible to enable the SOFTSERIAL feature. In 3.2.x it works like a charm. 

**Eays to reproduce the problem;**
Flash BetaFlight version 3.3.0 to your SPF3EVO flight controller and connect with the BF GUI. Try to enable the "SOFTSERIAL" feature with the GUI. Save and reboot. After reconnect to the fc, the feature is disabled. Same issue by setting it in the cli.
```
feature SOFTSERIAL
serial 31 2048 115200 57600 0 115200
```

The softserial feature is needed for SMART AUDIO, because all other RX/TX resources are in use.
Please readd the softserial feature in BF 3.3.x.

jflyper, 2018-03-20:
Software serial was disabled for SPRACINGF3EVO target to make some room for F.Port by #4158.

Changes since then seems to make it fit again, so I will submit a PR for reinstation.

closed 2018-03-21 by mikeller

jflyper, 2018-03-21:
@JBlocks 
We are sorry, `SPRACINGF3EVO` target is still overflowing, with all new features including the new filter facility devs are working on. Your only workaround would be to build your own custom firmware.

JBlocks, 2018-03-21:
Thanks for your support and the pull request. I will wait until version 3.3.1 including the fix.

jflyper, 2018-03-21:
@JBlocks Please note that there is no guarantee that software serial will resurrect in 3.3.1.

mikeller, 2018-03-21:
@JBlocks: I tried to re-enable softserial on SPRACINGF3EVO for 3.3.1, but there is not enough space: https://travis-ci.org/betaflight/betaflight/builds/356149860 :-(

JBlocks, 2018-03-21:
Would it be possible to remove modules like "Race transponder" or "Sonar" from the build? Or to release different builds like SPF3EVO_SoftSerial, SPF3EVO_default, ...?

mikeller, 2018-03-21:
@JBlocks: sonar (or rangefinder as it is called these days) is already disabled. The problem with disabling things like transponder is that it will leave users that are using this high and dry. Even without softserial, you still have 3 UARTs available that you can use.

PR #5480 (2018-03-21): SPRACINGF3EVO Reinstate software serial
Fixes #5479.