betaflight/betaflight #5482: OMNIBUS/OMNIBUSF4 fails to compile with current master code

Arakon, 2018-03-20:
%% (size optimised) vtx_string.c
%% (size optimised) vtx.c
%% (size optimised) vtx_rtc6705.c
%% (size optimised) vtx_smartaudio.c
make[1]: *** No rule to make target 'lib/main/CMSIS/CM4/DeviceSupport/ST/STM32F4xx/stm32f4xx_rcc.c', needed by 'obj/main/OMNIBUSF4/stm32f4xx_rcc.o'.  Stop.
make[1]: *** Waiting for unfinished jobs....
%% (size optimised) vtx_tramp.c
%% (size optimised) vtx_control.c
make[1]: Leaving directory '/cygdrive/d/betaflight'
make: *** [Makefile:408: hex] Error 2

No such issue i.e. with the BFF3 target.
Compiling on cygwin64, worked fine with master from a week ago or so.

lostcontrol, 2018-03-20:
Did you try to "make clean" first?

AndersHoglund, 2018-03-20:
If `make clean` does not help, try `rm -rf obj` instead. I had the same problem and  `rm -rf obj` fixed it.
CMSIS has moved, if you have old build artefacts remaining that refers to the old installation the build may fail.

Arakon, 2018-03-21:
Make clean didn't fix it.

Deleting obj did indeed fix it, though.
Thanks.. might be worth mentioning this somewhere in the compilation docs so people know the step to fix it.

closed 2018-03-21 by Arakon

DieHertz, 2018-03-21:
Having to clean build directory after switching revisions is pretty common.

If you performed a checkout/rebase/merge/cherry-pink or similar operation, you have to clean build directory in general with about any project.

lostcontrol, 2018-03-21:
`make clean` could do the rm job. Or another target like `make mrproper`.

Arakon, 2018-03-21:
`make clean` didn't do the trick, that's the problem.

mikeller, 2018-03-21:
@lostcontrol: I'm all for having `make mrproper` as the 'nuclear option'.

ledvinap, 2018-03-21:
IMO `make clean` should remove .d files too ...