betaflight/betaflight #5487: Issue with telemetry for Dshot1200

sskaug, 2018-03-21:
We had reports that BLHeli_32 gives intermittent zero output for telemetry: 
https://www.youtube.com/watch?v=dUmX3QUhzw8&feature=youtu.be

So I set up a system here to reproduce, I took a Typhoon 4in1 ESC, and a Bluejay F4 FC, set it up for telemetry (usart1) and used 16/16 rates. Flashed the latest 3.3.0 FW.

I hooked up a a scope on the telemetry wire, and for Dshot600 and Proshot1000, there are packets being sent every 20ms, which looks nice and clean.

Also for Dshot1200, when not armed, package transmission is clean at 20ms intervals. But when arming, and particularly when running the system, telemetry packets appear all over the place in the timescale. This is most likely the cause for the issues seen by users.

jflyper, 2018-03-21:
So it happens only with armed Dshot1200?

mikeller, 2018-03-21:
Also, on an unrelated note, does BLHeli32 now support ProShot1000?

sskaug, 2018-03-21:
Yes, in my setup the problem is there for armed Dshot1200.
And from Rev32.4, BLHeli supports Proshot. Proshot is more noise sensitive than Dshot for the same speed, I'm not sure how robust it will be in the field. So far I have not heard any reports on Proshot (neither good nor bad).

mikeller, 2018-03-21:
@sskaug: Interesting, and good to know. I always thought that ProShot was a good idea, and worth exploring some more, but I didn't know I had ESCs that support it.

RoadyFPV, 2018-03-24:
Confirmed at Typhoon32 V2 35A 4in1 ESC FW 32.4  with DSHOT1200 on Omnibus F7 V2.
Inflight the Voltage drop to Zero for 1 ore 2 seconds.

Saxin, 2018-03-24:
Confirmed on Wraith 35a v1, Omnibus Pro corner, Dshot1200, voltage dips to unreasonable levels in flight, happens in random manner. 
I can upload dvr if needed.

ave1, 2018-03-25:
Fighting with this issue for days, glad I finally saw this as at least now I know it is not something I am doing wrong as the same is occurring on the omnibus nano V5.1 and, ORI32.  I have been speaking to AirBot about this issue. I should (will)  try DSHOT 600.

controlflow, 2018-03-26:
Same here, AirBot NOX 2.17 @ DSHOT1200, getting zero voltage reading for a few seconds

SnappyFPV, 2018-03-27:
Confirmed Holybro Kakute F4 V2 + Tekko32 35A ESC (aka Wraith32 V2) DropOut to zero for Voltage & Current

mikeller, 2018-03-28:
@sskaug: Just to clarify: When you say 'telemetry packets', you mean the telemetry frames that are sent from the ESC back to the flight controller?

If so, this would hint at the most likely location of the problem being the ESC side. Triggering of the 'telemetry' flag in Dshot is agnostic of the Dshot protocol speed, and if it was a problem with setting the telemetry bit in the Dshot frame, this would cause the checksum of the telemetry frame to be corrupted.

Hint: There is a plugin for saleae logic that allows decoding of Dshot frames (but it does not yet support Dshot1200 I think).

sskaug, 2018-03-29:
Yes, I mean the telemetry frames.

Now I have dug a bit more into it, and I am surprised that systems work at all on Dshot1200 (at least if they behave like my FC (Bluejay F4 with 3.3.0 FW)). 

I see very bad behavior on motor #2 (for Dshot1200, not for Dshot600).
When I connect a KISS 32A ESC, it arms when the FC is disarmed. But when I arm the FC, the ESC disarms, and does not run at all. With a BLHeli_32 ESC, it runs intermittently and when debugging I see that there are lots of errors on the input signal frames. Which is probably also the reason why at times the telemetry bit is also detected.

Can you please have a look at channel #2 in a setup like mine?

ave1, 2018-03-29:
Its not only happening on Dshot1200. I have now noticed that Dshot 600 also can display the problem. I might be mistaken but, since I have been updating ESC firmware on some older FC/ESCs originally built 3-6 months ago, I do not always see the issue. It seems to me that some of my heavy larger motors, those with heavy clunker magnets like the Brother Hobby motors might also influence this temporary drop in voltage telemetry. I reference those motors drawing more current.

sskaug, 2018-04-02:
Just tested the BJF4 setup with a KISS32A ESC on channel #2, and the behavior is the same as with a BHLHeli_32 ESC on that channel. Telemetry responses way too often in a kind of random manner for Dshot1200.

skizztle, 2018-04-06:
Seen it with my setup as well. Haven't tested Dshot600 and below though.

ave1, 2018-04-06:
I have noticed that as the quad rest on ground, with Lipo plugged in, armed or not, the mAh used does not advance. Only once armed and in the air does the mAh telemetry number advance. I only have used the Ori32. Airbot has no explanation for this behavior. Each build responds differently where the mAh used is concerned. I have not tested separate 32bit ESCs or other model of 4 in 1 ESCs yet.

therealjonnyd, 2018-04-06:
^both of my Ori32s do the same. Even with the scale changed in blheli_32 FW, even low throttle still shows 0 amps while in the air. It doesn't matter if it's dshot600/1200 or Proshot1000.

mikeller, 2018-04-07:
@ave1: If you are using ESC telemetry, this is the expected behaviour. All that the ESCs are measuring is the current going into the ESC, which will be below the threshold for being picked up when the motors are not spinning. The consumption of the flight controller / RX / VTX / camera and any other devices (which is most of the consumption when not armed) can not be measured by the ESCs.

mikeller, 2018-04-07:
@ave1: Not sure about that. The power consumption is measured and reported by the ESCs, and then only summed up / displayed by Betaflight. We don't have any influence on how accurate and well calibrated the ESCs are. I think the latest version of BLHeli_32 allows the current sensor to be calibrated, maybe look into updating your ESCs and using this? @sskaug should know more.

etracer65, 2018-04-16:
@sskaug Does the telemetry CRC error fix in 32.4.1 resolve this problem?
```
Some fixes:
- Frequent CRC errors on telemetry frames shall now be fixed.
```

sskaug, 2018-04-17:
No, Rev32.41 does not fix the problem that I see.
Which makes sense, as it is only a problem for Dshot1200 and not lower speed Dshot. And the CRC issue is the same for all Dshot speeds.

sskaug, 2018-04-17:
Here are two log files (renamed to .txt).
I don't know how to read them, the BB explorer does not open them (it seems to hang).

[BTFL_BLACKBOX_LOG_20180417_201017.txt](https://github.com/betaflight/betaflight/files/1921128/BTFL_BLACKBOX_LOG_20180417_201017.txt)
[BTFL_BLACKBOX_LOG_20180417_203113.txt](https://github.com/betaflight/betaflight/files/1921129/BTFL_BLACKBOX_LOG_20180417_203113.txt)

mikeller, 2018-04-17:
@sskaug: The traces look like the telemetry responses can not be properly decoded when not armed, causing the receiver to time out after 100 ms, before sending the next telemetry request. Can you produce the same traces for Dshot600 please so that we can compare?

etracer65, 2018-04-17:
@sskaug I can't open the blackbox logs either.  I think you said you're running at 16/16K and there was a bug for blackbox logging with loop times faster than 8KHz that's been fixed but was not in BF3.3.  Can you set your gyro/PID to 8K/8K?

sskaug, 2018-04-18:
@mikeller: The disarmed trace shows telemetry responses every 20ms, presumably in a round robin style between the ESCs. Dshot600 also looks exactly like this (both when armed and disarmed).
If, however I unplug one ESC, I can see the 100ms timeout waiting for one ESC.
The second, armed trace, has telemetry responses much more often, and in a random manner. If I unplug ESC #2, then there is the 20ms spaced responses and the expected 100ms timeout. If I unplug any other ESC, the random response issue is still present.

sskaug, 2018-04-18:
@etracer65: I was running 8k/8k, but with a 32k gyro sampling. 
Here is 8k gyro and 8k pid - it opens in BB explorer.
[BTFL_BLACKBOX_LOG_20180418_072814.txt](https://github.com/betaflight/betaflight/files/1922537/BTFL_BLACKBOX_LOG_20180418_072814.txt)

mikeller, 2018-04-18:
@sskaug: Ok, I misread the timing of your scope traces then.

mikeller, 2018-04-18:
That looks like the ESCs are sending telemetry packets even when no telemetry requests are sent - in the cases where this is causing collisions this also explains the CRC errors.

mikeller, 2018-04-18:
@sskaug: These are the telemetry frames sent by the KISS ESC?

sskaug, 2018-04-19:
Yes, most of them. It is the combined frames from all 4 ESCs (with KISS on motor 2). And as we saw before, by removing ESC 2, the pattern becomes as expected again (the 100ms timeout).

mikeller, 2018-04-19:
@sskaug: The fact that this only seems to be affecting motor 2 seems to be oddly coincidental, and almost sounds like there is a hardware limitation on the resources driving the pin for motor 2 affecting the output. Do you have a different make of flight controller that you could use to cross check?

mikeller, 2018-04-19:
I tried this on my test bench. Unfortunately I only have 2 BLHeli_32 ESCs set up, and my KISS24 ESCs do not seem to be returning telemetry when running Dshot1200.

But the result with 2 ESCs is just the way it should be:

[image]

```
# version
# Betaflight / OMNIBUSF4 (OBF4) 3.4.0 Apr 19 2018 / 22:59:58 (0f6d4a2de) MSP API: 1.39

# set motor_pwm_protocol
motor_pwm_protocol = DSHOT1200
Allowed values: OFF, ONESHOT125, ONESHOT42, MULTISHOT, BRUSHED, DSHOT150, DSHOT300, DSHOT600, DSHOT1200, PROSHOT1000

# dshotprog 0 6
Using output 0.
Info for ESC 0:
ESC Type: #Airbot_Wraith32_ST#            
MCU Serial No: 0x
Firmware Version: 32.40
Rotation Direction: normal
3D: off
Low voltage Limit: 3.0
Current Limit: 40
LED 0: on
LED 1: off
LED 2: off
LED 3: unsupported
Command Sent: 6

# dshotprog 1 6
Using output 1.
Info for ESC 1:
ESC Type: #Airbot_Wraith32_ST#            
MCU Serial No: 0x
Firmware Version: 32.40
Rotation Direction: normal
3D: off
Low voltage Limit: 3.0
Current Limit: 40
LED 0: off
LED 1: on
LED 2: off
LED 3: unsupported
Command Sent: 6

# 
```

sskaug, 2018-04-19:
So for the BJF4, the issue is still there even if there is only one ESC, connected to motor 2. I do not have many other F4 FCs, but I have a REVO, and it does not exhibit the issue, as far as I can see.
Maybe you need to test with a BJF4? Maybe there is some sort of resource conflict that affects some FCs but not all?

mikeller, 2018-04-19:
Unfortunately I do not have a BLUEJAYF4 that I can test with. But can you generate traces showing the details of a Dshot1200 frame from motor2 on the BLUEJAYF4, and from one other motor? If there is a hardware limitation causing this then it should be apparent in the frame.
@blckmn is the designer of the BLUEJAYF4, maybe he is already aware of a hardware limitation there?

sskaug, 2018-04-21:
Here a pulse for every error on motor 2. Seems close to half the frames are in error. And the errors come in bursts:
[image]

etracer65, 2018-04-21:
@sskaug Separate from the reason that motor 2 on the BLUEJAYF4 seems to be having a hardware conflict, why would a corrupted frame be considered a telemetry request even if the bit was set?  Shouldn't the failed CRC just cause the frame to be thrown away?

sskaug, 2018-04-22:
Sure. Frames with XOR errors are thrown away. But there are also frames with correct XOR and telemetry bit set when it should not have  been set.

mikeller, 2018-04-24:
With that number of errors, could it be that the unaccounted for telemetry requests are caused by invalid frames with multiple bit errors, causing the CRC to appear valid?

migarnt, 2018-05-25:
I have a Holybro 4 in 1 35A and was having these same issues. I am on Typhoon 32 Rev. 32.41 Multi BLHeli Version 32.4.0.5. I turned off auto telemetry and these problems went away. Working as normal on dshot 600 1200 and Proshot1000

storm13, 2018-06-04:
Is there a solution in the works?

mikeller, 2018-06-05:
@storm13: As this is a BLHeli_32 problem (KISS ESCs don't have it), you should probably ask on https://github.com/bitdump/BLHeli/issues and not here.

Also, since Dshot1200 isn't recommended for general use, this is not an issue at all.

closed 2018-06-05 by mikeller

sskaug, 2018-06-05:
There can be various reasons for closing this issue. But moving the issue to BLHeli_32 is not the right action. The problem that opened this issue is not just a BLHeli_32 issue, it also affects KISS. See my post from April 2.

Saxin, 2018-06-05:
@mikeller - please reconsider leaving open at least for tracking purposes - this is big issue for many users, people want and will benefit from this issue resolved (no matter where the problem is, blheli, BF or the cable that connects them)

Meanwhile on RCG:
https://www.rcgroups.com/forums/showpost.php?p=39716815&postcount=2701

[image]

mikeller, 2018-06-05:
@Saxin: I don't see why more time should be wasted on something that has no benefit whatsoever. When we have a use case for Dshot1200 we can revisit this. Also, this cannot be replicated.

sskaug, 2018-06-06:
Maybe related to this?: https://www.rcgroups.com/forums/showpost.php?p=39760533&postcount=2747

TonyBazz, 2018-07-03:
@sskaug definitely related to it. The timer/dma HW seems to struggle above 1Mhz. Doesn't affect all channels though.

With custom builds of betaflight (where im testing dshot2400), its starts affecting all the output channels.

TonyBazz, 2018-07-03:
@sskaug Have a look at this video: https://www.youtube.com/watch?v=z6AdLGMXc7M&feature=youtu.be
At 0:42 seconds, full throttle, the telem bit is always set when it shouldn't be.

sskaug, 2018-07-03:
@TonyBazz Great input. Hopefully the BF team will dig more into this.

sskaug, 2018-07-04:
@TonyBazz Did you try the zero padding that Felix proposes?: https://www.rcgroups.com/forums/showpost.php?p=39691310&postcount=4194

TonyBazz, 2018-07-04:
Nope haven't tried it. But i dont see how the Zero padding will fix the the telemetry and frame corruption that occurs mid frame. It might fix the over/under run issues. Ill look into it further when as soon as i have time.

RoadyFPV, 2018-07-07:
I load the new BLHeli32 and read the manual.
- Rev32.5 
1 Added programmable sine modulation mode                  
2 Added programmable soft bidirectional mode                  
3 Added programmable autonomous periodic (every 32ms) telemetry mode                 
4 Activated hardware noise filter on the signal input                  
5 Fixed issue of corrupted telemetry CRC                  
6 Fixed issue in bidirectional mode where motor would twitch before stopping                  
7 Improved input signal detection, particularly for Proshot and also for Dshot at high input signal frequency                                   
Some smaller fixes 

Is it possible that point 5 can fix the telemety bug?
I cant load the new 32,5 in the moment.

sskaug, 2018-07-07:
@RoadyFPV Most likely not, the bug that is the topic of this issue, is improper Dshot signaling from the FC

mikeller, 2018-07-07:
@sskaug: You really have to stop blaming the bug in your firmware that makes it interpret frames with invalid checksum as valid on anything else than your firmware, while preventing everybody else from finding and fixing the bugs in your firmware.