betaflight/betaflight #5488: Lets get real RPM into OSD

Rakinroul, 2018-03-21:
I used hyperlite F4, AIKON 4in1 32BIT ESCS with telemetry, and i got some 4 digit number as RPM.
On 5percent idle i got number like 150.
I used first stable 3.3 firmware - i think. And last firmware from blheli 32 - flashed 15.3.2018 (dont know exact version now)

I belive all betaflight users will apreciate to get "real" rpm to OSD.
Question is, how to do it, when there is only one number , and motors are 4 :)
Add another 3 records to osd to have all 4 ?

Dont know. 1 can be enough, or aproximate number from all 4 ?
Its on you.

RipperDrone, 2018-03-21:
someone on #slack said it's erpm currently, so in order to know the true motor shaft rpm, you need to input the number of poles on your motors. So all I could imagine is adding another factor (# of poles) for scaling, like on the battery & power tab...

Rakinroul, 2018-03-21:
Yea, but i am worried it will not be enough..

Because eRPM should be much higher then number 150 in indle (5 percents) what i saw in my case.

Maybe there is some another "issue" between blheli32 and betaflight in communication.
Or is is just "some" ESCS telemetry problem in this esc whitch i using. Dont  know.

Someone should test how blheli32 (aikon 4in1 35A (?) ) telemetry work now curent betaflight ?

Dont know just droping tips and ideas.

controlflow, 2018-03-26:
I also wonder about the value OSD outputs when only one of four ESCs is wired to UART... is it averaged with three zeroes? With only one ESC wired my values are lesser than 500 even on throttle punches...

Rakinroul, 2018-03-26:
Yea, someone should get this working.

I must admint , have no idea, if it is blheli issue or betaflight.

If you have "telemetry" AIO cable from escs, i expect all info can be there without problem, when you use some kind of digital protocol.

I saw some kiss system video,s and looks like they have it working, separated rpms, current on all 4 motors in same time..

theSkonk, 2018-04-06:
i'd like this too its great for testing different props. to be honest you only need an average rpm or rpm of one motor as its really only useful for determining if your getting more or less rpm out of a certain set of props. also having all the motors on your osd is going to take up way too much space and cause a distraction. can we have a cli function to input the number of poles so the rpm displayed is actual @fujin ?

mikeller, 2018-04-06:
@theSkonk:

> i'd like this too its great for testing different props. to be honest you only need an average rpm or rpm of one motor as its really only useful for determining if your getting more or less rpm out of a certain set of props. 

If that is your concern then the current system will work perfectly fine, since it allows quantitative comparisons of eRPM for different props. No point in scaling them up with an arbitrary factor and potentially cofusing things.

theSkonk, 2018-04-06:
@mikeller that is true. The info is more useful afterwards and then it's not hard to multiply it by the number of poles to get the actual numbers. Still it is just a mulipler and would be Kool to get real rpm

Rakinroul, 2018-04-06:
It isnt about multiplier only i guess. As i mentioned above. Fact is i am using/used dshot1200

theSkonk, 2018-04-06:
@rakinroul it is a multipler. Its nothing more than the number of poles* the number Erpm. I'm running f40pro v2 2600kv motors on dshot 1200 if I take Te stator config, 12N14P.. if I times my max Erpm ( 2000 ) by the number of poles ( 14 ) I get 28000rpm. Excactly what the motor should be making under load. If you want all the motors etc then yeah much more complex. But coverting the current value to real rpm not so much

Rakinroul, 2018-04-06:
So in cli should be possible to set pole number, but 14 poles should be default right ?

theSkonk, 2018-04-06:
It could be if implemented. ( think and I could be wrong ) to take it to the next level would need some one to write the appropriate code.

closed 2018-05-13 by stale[bot]