betaflight/betaflight #5496: Board flashed with #693 (4786e1a) no longer shows up in com ports, related to PR #5478

flint723, 2018-03-22:
Related to "Add Composite CDC+HID device option. (#5478) (commit: 4786e1a)"

After flashing this, the board rebooted, made normal usb connect sound, but did not appear on com port list.  I guess it must be appearing as the usb composite device but it should not default to this.  The blue activity light does come on (not like a board with corrupted firmware).

I used boot button to put the board in DFU and flashed back to #692, all is well.  Flashed the board again with #693, same issue, cannot connect to configurator.

I tried MatekF405 and Omnibusf4 target, same thing.

This was on Windows 10 pro 1709.

richard-scott, 2018-03-22:
Same for me. the merging of "Support programming of Arduino devices in serial passthrough mode" in build [685](https://github.com/betaflight/betaflight/pull/5129#issuecomment-375073841) is when my FC stops responding on the USB serial port.

mikeller, 2018-03-22:
That's a shame. I tested #5478 on linux, and it works fine there. Unfortunately I only have Windows in a VM, and in this set up HID is handled separately from other USB devices, so I can't really test for Windows for this.

I'll get somebody to look into it ASAP.

DieHertz, 2018-03-22:
I can look on a windows machine tonight (in 5-6 hours).

mikeller, 2018-03-22:
@conkerh: What host OS did you test with? If Windows / OSX, did you have to change anything in the host / driver config to make this work?

jflyper, 2018-03-22:
MacOSX 10.13.2
Configurator 10.3.0
Flashed Omnibus F4 AIO (1st gen), Omnibus F4 V3 and Matek F405-Mini.

All came up normally and can be connected from the configurator.
In fact, the device is not recognized as composite device, but a regular CDC device. No HID device is visible (of course).

conkerkh, 2018-03-22:
Yeah I'm afraid we need to provide proper inf for Windows. It works fine under OS X and Linux, though OS X hides HID under devices but for sure it works in games.

conkerkh, 2018-03-22:
I'll have a fix in few hours need to work for now...

conkerkh, 2018-03-22:
@richard-scott can we narrow down the problem? Is passthrough on Arduino an issue or is Composite Device? Imho in windows Composite device without proper drivers would simply mask out HID interface.

richard-scott, 2018-03-22:
The same com port number (COM5) shows up in Windows and appears in the top of the Configurator... does that help??

BrenLaw, 2018-03-22:
I had no issues post build #685. But I can verify build #693 definitely breaks support for non-VM Windows 10 installations. Like @richard-scott , my COM port still shows up like normal in Betaflight configurator, but I cannot connect. If I go to Device Manager, the STMicroelectronics VCP driver has an error sign. Viewing its device status shows this text, "This device cannot start. (Code 10) A device which does not exist was specified."

flint723, 2018-03-22:
Agreed, #685 was fine for me, I'm only affected by #693+ but in my case it doesn't show up as a com port at all.  I do see the same message in device manager.
[image]

richard-scott, 2018-03-22:
For me, the com port on my Windows 10 Laptop with the AnyFC F7 board is OK according to device manager.

conkerkh, 2018-03-22:
@richard-scott CDC+HID is only on F4 atm, I'm working on a hotfix for Windows

conkerkh, 2018-03-22:
Can everyone who posted here provide Windows version they are using along with what is exactly in device manager?

flint723, 2018-03-22:
@conkerkh anything more I can provide besides the above images?  Mine is W10 Pro 1709.

conkerkh, 2018-03-22:
@flint723 32bit or 64?

flint723, 2018-03-22:
Whoops, 64 bit.

etracer65, 2018-03-22:
This also breaks support on older versions of OS X but the latest 10.13 seems okay.

etracer65, 2018-03-22:
I would suggest we need to revert this until it can be further tested and resolved.  I'm completely dead in the water and unable to test any development.

apoc, 2018-03-22:
You can revert it localy

BrenLaw, 2018-03-22:
Sorry for the wait. I'm running Windows 10 Pro build 1709 x64. 
My flight controller (F4) is still recognized in Device Manager as STM VCP device (COM5) like it typically is when everything is working correctly. Except there's that error icon I mentioned next to it. That's sums up everything for me.

mikeller, 2018-03-22:
@conkerkh: Seeing that this is not as seamless, and causes connection problems with a number of host OS, I think for now the functionality should be changed to be switchable with a parameter, and default to off.
Do you have time to open a pull request (soon-ish) to add this parameter, or should somebody else take care of this? Your code is already nicely modularised, so it should be straightforward.

conkerkh, 2018-03-22:
@etracer65 @BrenLaw @flint723 please checkout #5512 compile with USB_CDC_HID define in your target and let me know if it works. I tested this on OSX 10.13 and WIN10 both work fine. Windows required to replug device once drivers where installed and then worked ok. 

However I have no idea on pre W10 so I would kindly ask to test on older Windows (also OSX)

etracer65, 2018-03-22:
@conkerkh I can confirm it works on OS X 10.13 but also **does not work on older versions of OS X**.  I don't know exactly where the version cutoff is though.  Tested on two old machines (10.8 and 10.9) and they didn't work.  Don't have any other machines to test.

Do you have time to put some put an on/off parameter around it for now?  If not I can go ahead and submit a PR to implement this to default to disabled to allow people to continue testing.  People will be able to set the parameter back to "ON" for testing.

etracer65, 2018-03-22:
@conkerkh Disregard - read your post on email and didn't see the extra change.  I was referring to the original PR - I'll try to test #5512.

conkerkh, 2018-03-22:
@etracer65 thx

flint723, 2018-03-22:
Wish I could, am unable to compile myself.

conkerkh, 2018-03-22:
I'm on slack chrish drop me a line I'll send you hex

flint723, 2018-03-22:
Sent but maybe didn't go through since you are asleep!

etracer65, 2018-03-22:
@conkerkh I commented on the PR but the changes fixed the problem with older versions of OS X.

flint723, 2018-03-22:
Confirmed it works on W10, it created a completely new com port.

conkerkh, 2018-03-22:
Ideally someone could test this on W8 and W7 which are the only one that worry me, but I don't have such old OS anywhere:/

flint723, 2018-03-23:
I can try w7, will let you know

BrenLaw, 2018-03-23:
Yep, this works for W10. Good job! My COM port is still the same as it has always been. Trusty old COM5. And the STM VCP displays in Device Manager how it's supposed to.

I can test a Windows 7 machine right now. Well edit this post in a few minutes after I test it.

flint723, 2018-03-23:
After flashing from master again (700) it went back to previous com port.  Great!

@conkerkh I'm going to close this unless you would like to wait for W8 confirmation.

conkerkh, 2018-03-23:
I'm more concerned if joystick shows up as a HID device in W7 and W8 assuming you compiled f/w with define I mentioned.

BlueFishUK, 2018-03-23:
I have same issue and cannot get PC to connect no matter what I try at the moment.
[image]

[image]

[image]

richard-scott, 2018-03-23:
@BlueFishUK Try a hex from this build.
https://ci.betaflight.tech/job/Betaflight/684/artifact/obj/

This is the last one that works for me, anything newer than that and the Serial Port shows up cleanly in Windows, but does not communicate with the Configurator.

If build 684 works for you but build 685 does not then please add a comment onto #5502 to keep it all in the same place.

conkerkh, 2018-03-23:
Just use current build fix was merged couple of hours ago...

closed 2018-04-03 by flint723