betaflight/betaflight #5497: OSD Post Flight Statistics bleed through into menus 3.4.0 (minor issue)

flint723, 2018-03-22:
It happened somewhere between commit ab231b7(is fine) and b98bdcc(has the osd issue).   The issue is still there in b2e807b.  I'll try to narrow it down more tomorrow.  I had the issue on matekf405 and ominbusf4 target.

Short video of it.
https://youtu.be/7QHTt69wHc8

etracer65, 2018-03-23:
I think I know what it is.  I'll look into it.

etracer65, 2018-03-23:
It's an unintended side effect of #5463 which made the stats screen "live" and continue to refresh so that it could follow the state of the OSD disable switch mode.  I need to change it to exit the OSD stats if you enter the OSD menus.

As a temporary workaround while I work on it you can either wait for the stats to clear on their own (takes 60 seconds), or manually dismiss them with either throttle high or pitch high before entering the OSD menus.

flint723, 2018-03-23:
Yep, it's not a big deal.  Thanks!

etracer65, 2018-03-23:
Created https://github.com/betaflight/betaflight/pull/5522

@flint723 If you want to test and can't build your own hex, let me know what target you need and I'll make one for you.

flint723, 2018-03-23:
Should I just dm you on slack?

PR #5522 (2018-03-24): Fix OSD stats not clearing when entering OSD menu
Fixes #5497 

The pitch stick high can be used to clear the OSD stats screen, but it's also part of the sequence to enter the OSD menu. Entering the OSD menu from the stats page should have cleared the stats, but because of a race condition they weren't always cleared and the OSD menu would draw on top of the stats.  Changed the logic to ensure that if the OSD menu is entered the stats page is exited and cleared.
closed 2018-03-24 by mikeller