betaflight/betaflight #5498: BF3.3 - OMNIBUSF4SD - FPort - Control but no Telemetry

ArchmageAU, 2018-03-22:
F.Port receives channel values, but no telemetry is returned to TX (VFAS, Curr, AccX, etc..).  Move TX sticks and channels move in the BF configurator receiver window.

TX: X9D - D16 (Ch1-8)  (OpenXT 2.2.1)
RX: R-XSR - (s.port wired to uninverted pad - Firmware: 180111)
FC: OmnibusF4SD board (OMNIBUSF4SD (OBSD) 3.3.0)
Serial RX is on UART3 (normal/uninverted UART)

CLI
# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.3.0 Mar  2 2018 / 03:56:35 (177472b4f) MSP API: 1.37

# feature list:
Available: RX_PPM INFLIGHT_ACC_CAL RX_SERIAL MOTOR_STOP SERVO_TILT SOFTSERIAL GPS RANGEFINDER TELEMETRY 3D RX_PARALLEL_PWM RX_MSP RSSI_ADC LED_STRIP DISPLAY OSD CHANNEL_FORWARDING TRANSPONDER AIRMODE RX_SPI SOFTSPI ESC_SENSOR ANTI_GRAVITY DYNAMIC_FILTER

# set serialrx
serialrx_provider = FPORT
Allowed values: SPEK1024, SPEK2048, SBUS, SUMD, SUMH, XB-B, XB-B-RJ01, IBUS, JETIEXBUS, CRSF, SRXL, CUSTOM, FPORT

serialrx_inverted = OFF
Allowed values: OFF, ON

serialrx_halfduplex = ON
Allowed values: OFF, ON

#diff 
# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.3.0 Mar  2 2018 / 03:56:35 (177472b4f) MSP API: 1.37

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE
feature ESC_SENSOR
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map
map TAER1234

# serial
serial 0 8192 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 1 2 1700 2100 0
aux 2 2 2 1300 1700 0
aux 3 13 0 1150 1350 0
aux 4 35 0 1400 1850 0

# adjrange
adjrange 0 0 3 900 2100 12 3

# rxrange

# vtx

# rxfail
rxfail 3 h
rxfail 6 s 2000

# master
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1990
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set motor_pwm_protocol = DSHOT1200
set current_meter = ESC
set battery_meter = ESC
set vbat_scale = 120
set beeper_dshot_beacon_tone = 1
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_vbat_pos = 2455
set osd_rssi_pos = 2106
set osd_tim_2_pos = 2081
set osd_craft_name_pos = 2442
set osd_avg_cell_voltage_pos = 2433
set debug_mode = FFT
set vtx_channel = 5
set vtx_freq = 5820

# profile
profile 0

# rateprofile
rateprofile 1

set roll_rc_rate = 60
set pitch_rc_rate = 60
set yaw_rc_rate = 60

#

jflyper, 2018-03-22:
@ArchmageAU 
Can you tell us how you exactly wired your R-XSR to the UART port?

A pic showing both UART and R-XSR would be the best.

Arakon, 2018-03-22:
Did you re-scan for sensors? Do you get RSSI returned? If RSSI is missing, the issue is entirely between transmitter and receiver, not related to BF.

ArchmageAU, 2018-03-22:
R-XSR ground to ground.
R-XSR voltage to +5V.
R-XSR s.bus un-inverted pad to TX3 (B10)

[image]

(Excuse the poor soldering, but if connections did not work, no signal would be received).

I do get RSSI returned (and RxBt) 
I did re-scan for sensors with the battery connected.  RSSI and RxBt are the only ones discovered.

Arakon, 2018-03-22:
You have ESC Sensor active. Do you actually HAVE ESCs with sensors? Because if not, you'll get no data.

ArchmageAU, 2018-03-22:
iPeaka BORG 35A 32bit BLHeli Dshot1200 4in1 ESC (with heat sink :) ).
http://www.iflight-rc.com/ipeaka-borg-32bit-blheli-dshot1200-4in1-esc.html

ESC does have telemetry and the telemetry does work (Displays voltage in Betaflight Configurator).

ArchmageAU, 2018-03-22:
Noticed there are updates to R-XSR firmware (In FPort 180228  - https://www.frsky-rc.com/r-xsr/ ).

Should I apply this firmware to the R-XSR?

jflyper, 2018-03-22:
@ArchmageAU 

# You have to connect "Uninverted _*S.Port*_" to TX3.

ArchmageAU, 2018-03-22:
Note the photo.  (And where the green wire is soldered to the R-XSR and the TX3)

The S.port wire is connected to the un-inverted s.port pad on the R-XSR (not an inverted connector on the plug).  The S.Port connector on the plug has been removed.  Only power and ground remain on the plug (red and black).

jflyper, 2018-03-22:
Oh, then your
> R-XSR s.bus un-inverted pad to TX3 (B10)
was an incorrect description.

I have locally flashed Omnibus F4 V3 (OMNIBUSF4SD target v3.3.0), loaded your config diff, connected XSR (not R-XSR) flashed with F.Port firmware.

This setup works without any further tweaking.

Since you have RC channels working, I suspect a bad flash on your receiver.

ArchmageAU, 2018-03-22:
Sorry for the wrong description (which is why I included the photo).

R-XSR **s.port** un-inverted pad to TX3 (B10)

I'll re-flash the R-XSR (this time with 180228) and report back.   Was originally flashed with 180111 over a month ago.

ArchmageAU, 2018-03-22:
Re-flashed R-XSR with 180228.  Telemetry returned.  Signals still being received.

Can close issue.

Will re-open if problem re-surfaces.

closed 2018-03-22 by mikeller

frizzle101101, 2018-05-16:
i still for the life of me on non-eu 180228 cannot get telemetry. This is on a nox on serial rx 2, i get teh controll just not telemetry!

mikeller, 2018-05-16:
@frizzle101101: It won't work properly on an RX pin, it needs to go onto a TX pin.

SangaeLama, 2018-07-12:
Same here
I am on the 180228 firmware but same problem, I have connected the Fport(S-port) wire to the Sbus pin on my board. Everything(Sbus controls) works fine including the RSSI and RxBT but no other sensors are discovered.
Please help.

SangaeLama, 2018-07-12:
@ArchmageAU

ArchmageAU, 2018-07-12:
@SangaeLama 
Sbus pad on F405CTR is an inverted RX.

F.Port needs either the un-invert hack and to be wired to a UART TX port, OR no un-invert hack and wired to a TX port with a bi-directional inverter.

The F405CTR does not have a dedicated S.port pad (inverted TX), and most instructions get s.port working by using motor 5 and using soft-serial for telemetry (not good for F.port).  Best option is doing the un-invert hack on the R-XSR and using a regular UART.

Hope that all makes sense.

dhirennaidoo, 2019-05-29:
I know this is quite late, but for others who experience this issue, I had to rebind transmitter to start receiving telemetry.

R-XSR
Taranis Q X7
OMNIBUS F4 SD (LDARC)

Uninverted SPort (soldered pad) to TX3 on FC

yagormeister, 2019-07-25:
hello!! another late comment in this thread! but... im experiencing same issue.
i was trying to get to work an rxsr with a wizard x220s, which has an F4 v3. 
My problem was that I couldnt get any comunication between reciever and FC on any of the UARTS, not even with the uninversion hack, and playing and trying all permutations of halfduplex and noninverted commands in cli.
Eventually I decided to connect the s.port wired from reciever to the sbus pad on the FC, and seems like Im getting comunication between transmitter and FC (enabeling Rx on UART 6??) but no telemetry at all!!!
Whyyyyyyyyy!!! any help would be appreciated!!