betaflight/betaflight #5502: Board flashed with 3.4.0 #685 (5558174) won't open serial port, related to PR #5129

richard-scott, 2018-03-22:
PR https://github.com/betaflight/betaflight/pull/5129 has been merged into 3.4.x on build 685 and this has stopped my AnyFC F7 from booting. I get the serial port show up in Windows, but I can't connect.

If I roll back to build 684 BFC can connect, but anything 685 or above doesn't allow me to connect BFC to it via USB.

FC is working and I can fly the quad. However Black Box is showing gaps of data but I think that's a different issue as BB data is gappy in 684. I'll try and narrow down which release caused the gaps and open a separate issue for that.

BlueFishUK, 2018-03-23:
Tested with 684 Hex and it works also works with 685 on Windows 10 64 bit with Omnibus F4 V5

richard-scott, 2018-04-02:
It seems there is no PINIO1_PIN defined for the AnyFC F7 [target](https://github.com/betaflight/betaflight/blob/master/src/main/target/ANYFCF7/target.h), this was added to some other targets in this PR. 

Could that be it? It's still broken in the nightly build 733.

richard-scott, 2018-04-06:
phew, at least its not just me being a dope ;-)

richard-scott, 2018-04-06:
@RobbieHood any chance you could try [684 ](https://ci.betaflight.tech/job/Betaflight/684) to see if it works, and [685](https://ci.betaflight.tech/job/Betaflight/685) to see if its broken on that build?

RobbieHood, 2018-04-06:
Sure will do.

mikeller, 2018-04-07:
Just tested latest in master on a FPVM_BETAFLIGHTF7, and it's working just fine.

Can't test with ANYFCF7, since I don't have this board. Maybe @sambas can help?

Have you all tried flashing with 'full chip erase'?

RobbieHood, 2018-04-07:
Yes offcourse im not a noob ^^
Do you run windows on your mschine?

mikeller, 2018-04-07:
@RobbieHood:

No reason to feel offended - excluding potential reasons is part of the process to identify the real reason.

Just tested with Windows, and the FPVM_BETAFLIGHTF7 with the latest version of 3.4 is recognised just fine.

RobbieHood, 2018-04-07:
Oh no I was not offended in any way my friend.
Which version of windows is on that machine?

For me everything is fine and can flash, connect and configure on any build
except for the latest builds from this week. Did not check cleanflight if
it has the same issue but its the same for betaflight and butterflight so
far.

Rolling back to a release version or older build fixes the issue.

mikeller, 2018-04-07:
@RobbieHood: Windows 10.

RobbieHood, 2018-04-07:
Hmm now I am getting lost. 😂

Expected it to be a win10 issue of some sort.....

RobbieHood, 2018-04-07:
#685 lost connectivity
Rollback to 3.3.1 stable connectivity returns
#748 lost connectivity
#747 lost connectivity
#746 lost connectivity
#745 lost connectivity
#744 connectivity returns sometimes.... this gets weird

RobbieHood, 2018-04-07:
ok lets make this confusing.

Flashing 3.3.1 always gets a stable connection
Flashing from 3.3.1 to #684 always gets a stable connection.
Flashing from 3.3.1 to #744 gives a sketchy connection. (sometimes it works sometimes it won't)
Flashing from both these build to eachother looses the serial connection but return when rolling back to their respected selves (if it worked to begin with for #744.) , rest of the builds loose connection even when flashing from 3.3.1

#743 #742 ....... #685 no connectivity

No clue here.....

richard-scott, 2018-04-07:
Thanks for checking @RobbieHood, the nightly build of [685 ](https://ci.betaflight.tech/job/Betaflight/685/changes)is when my problems start too.

RobbieHood, 2018-04-07:
Yeah suspect it has something to do with the arduino passthrough interface.

mikeller, 2018-04-07:
Maybe @SteveCEvans can provide some help here? I don't know enough of Windows drivers to be able to tell what difference the addition of DTR support on the client side makes to the Windows device driver.

SteveCEvans, 2018-04-07:
Just tested using Windows 10 (albeit running under parallels), and it's working fine. I was able to connect to an OpenPilot Revolution (F4) using the Betaflight configurator, run `serial passthrough 5 0 rxtx 1`, connect using MW_OSD to access configs, and also use the GUI to reflash. This demonstrates correct DTR handling, and setting of the serial baud rate over USB. I'm using the Microsoft USB Serial Device driver, version 10.0.16299.15. So, I don't think this is a Windows driver issue.

SteveCEvans, 2018-04-07:
I'm afraid I don't have an F7 FC to test this. Has anybody tried rebuilding with some of the changes backed out from `git diff 3d4f0bb1 src/main/vcp_hal/usbd_cdc_interface.c`? Simply preventing baudRateCb or ctrlLineStateCb being set in CDC_SetBaudRateCb or CDC_SetCtrlLineStateCb respectively should back out the only behaviour that's changed. Apart from that the CDC_GET_LINE_CODING handling in CDC_Itf_Control has some extra checks, but I can see that making anything worse, only better

As serialSetBaudRateCb() and serialSetCtrlLineStateCb() are only called from cliSerialPassthrough() though I struggle to see how this could possibly make a difference at startup.

RobbieHood, 2018-04-08:
I can build it but not too sure what to change.
My git is rusty lol

RobbieHood, 2018-04-08:
Nice thanks for the explanation @SteveCEvans.
Going to walk my dogs and will compile a build with those edits.

SteveCEvans, 2018-04-08:
@RobbieHood Now we need to understand why. Can anybody with an ANYFCF7 debug this?

RobbieHood, 2018-04-08:
Ok will do

SteveCEvans, 2018-04-08:
@RobbieHood Whoops, sorry, yes 209. I was suggesting the change just after things broke, so commit hash  5558174d.

RobbieHood, 2018-04-08:
Great resetting, pulling, editting and building now ^^

RobbieHood, 2018-04-08:
Ok so the last edit also works on the FPVM_BETAFLIGHTF7.

SteveCEvans, 2018-04-08:
Good stuff. I can see what’s happening. I’ll create a PR to fix this tomorrow evening when I get a chance. Thanks for helping debug this @robbiehood

RobbieHood, 2018-04-08:
No problem @SteveCEvans glad to be of help and great to work with a bit of OOC again ^^

mikeller, 2018-04-08:
Thanks guys, good work debugging!

richard-scott, 2018-04-09:
I can build from a branch and flash to my AnyFC F7 to test if it helps?

RobbieHood, 2018-04-09:
Try the f7 opt branch

richard-scott, 2018-04-09:
@DieHertz - With the same config cut-and-pasted back in my F7 board now uses 2 - 3% CPU on the bench rather than 12 - 14% with 3.3.1. Nice work guys :-)

RobbieHood, 2018-04-09:
Build 3 did not work for me though ^^

RobbieHood, 2018-04-09:
And yeah andrey is doing a great job. 41% here 32/32, acc enabled, dynamic filter and fport, runcam device tbs smartaudio on uarts

SteveCEvans, 2018-04-09:
See https://github.com/betaflight/betaflight/pull/5659

DieHertz, 2018-04-09:
I'm happily great successful attempts in using it, soon merging to master then

richard-scott, 2018-04-11:
I'm happy to report that build [760](https://ci.betaflight.tech/job/Betaflight/760/) works for me.

closed 2018-04-11 by richard-scott