betaflight/betaflight #5507: ICM20602 Accelerometer Config

Neilgaede, 2018-03-22:
I couldn't find any reference to the accel_config2 register for the ICM-20602 and the register mapping looked wrong. Is there any provision for setting the accelerometer low pass filter separately from the gyro low pass filter on the gyros that have the capability?

etracer65, 2018-03-24:
No, the ACCEL_CONFIG2 register is not currently set to anything but defaults.  For the ICM20602 this results in 4 sample average with a 1KHz sample rate and 218Hz filter cutoff.  What looks wrong about this?

What is it you're trying to accomplish?

NeilGaede, 2018-03-24:
I was just wondering since the low pass for the accelerometer is set with the gyro on the MPU6000 if it makes any difference on the ICM20602. If the low pass filter is changed for the gyro, does the accelerometer need to be changed as well, since they are set separately?

etracer65, 2018-03-24:
You are correct in that for the MPU6000 the DLPF setting affects both the gyro and accel whereas they're separate in later gyros.  In our application with the relatively slow sampling of the accel (1KHz) and the filtering already in the firmware I'm not sure there's an argument to mess with the DLPF for the accel.  It's unfortunate that changing the gyro DLPF on the MPU6000 will also affect the accel (unavoidable in the hardware).  However the use cases for using `gyro_hardware_lpf` set to other than "NORMAL" on the MPU6000 are quite limited.  Technically the "EXPERIMENTAL" option (DLPF_CFG = 7) is officially marked as "reserved" by Invensense and no documentation is provided for this value.  If does seem to affect the DLPF and significantly raise the filter cutoff (observable by much noisier output) but it's unkown what the cutoff actually is (or possibly the DLPF is completely bypassed).  So anybody playing with this on a MPU6000 needs to be aware of the possible impact on accel performance as well.

etracer65, 2018-03-24:
In looking at the MPU6000 initialization code I noticed something odd though.  I'm not sure why the accelerometer is being explicitly configured with an 8G scale when the sensor is capable of 16G.  The MPU9250 is also set up this way whereas every other gyro is using full-scale 16G.

I'm going to raise a separate issue for discussion rather than just fixing it.  There might have been a valid reason at some point.

etracer65, 2018-03-24:
Opened this issue relating to the MPU6000/MPU9250 only set for 8g.

https://github.com/betaflight/betaflight/issues/5530

NeilGaede, 2018-03-27:
Ok, so to summarize, you are saying it is best to have the accel DPLF set the same as the gyro DPLF as on the MPU6000, or just let it stay at the default as on the newer gyros, regardless of what the gyro is set at? 
As for the accel maximum G scale, would the advantage/disadvantage to using a lower (8G) scale possibly be that it is more sensitive? 
Thanks again for the info.

etracer65, 2018-03-27:
@NeilGaede Not exactly what I said.  The point is that on the MPU6000 the accel DLPF runs at the same cutoff as the gyro - there is no separate option.  This is unfortunate as changing the gyro DLPF can have unexpected effects on accel performance.  For the other gyros I see no reason to adjust the accel DLPF from the defaults regardless of the gyro DLPF cutoff.

Technically the 8g scale is "more sensitive" but also runs the risk of losing useful data that might exceed 8g (because the output clips at 8g).  So as a result the attitude estimate might be less accurate which might affect self-level modes.  Every other chip is running at 16g and the changes for MPU6000 and MPU9250 clearly appear to be mistakes as they both were previously set at 16g in older versions.  Also we know that today's high-powered quads can exceed 8g punchouts.

NeilGaede, 2018-03-28:
Got it. Leave the accel DLPF at factory and set it to 16g. Is that correct?

etracer65, 2018-03-29:
I simply corrected the scale to 16g for both MPU6000 and MPU9250.  No other changes were made.

For chips that have separate gyro and accel DLPF settings (everything except for MPU6000 and MPU6050), I don't see any value in changing the accel DLPF cutoff.  And for MPU6000 and MPU6050 we can't change the accel DLPF cutoff anyway because it's connected to the gyro cutoff.

NeilGaede, 2018-03-29:
Sounds good. Thank you.

closed 2018-03-29 by NeilGaede