betaflight/betaflight #5516: STM32H7 support for BetaFlight

MJ666, 2018-03-23:
Not really an issue but Slack does not work fro me for some reason? Is anyone already working on STM32H7 support? The AFNG team start working on new H7 FC designs. I have ordered an Nucleo-H743ZI board but it will take a while till it gets available and delivered. The STM32H743IIK6 is already available and looks to be quite interesting to be used for an FC. I have started to work on an target for the Nucleo board. and will also planning to start work on the ALIENFLIGHTNGH8 target soon. If there is someone also out there working on H7 support also we may be able to work together and share efforts? Please let me know. Thanks.

DieHertz, 2018-03-23:
@jflyper worked on it to some extent, I will too once my Nucleo arrives

jflyper, 2018-03-23:
What I've done so far:
- Imported H7 libraries from CubeMX
- Trying to shrink as much features as possible to come up with a firmware that runs (may be as small as USB + CLI + Config flash).

I do have a couple of Nucleos, and 100QFP part waiting to be transplanted onto ANYFCF7 once something flyable is ready.

How are we going to collaborate? Slack channel & BF branch?

DieHertz, 2018-03-23:
Nice idea about transplanting, I have a bluejayf7 for that. Where do I get an h7 100pin part? Could someone help me and forward it?

conkerkh, 2018-03-24:
@DieHertz farnell, rs-components, mouser?

conkerkh, 2018-03-24:
I'll get NucleoH7 and can help you out as well.

MJ666, 2018-03-24:
I can publish my work in progress next days on my github. Not sure if it makes sense to make an early PR at this stage. Looks to be the Nucleo will come before end of April or even in May.

MJ666, 2018-03-24:
Using the ANYFCF7 and the 100pin LQF H7 version should really work to get an H7 based FC very fast.

In my knowledge the BlujayF7 is using the F722 wich is a 64pin LQFP case. There likely no way to get a matching H7 with the same footprint by now? 

Here is my actual working tree but it is just work in progress for now: https://github.com/AlienWiiBF/betaflight/tree/Nucleo_H743
I actually logging into fixing the loader file(s) since all flash sectors are equal in size and we can have the configuration space unlike the F4 and F7 back in the last sector. I will update this on a regular bases. may be an branch in BF would be good to work on this as a team?

If someone have an better plan how to cooperate please let me know. I tried to discus on slack but its not working fro me. Always get an empty page.

DieHertz, 2018-03-24:
BluejayF7 is using a 100pin F745

MJ666, 2018-03-24:
The F7 from Bluejay is named Nero and is definiely using an F722? Do you have a new prototype from Jay wich is not officially supported by BF yet?

DieHertz, 2018-03-24:
I have a prototype which was never released. BluejayF7 is not the same as Nero.

DieHertz, 2018-03-24:
Anyway, who could forward a chip to me? :) Digikey wants $120 for shipping

MJ666, 2018-03-24:
@DieHertz For now only the STM32H743IIK6 (176+25UFBGA version) is available all others versions have 30 weeks of delivery time at Digikey? Usually if i buy from Germany over 50€ the shipping is free. Where do you live?

DieHertz, 2018-03-24:
Moscow, Russia

jflyper, 2018-03-24:
rs-online seems to have some (72) LQFP100 part (VI) in stock.

brucesdad13, 2018-03-25:
I have the H7 Nucleo also. @jflyper and I ordered same time 😁 Happy to join in the efforts.

RicoKirstein, 2018-03-27:
I don't have devboards, but I made some custom boards for the 176+25 variant. They should arrive somewhen in the next 2 weeks. I haven't soldered any BGA yet, but got access to someone and the equipment for it. The SDio interface is on some pads to solder on an old micro sd adapter or something.
Didn't think anyone would be on the software side before I had the boards. I pretty much started on them as soon as they were available on digikey. I really hope the 100 TFBGA package becomes available later this year. Tolerances for vias are really tight on the UFBGA parts.
[image]

DieHertz, 2018-03-27:
We should aim for STM32H743VIT6, 100-pin LQFP package. I'm trying to get some from TaoBao and going to work on it with you guys as I'm finishing BF F7 improvements

MJ666, 2018-03-28:
Fortunately the interesting chips of the H7 generation are the H743 for now. The H753 don't make real sense since the additional features are not needed for an FC. This means if the H7 support will finally support all H743 chips and hardware devs have the freedom to chose what best fit there needs. This may change in the near future when more H7 chips get released (H7x5/H7x7).

DieHertz, 2018-03-28:
I'll have a Nucleo and a couple bare chips within one month from now, then I'll start working on it

MJ666, 2018-05-21:
My Nucleo board has been arrived some days ago. I will continously to work on this now when finding some time.

Sm1thson, 2018-06-25:
@DieHertz  did you get a H7 chip? you can order a free sample (or 2) from https://www.st.com/ if you have a corporate email address. I assume they will deliver to your part of the world.

conkerkh, 2018-06-25:
@Sm1thson there are no H7 samples on the portal, you have to contact ST sales directly and have good relations with them

jflyper, 2018-10-15:
@MJ666 You can't use slack?

MJ666, 2018-10-15:
I can use slack but not there very often.

tloki, 2018-12-27:
What is the current state of H7 development?

mikeller, 2018-12-29:
@tloki: Work in progress.

closed 2026-09-26 by blckmn