betaflight/betaflight #5521: Racoon Feature (Floaty)

quadmx, 2018-03-23:
Wondering if there is an application to add a Mario racoon feature, in which when you are upside down the props engage in the other way (similar to 3d) to give a floaty sensation when upside down. Maybe with the accelerometer we can define ceritain angles in which it engages when trottle is bellow certain threshold.
Regards,
QuadMx

mikeller, 2018-03-24:
@quadmx: Can you describe how what you are suggesting is different from 3D mode?

quadmx, 2018-03-24:
Yes, so basically is for those who fly normally without 3d.
So this is how it normally works; min throttle makes it feel floaty when throttle is off
[image]

With Mario Tail its basically the same as min throttle but, with the props engaged on the other direction. to give it certain floaty feeling when you are upside down. (if you aren't flying upside down, the drone behaves as always).
[image]
This inverse spining should turn on when you are upside down and within certain anglo and not turning above certain deg/s.
Regards

JonnyPhenomenon, 2018-03-25:
Ick.... If this was something that would happen "automagically", I would think It would probably end horribly... kind of like the "fpv cam angle mix", where all of a sudden your drone is doing something you didnt tell it to, but worse.  
But OK, lets think about it....

how about this: what if you set up the momentary switch on your TX to engage a reverse thrust mode, that you could either control with the throttle as normal, or  maybe just pre-set it to a certain throttle level.  To use as sort of an extra boost, for more hang time, or better yet, when you realize the powerloop you just tried to execute came up short, and you are going to have to roll out of it.... 

So you punch the throttle, steadily pitch back, then chop the throttle as you invert, see you want a little more nudge, flip the momentary for the extra surge, and release it when you are ready to return to normal flight... 

thats actually a kind of cool idea.  - but doesnt the standard "3d mode" already work like this? 

Well, the BF wiki says that the 3d mode is controlled by the position of the throttle, Not by a switch. so full throttle reverse is when the stick is fully low, and full throttle normal is when the stick is high.  So the output for the TX **Could** be configured so the throttle range would be 1500-2000ms stick low to stick high, but when the SH switch is Flipped on, then the range could be increased from say 1400-2000, which would give you 20% of reverse thrust if your stick was fully low....  

Interesting.....

mikeller, 2018-03-25:
@JonnyPhenomenon: There's a switch for that...
(sorry but couldn't resist)

See the paragraph on `3d_switched_mode` in the [release notes](https://github.com/betaflight/betaflight/blob/master/docs/Upgrading/Upgrading%20from%203.2%20to%203.3.md).

quadmx, 2018-03-25:
Jajajaja, good one, seems I need to read more.

JonnyPhenomenon, 2018-03-26:
Thanks Mike!