betaflight/betaflight #5524: Mag sensor output is always positive (negative gets cut)

dgrat, 2018-03-23:
Using the configurator or blackbox log viewer reveals that the compass is not working. 
I use an external gy-654 module with HMC5983.
The values of the sensor are always positive on all axis. The heading calculation is thus broken. 
Is there a way to fix it?

[image]

# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.3.0 Mar  2 2018 / 03:56:35 (177472b4f) MSP API: 1.37

# name

# resources
resource MOTOR 1 B01
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 A03
resource MOTOR 6 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature LED_STRIP
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 0 68 115200 57600 0 115200
serial 5 32 115200 57600 0 115200

# led
led 0 4,0::C:0
led 1 5,0::C:0
led 2 6,0::C:0
led 3 7,0::C:0
led 4 8,0::C:0
led 5 9,0::C:0
led 6 10,0::C:0
led 7 11,0::C:0

# color

# mode_color

# aux
aux 0 1 0 1100 1175 0
aux 1 5 1 1450 2100 0
aux 2 3 0 1275 1350 0

# adjrange

# rxrange
rxrange 1 2000 1000

# vtx

# rxfail

# master
set align_gyro = CW0
set align_acc = CW0
set align_mag = CW0FLIP
set magzero_x = -15
set magzero_y = 113
set magzero_z = 13
set baro_bustype = I2C
set baro_i2c_device = 2
set baro_i2c_address = 119
set rssi_channel = 16
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1000
set vbat_scale = 107
set ibata_scale = 625
set pid_process_denom = 2
set osd_warnings = 63
set osd_vbat_pos = 2485
set osd_rssi_pos = 2082
set osd_current_pos = 2517
set osd_altitude_pos = 2102
set debug_mode = NOTCH
set vcd_video_system = 1

# profile
profile 0

set dterm_notch_hz = 280
set yaw_lowpass = 80
set p_pitch = 50
set i_pitch = 40
set d_pitch = 10
set i_roll = 30
set d_roll = 10
set p_yaw = 50

# rateprofile
rateprofile 0

set roll_rc_rate = 145
set pitch_rc_rate = 145
set yaw_rc_rate = 200
set roll_srate = 68
set pitch_srate = 68
set yaw_srate = 50

etracer65, 2018-03-24:
This has already been reported and fixed for the next maintenance release.  https://github.com/betaflight/betaflight/pull/5451

It was just a bug in how the values were sent from the flight controller to the Configurator for display.  The actual sensor is working correctly in the flight controller.

closed 2018-03-24 by etracer65

dgrat, 2018-03-24:
@etracer65  Just noticed, my accelerometer has the same problem :/

jflyper, 2018-03-24:
Aaaaand, it has been reported and fixed for the next maintenance release, too.

MSP_RAW_IMU to convert accADC from float to integer #5417