betaflight/betaflight #5526: Dterm Notch Filter is not working

spatzengr, 2018-03-24:
Dterm Notch on BF 3.4.0, build #702 is broke.

[image]

[BF-FKF at 90hz.BFL.txt](https://github.com/betaflight/betaflight/files/1843962/BF-FKF.at.90hz.BFL.txt)

spatzengr, 2018-03-24:
https://github.com/betaflight/betaflight/pull/5458

mikeller, 2018-03-24:
Could this be related to #5513? @fujin?

flint723, 2018-03-24:
To be clear #5458 may not be the origination of this.  It's just the only other commit I had logs with dterm notch on.

DieHertz, 2018-03-24:
This commit changes PG layout and in not sure we have increased the version number. 
Go into CLI and check the value of Dterm notch hz and cutoff

fujin, 2018-03-24:
@DieHertz I increased the version of PG layout in my subsequent PR

mikeller, 2018-03-24:
@fujin: That would have 'nixed the configuration for this parameter group, potentially leading to just this outcome.

fujin, 2018-03-24:
.. realizing I only increased the gyro sensors PG

flint723, 2018-03-24:
@diehertz GUI filter tab numbers, when saved, transfer correctly into CLI for me.  Just checked.

DieHertz, 2018-03-24:
Looking further then.
Please provide a diff with every issue request for our convenience

flint723, 2018-03-24:
Added diff to earlier post.

DieHertz, 2018-03-24:
```
set dterm_notch_hz = 550
set dterm_notch_cutoff = 500
```
Are you sure there are the right settings? @flint723

flint723, 2018-03-24:
For the log in that post yes.  I picked a point where there was dterm noise in hover.

PR #5529 (2018-03-24): Fixed notch Q calculation to be dimensionless as it should
Fixed #5526, spotted by @apoc (great job, mate!)

During #5287, `filterGetNotchQ` was rewritten in such a way that Q acquired a dimensionality of `1 /  [time]`, which is not correct.

The original formula was:
```
octaves = 2 * log2(center / cutoff)
Q = sqrt(2^octaves) / (2^octaves - 1)
```

Let's rewrite Q, simplifying it:
```
Q = sqrt(2^2log2(center / cutoff)) / (2^2log2(center / cutoff) - 1) // move 2* into logarithm and get rid of 2 to the power of logarithm
= sqrt((center / cutoff)^2) / ((center/cutoff)^2 - 1) // get rid of sqrt
= [... cut]
closed 2018-03-24 by DieHertz