betaflight/betaflight #5530: MPU6000 and MPU9250 accelerometer scale only initialized to 8g

etracer65, 2018-03-24:
As the result of another issue discussion, I noticed that for both the MPU6000 and MPU9250 the accelerometer is initialized using the 8g full-scale setting whereas both parts are capable of 16g full-scale.  All other gyros are using the 16g full-scale.

Any ideas why this might be the case or rationale not to correct?

MPU6000: `drivers/accgyro/accgyro_spi_mpu6000.c`
```
    // Accel +/- 8 G Full Scale
    spiBusWriteRegister(&gyro->bus, MPU_RA_ACCEL_CONFIG, INV_FSR_8G << 3);
```

MPU9250: `drivers/accgyro/accgyro_spi_mpu9250.c`
```
    mpu9250SpiWriteRegisterVerify(&gyro->bus, MPU_RA_ACCEL_CONFIG, INV_FSR_8G << 3);
```

etracer65, 2018-03-24:
Ok, what I've found so far.

**MPU6000**
In BF3.1.7 the ACC scale was still set to 16g.  In BF3.2.0 it was changed to 8g.  Can't find the actual commit because the file history seems to cutoff before I can see far enough back.  Shortly after the change, @sambas submitted https://github.com/betaflight/betaflight/commit/3d0b25b62d710959b5b1d020ec2340e4676e3886#diff-e91165f8bfd1ca75980940a4c64c1dff to fix the scale to match the new 8g setting that was apparently missed when changing the register setting.  Not implying that @sambas made the other change, but maybe you remember something?

**MPU9250**
Previous to BF3.0 there was no separate MPU9250 configuration and it was rolled into MPU6500 (they're the same except 9250 has a MAG).  So previously it was running with a 16g setting.  In BF3.0 a separate config was added and this was initially created to set the 8g scale.

So to summarize:

MPU6000: 16g -> BF3.2 -> 8g
MPU9250: 16g -> BF3.0 -> 8g

So if anybody has any additional info to understand the decision to change to 8g scale it would help.  It seems like we should be using the full 16g scale otherwise we have a greater risk of clipping high accelerations which might be contributing to larger attitude estimation errors.

mikeller, 2018-03-25:
Looks like the MPU6000 change was accidental during refactoring:

https://github.com/betaflight/betaflight/commit/60c2b812d1113818948e9983bd5fb10053970f16#diff-1efa9c0745638b010044a2909c5d73d6R235

etracer65, 2018-03-25:
Yep, that looks like it.  Your github search-fu is better than mine! :)

The MPU9250 is a little different in that it was split out from the MPU6500 as part of BF3.0 and it just looks like it was set up as 8g from that point forward.

I'll wait a bit to see if there's any objection, but it seems like these should just be fixed (along with updating the scale calculations to match).

mikeller, 2018-03-25:
I know a bunch of accelerometers were updated to use 16 G in the past, and all of them seem to be working fine.

ctzsnooze, 2018-03-29:
It was around that time the tendency for severe yaw spins started.

PR #5559 (2018-03-30): Correct accelerometer initialization for MPU6000 and MPU9250
Fixes #5530 

Accelerometer should be initialized to 16G scale.

Previously both were set to 16G matching all other MPU's but they were mistakenly changed to 8G during other changes.

MPU6000 was correct before BF3.2.  Was erroneously changed by: https://github.com/betaflight/betaflight/commit/60c2b812d1113818948e9983bd5fb10053970f16#diff-1efa9c0745638b010044a2909c5d73d6R235

MPU9250 was correct before BF3.0.  As erroneously changed when MPU9250 was split out into a separate file from MPU6500.
closed 2018-03-30 by mikeller