betaflight/betaflight #5532: Betaflight 3.3.0 causes periodic video breakup if IRC Tramp protocol is used and 600mw setting is used on Matek FCHUB-VTX

rox-wolf, 2018-03-24:
After upgrading from Betaflight 3.2.5 to 3.3.0 IRC Tramp protocol causes Matek FCHUB-VTX to glitch if maximum power (600mw) setting is set from Taranis lua script. This was not happening on Betaflight 3.2. Video glitches each ~12 seconds and LED on the VTX board is constantly blinking. Normally it should blink a bit after power change and then stop blinking. Only max power causes this, using lower power value makes it work OK. Here's video demonstration:

[image]

I'm using Matek F405-OSD and Matek FCHUB-VTX stack. Matek's VTX max power is 500mw, but on Taranis lua script it's coded as 600mw. 

I'm using 2.2.1 OpenTX with Latest LUA script.

Here's the diff

```
# diff

# version
# Betaflight / MATEKF405 (MKF4) 3.3.0 Mar  2 2018 / 03:52:04 (177472b4f) MSP API: 1.37

# name
name Stretchy

# resources

# mixer

# servo

# servo mix

# feature
feature -SOFTSERIAL
feature MOTOR_STOP
feature LED_STRIP
feature ESC_SENSOR
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map
map TAER1234

# serial
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200

# led
led 0 5,15::A:2
led 1 6,15::A:0
led 2 7,15::A:0

# color

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 1950 2075 0
aux 1 1 1 925 1050 0
aux 2 1 1 1950 2075 0
aux 3 13 2 1250 1350 0
aux 4 28 1 1450 2050 0
aux 5 35 1 1250 1325 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 12
set serialrx_provider = CRSF
set dshot_idle_value = 280
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 15
set beeper_dshot_beacon_tone = 1
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 1
set pid_process_denom = 1
set osd_warnings = 63
set osd_rssi_alarm = 65
set osd_cap_alarm = 1500
set osd_vbat_pos = 2504
set osd_rssi_pos = 2500
set osd_tim_2_pos = 2103
set osd_current_pos = 2515
set osd_mah_drawn_pos = 2510
set osd_warnings_pos = 2089
set osd_pit_ang_pos = 2081
set osd_esc_tmp_pos = 2135
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 3
set vtx_channel = 4
set vtx_power = 5
set vtx_freq = 5645
set vcd_video_system = 1

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set anti_gravity_gain = 10000
set p_pitch = 65
set i_pitch = 60
set d_pitch = 40
set p_roll = 65
set i_roll = 73
set d_roll = 40
set i_yaw = 60

# rateprofile
rateprofile 0

set roll_rc_rate = 87
set pitch_rc_rate = 87
set yaw_rc_rate = 87
set roll_expo = 18
set pitch_expo = 18
set yaw_expo = 18
set roll_srate = 83
set pitch_srate = 83
set yaw_srate = 83

# 
```

mikeller, 2018-03-25:
Have you read and followed the instructions on #5348?

MATEKSYS, 2018-03-25:
We already fixed this in new firmware of VTX-HV.  we will release new firmware for FCHUB-VTX

rox-wolf, 2018-03-25:
@mikeller Yes I'm using latest v1.0.1 Lua scripts from this site. 

@MATEKSYS Is this hardware related issue or a regression of software as on Betaflight 3.2 it was working properly. Is it possible to updare FCHUB-VTX firmware?

MATEKSYS, 2018-03-26:
@rox-wolf  http://www.mateksys.com/?portfolio=fchub-vtx#tab-id-5

rox-wolf, 2018-03-26:
@MATEKSYS Thanks, you actually fixed it! :) Upgraded the firmware using FTDI adapter and now it's stable, no video glitches! :)

closed 2018-03-26 by rox-wolf