betaflight/betaflight #5538: Motor wobble after first init - PID or dynmic filter initialization bad?

RipperDrone, 2018-03-25:
Have done >25 builds over the past 2yrs, and there seems to be a recurring pattern after fresh initialization of FCs which leads to weird woblles instead of steady signal output to the motors when trying to hover, but only at the very first test hover. Here is how I keep running into this issue approx. every 5 builds:

1. Flash FC (no complete erase sometimes?)
2. Apply settings in BFC, including Dynamic Filter, PT1, leave all notches and filter cutoffs on default
3. Load switch parameters, OSD parameters from other file into CLI
4. Calibrate gyro

Then, when trying to hover or even without props on, quad spazzes out. When reflashing and/or turning dynamic filters off, everything is smooth and nice.

Start wondering whether there's an init issue that the dynamic filters (or PID controller?) Uses readings from FC memory to tune in on filters, so if memory has not been completely erased, it will build filter settings upon rubbish values? 

Since it might be dangerous for some of us pilots, I thought I'd bring this issue up here...

etracer65, 2018-03-26:
There is no filter or PID performance or feedback data saved back to the eeprom.  The filters/PID are reinitialized at every power cycle by the values stored in eeprom based on your settings and no previous operational data is retained/used.

The more likely scenario here is that you may be initializing the settings to some odd or incorrect values based on the dump you're reapplying possibly related to not doing a full chip erase.  Then after the "problem flight" you go into BFC and change settings possibly correcting the rogue values implicitly.

Personally I've never seen any issue even remotely like this - but then I almost always to a full erase and maintain a minimalist diff that I use to reapply my settings when upgrading versions.  Even then you need to look at the cli responses to ensure everything applied correctly as parameter names/ranges can change between releases.

Maybe try reviewing a `diff` immediately after having this occur to analyze the config before making changes in BFC.

RipperDrone, 2018-03-26:
Hmm, my diffs are pretty slim and straightfwd... but I agree it has to do with initializing the parameters. Not aware that I adjusted any of them in BF other than the filter settings. Mostly switching off Dynamic_Filter or selecting another basic filter type biquad --> PT1 or vice versa fixed the spasms...

Jahnkeanater, 2018-03-27:
Sounds like a setting with a bad value. After every update its a good idea to run the 'defaults' command before restoring your diff. I have had quads spaz out even after a full chip erase.

Its also a good idea to paste your diff into the CLI a few lines at a  time to verify that all the settings make sense and to make sure you don't overwhelm the FC. It might not be able to keep up.