betaflight/betaflight #5539: CPU_OVERCLOCK command influence on the barometer data

jinukyang, 2018-03-25:
When I use 32k, I always use cpu_overclock command.
But After I changed FC W/ baro(Matek F405-std), the barometer data continuously rise.

1. without Cpu_overclock command
[image]
2. With Cpu_overclock command
[image]

I didn't connect any other product. And I overwriting the FC.
firmware v. 3.3.o stable MKF4

DieHertz, 2018-03-25:
Looks like it's integrating some error

Sm1thson, 2018-03-26:
Could be due to temperature drift, over clock = board gets warmer? just a thought.

jflyper, 2018-03-26:
I suspect SPI clock getting faster beyond some limit when overclocking.