betaflight/betaflight #5540: Feature Request: Toss Start

FPVZaphod, 2018-03-25:
Hi all,

I am in love with the auto launch feature of iNav - would it be possible to add a "Toss Start" Lauch Mode in BF? 
Could be an additional Flight Mode. Activate while the quad is disarmed. Then Arm it, and raise throttle to a value which ensures hovering of climbing. Beeper beeps to show the "ready to throw" Mode. Then just throw it up in the sky. When the successful launch is detected (strong acceleration and then zero acc. ), the quad should level out, and climb with throttle value. When touching the right stick, it swiches back to the normal flight mode (whatever you choosed, Acro or Angle), and behaves normally. 
I do sometimes toss starts, looks very cool, but I am always scared to not find the arm switch right on time :-)
Thanks a lot!

JyeSmith, 2018-03-25:
This sounds neat but beyond the obvious safety issues, I can see people doing a mid air taz.

FPVZaphod, 2018-03-26:
iNav disables the Launchmode, once it's triggered. It's only available again after disarm, so unrequested mid air acrobatics should be impossibe. Anfs the safety-issue, I think it's quite relaxed? With a proper throw, the delay between motor start and tross detection can be easily 2 seconds.  A beeper for status indication is mandatory, of course.

mikeller, 2018-03-26:
Does iNav have a launch mode for multirotors, or is this only for fixed wing (where hand launching is sometimes the only viable option)?

FPVZaphod, 2018-03-26:
iNav was more an example, because it's already F3/4 based, I am not sure if the Nav Launch is available for multirotors. 
Where it is available and working (I tested it already) is ardupilot:
http://ardupilot.org/copter/docs/throw-mode.html

mikeller, 2018-03-26:
Ok, this seems to be a different use case: The multirotor goes into loiter mode at a couple of meters altitude, out of harm's way, right after the launch. The pilot does not even hold the transmitter when launching the craft. This makes some sense. Betaflight does not have loiter mode, nor does it have reliable altitude hold, so I can't see how this can be implemented in a way that is not a safety risk.

FPVZaphod, 2018-03-26:
Loiter Mode you can replace with Angle/Horizon plus Loiter Throttle. Of course it's not ok to throw it, walk to your chair, put the goggles... Currently I have a switch for arm, and a 2nd momentary switch for throttle zero plus disarm. I press this switch, move the arming switch ot "arm" (remains disarmed because the momentary switch), then I raise throttle to half, throw the copter, and release the momentary switch. This arms, and instantly raises the throttle. 
But yes, I agree - such a throw mode has some potential for unwanted results...

etracer65, 2018-03-26:
Also take into account that any "toss start" type implementation won't play well with runaway takeoff prevention (anti-taz) which likely trigger a disarm.  The problem is that to the logic it looks nearly identical to a runaway event (high pidSum and gyro motion).

Jahnkeanater, 2018-03-27:
I have done this a few times. There has been instances where the quad can not find level and quickly accelerates in a random direction. Sometimes straight down.

varkokonyi, 2018-03-27:
Maybe, if combined with an ACC, it is possible to set a throttle amount that the software will hold until the throttle command is changed (and cut off if it's not changed after 2 seconds as a safety feature), and it would also activate angle mode. This way, it would be possible to make the quad hover or slowly ascend in one spot, quickly put on the goggles and go.

FPVZaphod, 2018-03-28:
Yes, that's what I was thinking about. With throw mode activated, 1st action should be level. Auto-Arm and hover throttle after the throw is detected, then automatic switch to the desired flight mode as soon as you touch the right stick. 2 seconds timeout till throttle cut is a great idea. 
Plus the beeper ticking like a little bomb to indicate that it's ready to throw.

mikeller, 2018-03-28:
With a flight controller without GPS, or any other means to establish and lock onto a set position, I am left wondering how many tries, or how much of an unexpected gust of wind it will take until the craft comes straight back and blends your face before your fingers are even touching the sticks...

FPVZaphod, 2018-03-28:
I think a race copter will never replace human thinking. The common sense should make clear to throw it at least 3-5 meters up, and not against the wind. And of course not between trees or houses...
Should not raise the usual  shrinkage and spoilage a copter brings anyways :-D

mikeller, 2018-03-28:
How long does your quad take to accelerate from standstill, and cover 3 - 5m when you punch it out?

FPVZaphod, 2018-03-28:
I guess it takes around same time like throwing it. From hovering at 3 meters to reach 8 meters should take 1-2 seconds at full throttle.

Rakinroul, 2018-03-30:
I dont know why yet, but there are issues to be sure your quad will arm with properly accelerometer setup.

I am repluging almost every second start, because my quad wont arm because out of angle, even if placement is ok....

I dont belive my quads will arm everytime, when i place them on surface, for some reason.
I am not using small angle 180 mostly.

FPVZaphod, 2018-04-03:
I use small angle = 180 by default, to be able to to the "manual throw start" - works absolutely reliable. Maybe you move it too much while connecting the battery, so the gyro calibration fails?