betaflight/betaflight #5543: PIKO BLX F.Port Rebooting randomly

createthis, 2018-03-25:
I have a PIKO BLX that used to be running Betaflight 3.1.7 reliably with an XM+ receiver using SBUS.

I bought a R-XSR to replace the XM+ so I could have VFAS battery telemetry in the same form factor. I flashed it with the latest firmware. I only noticed after the fact that the latest firmware from FrSky is FPort firmware. How sneaky of them.

Anyway, I couldn't get SBUS working with this FPort receiver firmware and I had already desoldered the R-XSR header because this thing has to go in a super tight space on my 3" micro. Flashing older firmware would have been a pain. Instead, I decided to upgrade to Betaflight 3.3 and give FPort a try. I noticed there wasn't a stable release for my PIKO BLX, but I saw a recent release on the Releases page here on github, so I flashed betaflight_3.3.0_FF_PIKOBLX.hex and restored my settings from a dump and then set up FPort according to the instructions.

I now have stick input in the receiver tab and my telemetry works fine. CPU usage is in the 20s and everything functions.

Unfortunately, my PIKO BLX appears to be rebooting randomly. It seems more likely to reboot when I change tabs in the Configurator, but it doesn't always reboot changing tabs. I can fly it for a few seconds, but then it falls out of the sky and reboots.

Any ideas?

MJ666, 2018-03-25:
I fly an FuriousFPV PikoBLX and an FlyingEgg with an PikoBLX clone with 3.3.0 and don't have issues but i use and LemonRX DSMX Sat and an XM RX.

RoadyFPV, 2018-03-25:
that is a normal behavior as far as I know.
all my fc's do this so.

mikeller, 2018-03-25:
@createthis: That is the expected behaviour. Exiting the CLI triggers a reboot of the flight controller.

closed 2018-03-25 by mikeller

MJ666, 2018-03-25:
Initalization message I responded to contained much more an was not about cli reboot only?

createthis, 2018-03-25:
Yeah, I pruned it down when I realized removing the load from the BEC on the Piko BLX was fixing the random reboot issue. Sorry for the noise.