betaflight/betaflight #5551: RSSI value in receiver tab, OSD, setup tab and Taranis not consistent

BlueFishUK, 2018-03-26:
The RSSI value in the setup tab and OSD shows 9% but receiver tab AUX 6 shows 1940 and Taranis shows 92%.

I have setup the RSSI using the Taranis loop back method and double checked all settings. 

Matek F405 CTR Flight controller:

My CLI RSSI settings:

# Betaflight / MATEKF405 (MKF4) 3.3.0 Mar  2 2018 / 03:52:04 (177472b4f) MSP API: 1.37

# get rssi
rssi_channel = 10
Allowed range: 0 - 18

rssi_scale = 41
Allowed range: 1 - 255

rssi_invert = OFF
Allowed values: OFF, ON

osd_rssi_alarm = 40
Allowed range: 0 - 100

osd_rssi_pos = 2058
Allowed range: 0 - 3071

osd_stat_min_rssi = ON
Allowed values: OFF, ON

[image]
[image]

etracer65, 2018-03-26:
Try changing `set rssi_invert = ON`

BlueFishUK, 2018-03-27:
@etracer65 Thank you so much that worked perfectly. I guess I need to understand more clearly the subject of inversion as I see it mentioned quite a bit when referring to UART's and serial ports etc. 

Do you have by any chance a link to information whereby I might be able to read about this subject and gain a fundamental understanding of the principles, as I do not wish to waste the time of you guys who I am sure have more important issues to deal with than my ignorance?

Best wishes.

McGiverGim, 2018-03-27:
This "inversion" has nothing to do with UART. 

Your RSSI channel show 1940, the channel go from 1000 to 2000 so 1940 is a 94%. By default the RSSI level used is the difference between 100% and your channel, 6% at this sample. When you activate the inversion, is just the opposite, 94% is used.

BlueFishUK, 2018-03-27:
@McGiverGim thank you that really helps my understanding, I didn't realise this as my other quad with an Omnibus F4 does not have RSSI inversion set to on. I will check carefully the difference between the two setups and try and understand why this is the case.

In regard to my question regarding inversion in general I was trying to get my head around this information with regard to understanding inverted signal receivers. I realise it's not linked to my question about RSSI inversion.

"SBUS/PPM (J7) header (uart6) This is for inverted signal receivers; has fixed inverter that cannot be turned off (by CLI command sbus_inverter=on/off)
RX1 (on J2) (uart1), These are for non-inverted signal receivers."

McGiverGim, 2018-03-27:
Regarding to UART, simplifying, some protocols (like SBUS) uses a HIGH voltage signal for 0 and LOW signal for 1 (the habitual is the inverse). So this "electric" signal must be inverted to be in the form that the UART needs. So you need an inverter to make it work.

This inverter can be switchable with a parameter or hardcoded in the FC to some pads.

BlueFishUK, 2018-03-27:
Thanks @McGiverGim that gives me a good start I have found some useful links online and will get to some studying this week.

closed 2018-03-27 by etracer65

luireyes, 2018-11-14:
Hi, Could you please advise why my RSSI value in my OSD is flashing and only 1 digit?

I have Matek F405 Wing / Inav 2.0.1 / R9MM receiver

etracer65, 2018-11-14:
@luireyes This is the Betaflight project, not iNav. Maybe try here instead:

https://github.com/iNavFlight/inav

mikeller, 2018-12-23:
@rolial: This is the same question that you've already asked in #7275. Please refrain from posting the same question in multiple places - have a look at other questions, and then select the option that matches best with your question.

rolial, 2018-12-23:
> @rolial: This is the same question that you've already asked in #7275. Please refrain from posting the same question in multiple places - have a look at other questions, and then select the option that matches best with your question.

OK sorry.
I erased