betaflight/betaflight #5553: F.Port (halfduplex) and RunCam Device (duplex)

N4v1g4t0R, 2018-03-27:
Hardware Setup
BetaflightF7 with Betaflight 3.3.0 (Mar  2 2018)
FrSky R-XSR with F.Port Firmware 180228
RunCamSplit 2 with Fimware 2.0.0.

Software Setup
RX is on UART3 with the uninverted Signal (formal S.Port) from the R-XSR
RunCamSplit2 ist on UART2 RX/TX-TX/RX

Parameters
```feature TELEMETRY
serial 1 16384 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serialrx_provider = FPORT
serialrx_inverted = OFF
serialrx_halfduplex = ON
```

Problem
F.Port works fine with this setup, control as well as telemetry. But the RunCam does not react on commands.

My quess is, that the serialrx commands are global commands, affecting all UARTS, not just the UART for the receiver, UART3 in this case. So, the config works for F.Port (halfduplex), but not for the RunCam (duplex).

mikeller, 2018-03-27:
> My quess is, that the serialrx commands are global commands, affecting all UARTS, not just the UART for the receiver, UART3 in this case. So, the config works for F.Port (halfduplex), but not for the RunCam (duplex).

No, this is not the case. The `serialrx_` parameters are only applied to the port that is configured for serialrx (UART3 in your case).

etracer65, 2018-03-27:
I think the problem is according to the target files the TX pin for UART2 is not connected.  Not sure if this is really the case on the BFF7 (target FPVM_BETAFLIGHTF7) or just something inherited from the OMNIBUSF7 which it shares the target.

`/target/OMNIBUSF7/target.h`:
```
#define UART2_TX_PIN            NONE
#define UART2_RX_PIN            PA3
```

If the TX pin is truly not connected then the only option would be to use another UART.

@rotcehdnih Do you know if it is correct that UART2 TX pin is not connected or is this something that only applies to the OMNIBUSF7 and should be fixed for FPVM_BETAFLIGHTF7?

N4v1g4t0R, 2018-03-27:
The assumption with the UART2 TX makes sense.

F.Port works at TX3 for control and telemetry, but the RunCam control was on UART2, and not working. F.Port on TX2 did not work, so I could not test the RunCam control at UART3.

etracer65, 2018-03-27:
None of the pictures on FPVModel's website show it clearly, but from the manual:

[image]

So it looks like it's connected to the MCU but since it's labeled S.PORT it _might_ have a bidirectional inverter on it (but probably not).  Might just need a target update specific to FPVM_BETAFLIGHTF7.

N4v1g4t0R, 2018-03-27:
thanks @etracer65

DieHertz, 2018-03-27:
>since it's labeled S.PORT it might have a bidirectional inverter on it (but probably not). 
@etracer65 why would it? It's an F7, not F4

etracer65, 2018-03-27:
@DieHertz Yeah, I thought of that after I wrote it and just didn't bother to edit.

As I mentioned in the slack channel, it looks to be a simple matter to add a conditional define for the FPVM_BETAFLIGHTF7 to assign UART2 TX pin to PA2, but the timer for PA2 is being used for ESCSERIAL.

```
#define USE_ESCSERIAL
#define ESCSERIAL_TIMER_TX_PIN  PA2 // (Unwired UART2_TX)
```

So I'm assuming we'd also have to use a different timer for ESCSERIAL.

mikeller, 2018-03-27:
I contacted @svfalcon for a schematic for the board, so we can put this one to bed once and for all.

rotcehdnih, 2018-03-28:
I have them , i'll jump on and share them now . i did post them but removed the link when we had someone screenshoting the members group.

TX/RX2 are on the board and connected

mikeller, 2018-03-28:
@N4v1g4t0R: Ok, I think we've solved this. #5562 permanently adds UART2 TX. In the mean time, just doing `resource serial_tx 2 a2` (and `save`) in CLI should change this for the existing firmware. Please test.

@DieHertz: I don't think this conflict will matter. `ESCSERIAL_TIMER_TX_PIN` is only used when the ESC serial forwarding mode is invoked, at which point the flight controller ceases to function as a flight controller, and stops sending serial output on UARTs.

DieHertz, 2018-03-28:
@mikeller that probably was aimed at someone else, can't recall discussing `ESCSERIAL_TIMER_TX_PIN`

mikeller, 2018-03-28:
Hehe, yes, that was @etracer65, sorry @DieHertz.

N4V1G4T0R, 2018-03-28:
Thanks

I tested with the inverted F.Port Signal from the R-XSR on the TX2 after the CLI commands. Firmware as stated on top of the discussion. No luck with the settings that should work. I tried all permutations of serialrx_inverted and serialrx_halfduplex = ON/OFF. Still no luck. Neither control nor telemetry. Same settings on TX3 work. Maybe it's just a faulty FC.

I'm probably not able to test the next days, because I'm going on vacation.

mikeller, 2018-03-28:
Weird. I haven't tested the hardware, will do that later on to confirm the schematic is correct.

mikeller, 2018-03-31:
Tested with vanilla (not uninverted) XSR on TX2 on a FPVM_BETAFLIGHTF7. Confirmed working as expected.

Configuration used:
```
# diff

# version
# Betaflight / FPVM_BETAFLIGHTF7 (FBF7) 3.4.0 Mar 31 2018 / 16:38:54 (1a29e88f5) MSP API: 1.39

# serial
serial 1 64 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# master
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
```

closed 2018-03-31 by mikeller

N4V1G4T0R, 2018-03-31:
thumbs up :-)

yoodog, 2019-10-05:
F7 board UART2 is RC pin = RX only. Just use any other UART1,3,6 (UART 7 is ESC Telemetry usually taken)
I have it so on Omnibus F7v2
UART1 TX/RX   - RunCam split 2s
UART2 RX only - GPS
UART3 TX         - F.Port for r9mm-ota
UART6 TX        - SmartAudio2.1 for Unify pro32 nano

jflyper, 2019-10-05:
@yoodog You have commented on the 1.5 year old closed issue with out of context example.