betaflight/betaflight #5558: Please create a cli command to turn on the disarm switch that was removed

WiKDMoNKY, 2018-03-27:
BF 3.3.0 disabled the ability to disarm in the air, which in my opinion, is dangerous for all of us that have been landing using this setting since CleanFlight. Why can't there be a cli command that turns this back on?

mikeller, 2018-03-27:
Please see #5073 for a discussion of the reasons why this unsafe option was removed.

I can't see how using a configuration where the arming switch does **not** stop the motors would be **less** dangerous in any situation.

closed 2018-03-27 by mikeller

AndersHoglund, 2018-03-28:
@etracer65 had a much safer variant that does the job the way you want it:

> An alternate way to achieve the same result without the safety issues is to:
> 
>    - Enable motor_stop
>    - Disable feature airmode (so it can be set up on a switch)
>    - Configure your arming on a 3-pos switch so that you have disarmed -> armed -> armed + airmode
> 
> Now when you come in to land, switch to the middle setting to disable airmode. This will then enable motor_stop. Simply land and lower the throttle to zero and the motors stop. Then fully disarm. Same end result with fewer safety issues.
> 
> If you don't want to use a 3-pos switch, simply set up a "landing mode" switch that does the airmode on/off.

Maybe we should add this to some "Tips&Tricks" section in the WiKi? Easier to link to, than trying to  finding old comments to copy&paste.

codecae, 2018-03-28:
Or you can do it with OpenTX... **YOU SHOULD NOT DO THIS.  DON'T BLAME ME IF YOU DO AND YOUR QUAD FLIES TO THE MOON OR ATTACKS AN INNOCENT PEDESTRIAN!**

[image]
[image]