betaflight/betaflight #5563: LED Strip output configuration open drain

RicoKirstein, 2018-03-28:
Since a lot of people, including me continue to have problems with too low signal voltage on different types of LED strips, we should allow to change the output type of the ledstrip pwm to open drain with no internal pullup. 
As most of the pins of the f4 and f7 are 5V tolerant, this allows to have an external pullup to 5V >1kOhm to have a real 5V pwm signal without the need to have the first led on a lower voltage.
I have tried it on a F722 and it's working perfectly.

I could do a pull request for configuration per define, but I think having a cli configuration for it would be the way to go since this allows to change it on existing flight controllers and add the pullup directly on the LED if necessary. And I haven't read into the code enough to add a configuration variable yet.

This is my current pin configuration for testing:
IOConfigGPIOAF(ws2811IO, IO_CONFIG(GPIO_MODE_AF_OD, GPIO_SPEED_FREQ_VERY_HIGH, GPIO_NOPULL), timerHardware->alternateFunction);

mikeller, 2018-03-28:
Sounds like a good suggestion. How about you have a go at adding the change, with a config parameter (they are not that hard, just take an existing one like `ledstrip_visual_beeper` as a template, and grep for all  occurrences), and we'll help you along the way?

RicoKirstein, 2018-03-28:
sounds reasonable, I'll look into it tomorrow evening or on the weekend if I find some time.
I got some more changes going for a custom fc. Is there any board function for pin setups on boot? currently i'm using the hardware version detect function but it seems like an ugly hack.

mikeller, 2018-03-28:
I think `detectHardwareRevision()` is the best place for this kind of thing for the moment - unless it's something of general interest that can be made into an optional driver.

RicoKirstein, 2018-03-28:
It might be, I'm usind the PLL to generate the 27MHz for the OSD chip to save board space.

mikeller, 2018-03-28:
That does sound useful. :-)

jflyper, 2018-04-03:
@RicoKirstein Is this project still active? If not, I can take over and finish it.

RicoKirstein, 2018-04-03:
@jflyper I haven't had any time to work on it yet. And I probably won't be able to work on it before next week. If you want, go for it.

piodabro, 2018-04-04:
I've got one question.
Is LED_STRIP pin 5V tolerant on most configurations?
I'm making some hardware depending on LED_STRIP signal, and there is possibility that it will drive LED_STRIP pin on 4.5V level if there is no signal from FC...

RicoKirstein, 2018-04-04:
@piodabro I believe in most configurations it should be, on the F722 there are only 2 GPIO pins I think that aren't 5v tolerant. And even those only aren't if you use them as adc or dac pins. How are you driving those pins to 4.5V is the bigger question. 
Is it a reasonably high pullup >1k or some mosfet directly connecting it to 4.5V?
As long as the current is limited it should work even with the default pin config.
But if you want to drive it from the STM32 and need the 4.5V highs you need to switch it to open drain and disable the internal pullup.
Then again, if you're directly connecting it to 4.5V you should never drive the stm32 pin low since it would short to ground and kill the pin.

piodabro, 2018-04-04:
@RicoKirstein i have MCU on the other side wchich generates WS2812 signal. It has push-pull output on 4.5V logic level. I figured out how to make this working ;) just used resistor divider wchich makes 3.38V from 4.5V (3.3V tolerant pins has VDD + 0.3V typical voltage and absolute maximum is 4V).
There is only one question left. 
What is pin configuration if there is no LED configuration in FC. (No signal generated by BF).
It would be really awesome if it is Hi-Z ;)

RicoKirstein, 2018-04-04:
@piodabro are you programming the other mcu yourself? you should also switch it to opendrain with a pullup to your desired voltage if possible. Having two output devices on the same bus is never a good idea if one is in pushpull configuration. If you set the pin on the FC to input it will be highZ but if you have no way of syncing both outputs, the FC could drive the pin low while your mcu tries to drive it high resulting in a short. 
The voltage divider should give you some protection, if the values are high enough. It should be high enough so it limits the current to a few milliamps for your supply voltage. I don't know what the default init mode for GPIOs is in betaflight. Why would you connect it to the led pin if you expect it to have no configuration in BF?