betaflight/betaflight #5564: 100% cpu in pwm receiver configuration

menachg, 2018-03-28:
I'm running on sp-racing naze32 board on the latest version(3.3).
When tried arming noticed 100% cpu error
worked on debugging it a bit .
And when i switch to ppm/serial input cpu readings go down to normal(6%)
so i guess thats the culprit.
Works fine when i roll back to betaflight-2.6.1/cleanflight-2.2

mikeller, 2018-03-28:
I don't know what you are running, but NAZE is not supported in 3.3. any more. Can you do a `status` in CLI, and post the output?

menachg, 2018-03-29:
no its not an F1 board its of the sp-racing F3 type.
ii will take me some time to wire it up and take copies of cli again
but i can tell you right now:
In status it had "load" error in status.
And in tasks rx took more than 90%
it works fine with cleanflight and older versions of beta so i dont really mind
i just wanted to float this problem up 
if other  people have it too we should look into it
what changed in pwm rx decoding to possibly drive the cpu usage that high.

eephyne, 2018-03-29:
Hello
Same issue here

I just upgraded betaflight on a quad witha sp racing f3 (chinese).
it use an old pwm receiver who was working great (as far as it can be) before I upgraded.
Now I can't arm, after searching the beep code, it seem cpu usage is too high.
I look into cli, and I soon as I plug power I see this :
```
# tasks
00 - (         SYSTEM)      4       7       2    0.5%    0.5%         0
01 - (            PID)    932     204     165   19.5%   15.8%      4833
   - (           GYRO)   1865
02 - (            ACC)    471     122     112    6.2%    5.7%      1648
03 - (       ATTITUDE)     49      28      22    0.6%    0.6%        31
04 - (             RX)  10309     105      78  108.7%   80.9%      1657
05 - (         SERIAL)     16  100899       3  161.9%    0.5%       116
07 - (BATTERY_VOLTAGE)     24       9       4    0.5%    0.5%         2
08 - (BATTERY_CURRENT)     24       7       2    0.5%    0.5%         1
09 - ( BATTERY_ALERTS)      2       9       3    0.5%    0.5%         0
10 - (         BEEPER)     16      12       2    0.5%    0.5%         3
11 - (        COMPASS)      1     108     105    0.5%    0.5%        14
12 - (           BARO)     16     118     110    0.6%    0.6%       152
14 - (       ALTITUDE)      6      64      57    0.5%    0.5%        29
19 - (            CMS)      9       8       3    0.5%    0.5%         1
20 - (        VTXCTRL)      4       3       2    0.5%    0.5%         0
21 - (        CAMCTRL)      4       2       1    0.5%    0.5%         0
23 - (       PINIOBOX)     19       8       4    0.5%    0.5%         0
RX Check Function                  12       5                        91
Total (excluding SERIAL)                       141.1%  109.1%

```

I tested changing to ppm, and it go down to normal (but of course I don't receive any receiver data).
I precise that in pwm mode even if the cpu usage is full, I can read receiver data in configurator.

3.2.5 is okay but 3.3.0 RC and final have this bug

mikeller, 2018-03-31:
@eephyne: I just tested with an SPRACINGF3, and in 3.3.0 I get 

```
04 - (             RX)     49      94      80    0.9%    0.8%        72
```

which looks completely normal.

I don't have a PWM setup to test with, so I am testing without a valid input signal. Can you check if you get the high load even when there is no PWM input signal?

Also, please do a 'reset to defaults', and a restore from backup, and then try again.

eephyne, 2018-03-31:
All my tests was on clean install without any modification except input Rx type of course.

I'll try today to test it without putting the pwm wire

eephyne, 2018-03-31:
Yes. diff in cli show only pwm feature.

etracer65, 2018-03-31:
I put together a PWM setup and was able to replicate this problem off of current master.  When the PWM receiver is disconnected (or not putting out pulses) the RX task is normal.  However when there are PWM signals present the RX task consumes all the processing cycles.

I'll look into it.

DieHertz, 2018-03-31:
If you need a hand with a debugger/logic analyzer I can help.

dronematt, 2018-04-01:
Hi guys,
I'm new to this quad building & I have come across this issue with my first build. It's kinda been doing my head in, but now I've finally found people with the same issue!!!!

I'm using PWM on f3 controller using latest version of betaflight on the betaflight configurator. Everything works fine, transmitter, motors run in motors tab etc. The issue is when I turn on the transmitter & try to arm, CPU jumps from 8% to 100% & obviously won't arm, had the same issue on cleanflight as well.

Are there any remedies?
Thanks in advance

eephyne, 2018-04-01:
Same here, if you need a hand to get debug info, I'll be happy to help

mikeller, 2018-04-01:
Thanks guys. We've identified the problem, and are working on a solution.

dronematt, 2018-04-01:
Just letting you know, went back to earlier version of betaflight and I can now arm my quad!!!!!!!!!!!

eephyne, 2018-04-01:
Yes this is starting with 3.3.0, 3.2.5 is fine

E120103016, 2018-04-01:
The problem with dronematt. The only way to solve this problem is to convert the PWM to single-line PPM and use 3.3.0 firmware.

etracer65, 2018-04-01:
No need to make changes to your receiver configuration.  The fix is in testing and the problem will be solved soon.

mikeller, 2018-04-01:
@menachg, @eephyne, @dronematt, @E120103016: Can anybody test the fix in #5589 please? If you can't build the firmware yourself let me know what target you need and I'll build it for you.

eephyne, 2018-04-01:
I have the quad ready for it, I can test it tomorrow morning (GMT+2 here so in about 10 hours). But I can't build it right now, my controller Target is sp racing f3.

mikeller, 2018-04-01:
@eephyne: Here you go: [betaflight_3.4.0_SPRACINGF3.zip](https://github.com/betaflight/betaflight/files/1866649/betaflight_3.4.0_SPRACINGF3.zip)

dronematt, 2018-04-01:
Hi Michael,

Flashed & tested betaflight_3.4.0_SPRACINGF3.zip   on betaflight this morning. Seemed to work fine, no more CPU issues,

Cheers,

Matt

From: Michael Keller
Sent: Monday, 2 April 2018 5:50 AM
To: betaflight/betaflight
Cc: dronematt; Mention
Subject: Re: [betaflight/betaflight] 100% cpu in pwm receiver configuration(#5564)

@eephyne: Here you go: betaflight_3.4.0_SPRACINGF3.zip

mikeller, 2018-04-01:
Thanks for testing @dronematt!

closed 2018-04-01 by mikeller

mikeller, 2018-04-01:
Sorry to bother you again, @dronematt, but can you please also test this one: 
[betaflight_3.4.0_SPRACINGF3.zip](https://github.com/betaflight/betaflight/files/1866765/betaflight_3.4.0_SPRACINGF3.zip) (this is the change adapted to 3.3, which will go into the next maintenance release, but it wasn't straightforward to get it in there, so another test  would be great.

menachg, 2018-04-01:
i too am using spracingf3
i flashed both hex's for 3.4 you included
and both seem to work fine with pwm input
cpu usage nominal.
great work guys!

mikeller, 2018-04-02:
Thanks for the feedback, @menachg!

PR #5598 (2018-04-01): Fix PWM rx 100% CPU load
Fixes #5564 

Previous logic changes were causing the RX task to run every loop for PWM rx leading to 100% CPU load. Reverted PWM processing to be based on a time interval rather than whether new data is available. With PWM we can't differentiate when new data is available as it's a continuous stream.