betaflight/betaflight #5565: CRASH FLIP warning "stuck" in OSD - differing from actual state

bkeinert, 2018-03-28:
The CRASH FLIP warning is sometimes shown incorrectly while the actual state of the quad is not crash flip mode in recent master.
Reproduce: Switch on crash-flip, warning is shown, disable crash flip and the warning will still be shown until next arm where the warning disappears and you have a normally armed quad.

Also noticed another oddity where crash flip wasn't shown and the quad seemingly armed in crash flip mode or some other faulty state. Happened twice in two days (a lots of crash flip testing) and seems hard to reproduce.

etracer65, 2018-03-28:
I’ll take a look when I get home tonight.

etracer65, 2018-03-29:
Turns out this this was a bug with the crash flip state only being set/cleared on arming.  So the OSD was actually displaying the correct information.

Created https://github.com/betaflight/betaflight/pull/5566 to fix.

bkeinert, 2018-03-29:
Nice! Thank you. Maybe this also fixes the oddities - will try in a couple of hours.

PR #5566 (2018-03-30): Clear crash flip mode flag on disarming
Fixes #5565 

Previously the flag was only set or cleared during arming so after crash flip mode was used the flag would stay active until the next arming. This caused the "CRASH FLIP" warning message to erroneously display on the OSD when disarmed.

This problem was exposed by #5253 which changed the OSD logic to display the actual state of crash flip mode flag rather than simply track the switch.
closed 2018-03-30 by mikeller