betaflight/betaflight #5567: feature request: throttle limit in FC

theSkonk, 2018-03-29:
basically the ability to clip the top of the throttle with a configurator  or  cli command that is attached to the rate profile in use..

something like set max_throttle% = 80
allowed input   70-100

a lot of racers do this any with high kv setups to but do it via the radio..
i personally fly acro with my racing quads, so use different rate profiles.
would be nice to be able to have this attached to my race rates but not my acro rates ( sometimes you needs all the powers to get out of trouble) 

thoughts?

mikeller, 2018-03-29:
I think ideally this should be integrated into #5528, allowing users to set a throttle limit, and then a boost on the throttle when the battery voltage is droping. @etracer65, @ledvinap, what's your thoughts?

Rakinroul, 2018-03-29:
Why boost ? I dont thing there is reason for any boosting.

mikeller, 2018-03-29:
@Rakinroul: Because both of these features modify the mapping between throttle input and motor output. Even if the net result isn't a boost with a configuration designed to work with weak batteries, having a unified configuration that incorporates both is still better than having two independent configurations competing with each other.

Rakinroul, 2018-03-29:
I dont understand.

Whitch both features ?

You mean battery voltage compensation ?

I really dont think you can compensate throttle in relation with voltage (drop)

It is nonsense.

Ant it have risc whitch boris noticed.

mikeller, 2018-03-30:
Yes, battery voltage compensation. This modifies how the throttle behaves. You are suggesting the addition of another way to modify how the throttle behaves. The configuration and application of both of them should be combined, and not done independently, and possible in ways that conflict with each other.

Rakinroul, 2018-03-30:
I understand, i prefer have throttle (i call it) rescaling on FC.
When you want limit throttle maximum but wanna keep accuracy.
If set on radio you loose some.
Radio have 1k points ,and escs usually 2k.
I also prefer keep that 1k, when you limit that on FC / ESC side you still have at least something around 1.5k "points"

theSkonk, 2018-03-30:
I agree if you can keep the throttle Rez that's also a bonus to having it handled by the FC

etracer65, 2018-04-02:
Created #5608

PR #5608 (2018-04-05): Throttle limit feature
Fixes #5567 

Adds new `throttle_limit_type` and `throttle_limit_percent` parameters that allow the pilot to limit the maximum commanded throttle seen by the flight controller and determine how the limiting is applied.  `throttle_limit_type` can be `OFF | SCALE | CLIP`.  The `SCALE` value will scale the commanded throttle linearly to the limit specified by `throttle_limit_percent` based on the full stick range.  The `CLIP` type will leave stick scaling unchanged below the limit, and cap the throttle so that additional throttle input above the limit has no effect.

The se [... cut]
closed 2018-04-05 by mikeller