betaflight/betaflight #5576: spi rx twitch while flying in d16 mode

turboed13b, 2018-03-30:
Crazybee flight controller with spi-rx. D16 mode causes twitching while flying. Dshot600 mode causes failsafe while in d16 mode.

D8 mode works fine in multishot but in dshot600 it arms then immediately disarms with a telemetry lost from tx.

mikeller, 2018-03-30:
This sounds a lot like CPU overload. What loop frequencies are you using? Have you tried reducing them? A `diff` would help, just like it says in the text you get when opening an issue.

turboed13b, 2018-03-30:
I have tried 4k/4k. Previously on 3.3 then flashed 3.4 to see if it fixed it with no luck. The twitch only happens in d16 mode almost feels like a failsafe but quickly comes back. Dshot makes the twitch worse and occasionally fully failsafes and I have to unplug the quad to get it to fly again.

``` # diff
# Betaflight / CRAZYBEEF3FR (CBFR) 3.4.0 Mar 27 2018 / 19:22:44 (7ad4bbfa0) MSP API: 1.39
name BLACKHORSE
resource MOTOR 1 B09
resource MOTOR 2 B08
resource MOTOR 3 A02
resource MOTOR 4 A03

feature -TELEMETRY
feature DYNAMIC_FILTER
map TAER1234
aux 0 0 0 1700 2100 0
aux 1 1 1 900 1300 0
aux 2 28 1 1700 2100 0
aux 3 30 0 1700 2100 0
aux 4 35 2 1700 2100 0
set align_gyro = CW270FLIP
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set align_acc = CW270FLIP
set acc_trim_pitch = 4
set acc_calibration = -29,-75,114
set min_check = 1010
set rssi_channel = 8
set fpv_mix_degrees = 10
set rx_min_usec = 850
set rx_max_usec = 2150
set rx_spi_protocol = FRSKY_D
set motor_pwm_protocol = MULTISHOT
set bat_capacity = 250
set vbat_max_cell_voltage = 44
set vbat_min_cell_voltage = 29
set vbat_warning_cell_voltage = 32
set yaw_motors_reversed = ON
set pid_process_denom = 1
set osd_warnings = 63
set osd_cap_alarm = 250
set osd_vbat_pos = 2455
set osd_rssi_pos = 2433
set osd_craft_name_pos = 2442
set osd_warnings_pos = 329
set frsky_spi_tx_id = 32,14
set frsky_spi_offset = -26
set frsky_spi_bind_hop_data = 1,146,56,201,111,21,166,76,223,131,41,186,96,6,151,61,206,116,26,171,81,226,136,48,191,101,11,156,66,211,121,31,178,86,231,141,51,196,106,16,161,71,216,126,36,181,93,0,0,0
set frsky_x_rx_num = 1
profile 0

set dterm_lowpass_type = PT1
set dterm_setpoint_weight = 200
set p_pitch = 70
set i_pitch = 75
set p_roll = 65
set i_roll = 70
set p_yaw = 90
set i_yaw = 70
set p_level = 70
rateprofile 0

set roll_rc_rate = 114
set pitch_rc_rate = 114
set yaw_rc_rate = 108
set roll_srate = 71
set pitch_srate = 71
set yaw_srate = 60

# 
```

mikeller, 2018-03-30:
Try lowering your loop timings to 4k/2k to see if this improves things. Since the RX stack is run on the flight controller MCU, it will require some of the processing power, and it's sensitive to CPU overload. FrSky X (D16) is a lot more complex in terms of telemetry, which causes it to use more CPU cycles than FrSky D (D8).

turboed13b, 2018-03-31:
OK I tried 4k/2k and 2k/2k and it is not working still.

mikeller, 2018-03-31:
Thanks for testing. Sounds like a resource issue. Can you please post the output of a `resource list` and a `dma list`?

turboed13b, 2018-03-31:
```# resource
resource BEEPER 1 C15
resource MOTOR 1 B09
resource MOTOR 2 B08
resource MOTOR 3 A02
resource MOTOR 4 A03
resource SERIAL_TX 3 B10
resource SERIAL_RX 3 B11
resource LED 1 B03
resource RX_BIND_PLUG 1 A09
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource ADC_BATT 1 A00
resource ADC_CURR 1 A01

# dma

DMA:
DMA1 Channel 1: ADC
DMA1 Channel 2: FREE
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: FREE
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE
```

turboed13b, 2018-03-31:
```# resource list
IO
A00: ADC_BATT
A01: ADC_CURR
A02: MOTOR 3
A03: MOTOR 4
A04: MPU_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI
A09: RX_BIND
A10: LED
A11: FREE
A12: FREE
A13: FREE
A14: FREE
A15: FREE
B00: FREE
B01: OSD_CS
B02: FREE
B03: LED 1
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: MOTOR 2
B09: MOTOR 1
B10: FREE
B11: FREE
B12: RX_SPI_CS 2
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI
C14: FREE
C15: BEEPER
F00: FREE
F01: FREE
F04: FREE

# dma list

DMA:
DMA1 Channel 1: ADC
DMA1 Channel 2: FREE
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: FREE
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE
```

mikeller, 2018-03-31:
That looks like it was done when Dshot was not enabled (because it's not using any DMA for Dshot). Can you do it again please, with Dshot enabled?

turboed13b, 2018-03-31:
```# resource list
IO
A00: ADC_BATT
A01: ADC_CURR
A02: MOTOR 3
A03: MOTOR 4
A04: MPU_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI
A09: RX_BIND
A10: LED
A11: FREE
A12: FREE
A13: FREE
A14: FREE
A15: FREE
B00: FREE
B01: OSD_CS
B02: FREE
B03: LED 1
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: MOTOR 2
B09: MOTOR 1
B10: FREE
B11: FREE
B12: RX_SPI_CS 2
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI
C14: FREE
C15: BEEPER
F00: FREE
F01: FREE
F04: FREE

# Dma list

DMA:
DMA1 Channel 1: ADC
DMA1 Channel 2: FREE
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: MOTOR 3
DMA1 Channel 6: FREE
DMA1 Channel 7: MOTOR 4
DMA2 Channel 1: MOTOR 1
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: MOTOR 2
```

mikeller, 2018-03-31:
Ok, I suspect there is a resource conflict between the DMA that is used for the SPI RX and the DMA used for motor 3.
I have reshuffled the DMA for Dshot and squeezed into 2 DMAR channels: #5590.

@turboed13b: Can you please try this: 
[betaflight_3.4.0_CRAZYBEEF3FR.zip](https://github.com/betaflight/betaflight/files/1865167/betaflight_3.4.0_CRAZYBEEF3FR.zip) (requires a full chip erase / restore for all settings to take)

turboed13b, 2018-03-31:
That fixed dshot in d8 mode it flies now. 

Second problem with dshot in d16 mode and failsafe/twitches are still present. When it failsafes I can't rearm.

I did remap my motors so 1/2 are switched and 3/4 switched. I don't know if that was what is causing the problem with d8 mode.

turboed13b, 2018-03-31:
I tested multishot in d16 and the twitches are still there also.

mikeller, 2018-03-31:
Ha. That means that the firmware on the on board ESCs is too old, and can not handle the 'normal' / 'reversed' commands that are sent on arming when crash flip is enabled. Try updating the ESC firmware to 16.67 or 16.7 first, then try again.

(Also, if the hex file I sent you works this will be an improvement anyway, as it optimises the resource usage.)

turboed13b, 2018-03-31:
I am on 16.7

mikeller, 2018-03-31:
Ok. This is weird then. There is no difference between crash flip being disabled and enabled, except for the commands that are sent when arming, and 16.7 can handle these.

Have you tried Dshot300? (Just taking pot shots here, but there are some ESCs with slow MCU that can't handle Dshot600.)

turboed13b, 2018-03-31:
Just tried dshot300 and it didn't work. Messed with lower loop times also. This is in d8 mode.

It does work in d16 mode and flip over crash works also but it has the twitch and failsafe issue. I think it has something to do with the rx. I get telemetry lost and telemetry recovered as soon as i hit the arm switch.

mikeller, 2018-03-31:
I am a bit puzzled here. Both FrSky D and FrSky X were tested on another F3 board design, with Dshot, and confirmed to work fine there. This looks like this is an issue inherent to this board's design. Maybe @githubDLG can provide us with a known good configuration for this board?

turboed13b, 2018-03-31:
I tried pretty much every setting. I ordered a couple more boards to see if this one is just faulty.

githubDLG, 2018-04-04:
hi, guys~~ @mikeller  @turboed13b @brycedjohnson 
i am here now , sorry for being late. :stuck_out_tongue_winking_eye:
`Crazybee flight controller with spi-rx. D16 mode causes twitching while flying. Dshot600 mode causes failsafe while in d16 mode.`
`D8 mode works fine in multishot but in dshot600 it arms then immediately disarms with a telemetry lost from tx.`

Yes,yes~~    Actually, we have already know `twitch` and `d8/d16 arm/disarm` issue, and  `Flysky SPI rx` seems  also suffered twitch issue... We have disscued this problems in slack. :joy:

And , don't be worrid about them , i think our BF devs will fix it soon ~~:joy:

githubDLG, 2018-04-04:
And , @turboed13b   
`I tried pretty much every setting. I ordered a couple more boards to see if this one is just faulty.`
Do not do that, i'm seriously~~~ :joy:
I strongly recommand you that, fly with D8 first, and wait for D16 to be fixed  :joy:

turboed13b, 2018-04-04:
I am in the U.S. D16 mode works but it is pretty much impossible to fly with the twitching and failsafes.

I am flying d8 mode in dshot600 right now without any problems. I just can't enable dshot motor reversal it won't arm.

brycedjohnson, 2018-04-04:
```

# diff all
# Betaflight / CRAZYBEEF3FR (CBFR) 3.4.0 Mar 31 2018 / 18:16:36 (14d1c52cf) MSP API: 1.39
defaults nosave
name Bryce[\]^_

feature AIRMODE
aux 0 0 0 1200 2100 0
aux 1 1 2 1200 2100 0
aux 2 13 1 1700 2100 0
aux 3 35 1 1300 1700 0
aux 4 36 3 1700 2100 0
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set acc_calibration = -37,-89,50
set min_check = 1010
set max_check = 1990
set dshot_idle_value = 900
set motor_pwm_protocol = DSHOT300
set bat_capacity = 260
set vbat_max_cell_voltage = 45
set vbat_min_cell_voltage = 28
set vbat_warning_cell_voltage = 29
set vbat_scale = 113
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warnings = 63
set osd_vbat_pos = 385
set osd_rssi_pos = 40
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2455
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 352
set osd_current_pos = 2400
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 2441
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2056
set osd_avg_cell_voltage_pos = 2433
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 465
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 472
set osd_rtc_date_time_pos = 2048
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF
set osd_stat_rtc_date_time = ON
set vcd_video_system = 2
set frsky_spi_tx_id = 72,197
set frsky_spi_offset = -18
set frsky_spi_bind_hop_data = 6,219,199,179,159,139,119,99,79,59,39,19,234,214,194,174,154,136,114,94,74,54,34,14,229,209,189,169,149,129,109,89,69,49,29,9,224,204,184,164,144,124,104,84,64,44,24,0,0,0
set frsky_x_rx_num = 5
profile 0

set crashflip_motor_percent = 50
set dterm_setpoint_weight = 254
set p_pitch = 80
set p_roll = 80
set i_roll = 50
set d_roll = 35
set p_yaw = 80
set p_level = 120
set level_limit = 60

profile 0
rateprofile 0

set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80
rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

rateprofile 0
save
# ```

turboed13b, 2018-04-13:
Let me know if you guys get it figured out.

githubDLG, 2018-04-13:
Okay~~

evnngsky, 2018-04-18:
Moreover, I found motor run without arm in my case(D16, DSHOT600). It should hurt me!!!!!

mikeller, 2018-04-18:
@evnngsky: What ESCs are you using?

evnngsky, 2018-04-18:
@mikeller I've apply Racerstar 40A BLHeli-S(BB2) ESCs to my quad.
I thought It has a possibility to set turtle mode with wrong signal or internal error.

githubDLG, 2018-04-19:
@evnngsky  What FC are you using? Crazybee or ?    :smiley:

evnngsky, 2018-04-19:
@githubDLG It is Matek F411ONE (F411RX target)

githubDLG, 2018-04-19:
haha , okay , so , this problems are  seems related to rx code other than in a unique FC 

:smiley:

githubDLG, 2018-04-19:
If it is really need to open a new issue, reopen this #5576 maybe a better choice? :smiley:

rustbg, 2018-04-19:
Same board (crazybee), same problems. Even on 1khz gyro/pid.

githubDLG, 2018-04-19:
@rustbg @turboed13b  @brycedjohnson   
Sorry for bad user experiences on your guys:sob::disappointed_relieved::upside_down_face:

evnngsky, 2018-04-19:
I will upload some videos contains symptoms of my quad to find out reasons after several hours.

brycedjohnson, 2018-04-19:
D16 and D8 issues appear to be different.  I have a fix for the D8 failsafe in progress here #5707.  There was a bug where the dshot commands were holding off the spi-RX a bit on arming and disarming.

turboed13b, 2018-04-20:
f4 whoop pro same issue but worse. Matek411 target. D16 mode is unflyable it just failsafes. D8 mode range is bad and has the same twitch when only when far away. when I set guard time to 0 the twitches go away but I can tell there is something going on still because it doesnt react as fast as my stick inputs sometimes.

```
A00: FREE
A01: MPU_EXTI
A02: FREE
A03: FREE
A04: MPU_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI
A14: RX_SPI
A15: RX_SPI_CS 3
B00: ADC_BATT
B01: ADC_CURR
B02: RX_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 1
B07: MOTOR 4
B08: MOTOR 3
B09: LED
B10: MOTOR 2
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI
C15: BEEPER
D02: FREE
```

```DMA1 Stream 0: MOTOR 1
DMA1 Stream 1: MOTOR 2
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 3
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

jonnyvolts, 2018-04-20:
Hey guys also running matek f411 and Qx7s in either d8 or d16 just failsafes all the time.

brycedjohnson, 2018-04-20:
@turboed13b @jonnyvolts If you want to try the changes in #5707 that helped with fix all the issues in D8 mode and some of them in D16.

[betaflight_3.4.0_CRAZYBEEF3FR.hex.zip](https://github.com/betaflight/betaflight/files/1932544/betaflight_3.4.0_CRAZYBEEF3FR.hex.zip)
[betaflight_3.4.0_MATEKF411.hex.zip](https://github.com/betaflight/betaflight/files/1932545/betaflight_3.4.0_MATEKF411.hex.zip)

Also if you turn the OSD menu to Debug, then the 3rd number tells when you start to failsafe and miss packets and how many.  Also you'll see it increment if you radio is off.  If you could grab a dvr when that is happening it would be helpful.

[image]

I don't have a MatekF411, so if it won't arm try and turn dshot_burst = off in the CLI (for now, a bug I'm trying to work around).

I believe the D16 where it 'failsafes' is actually a lockup.   So try on D8 first.  Turtlemode and everything should be working there so you can have it on.  This doesn't address any of the twitches in D16 mode either.

Also @turboed13b, when you are flying far away it actually just could be getting out of range, Having the debug OSD would be helpful to see the dvr of.

Thanks!

johnblackj, 2018-04-20:
I tried D8 after read this and had the same limited range and failsafe.
Also I noticed that when testing the dshot beeper, every time I beeped and stop the QX7 would report "telemetry lost, telemetry recovered "
Also when disarmed if the radio was close to the quad, I would get RSSI critical and the motors would pulse for a second. Also caught a finger pick up my quad after landing  with this one

jonnyvolts, 2018-04-20:
I had my motors do a low rpm spin up while disarmed as well... and on the bench near the quad constant telemetry and rssi alarms

brycedjohnson, 2018-04-20:
@johnblackj Yes that is the issues that I fixed with the hex above, test if out if you want and let me know if it works.

Also when your quad is really close to your radio it will swamp the signal and you'll lose connection/failsafe.  Does it do that when you are >1m away?  Is that with dshot or multishot? @jonnyvolts 

Please put a diff all on your post if you want to help fix this issue. There a few variables they may affect this and I have no idea how you have it configured.

johnblackj, 2018-04-20:
Dshot 1200. I knew rssi and telemetry gets mess up as close range. 
The first time it pulsed the radio was on a strap around my neck and I had just picked up the quad. I was then able to reproduce but just placing the radio next to the quad. 1m away and it was good. Unless I tried the range test and it failsafed  then it did the same thing. 
I was kinda thinking the odd random twitch I got was made that pulse happening on a momentary failsafe during flight. I never crashed on a failsafe, I'd get the OSD failsafe warning.  But I cant confirm the twitch and failsafe happened at exactly the same time.
I swapped out the flight controller but I have another build I might test on with this hex

brycedjohnson, 2018-04-20:
@johnblackj. What FC?  I assume the MATEKF411RX-AIO board?  Did those issues happen on D8 or D16 or both.  What version of betaflight?

`Diff all` would get all that info and more! :)  DVR I'm sure would help as well.

turboed13b, 2018-04-20:
Seems like matek have slightly different issues. Both D8 and D16 have really bad range. D8 I can fly in although not far and d16 is pretty much impossible just failsafes.

The pulse seems like a symptom of rx loss I get the same pulses when flying on the edge of range and it will make my tiny whoop gain altitude.

johnblackj, 2018-04-20:
D8-D16 experience seems the same as turboed13b, I was running the MatekF411-one board with betaflight 3.3.0  which is what it comes with.  
I can diff the board when I get home. I'd need to look through the card in my googles to see if I have DVR video.

brycedjohnson, 2018-04-20:
I took my crazybee f3 for a quick run in the backyard to see if I got bad distance (I was partially in the basement of my house so i didn't get rained on).  I got about 500ft before I got bad video (25mW dipole) and I hadn't had any missed packets.  I did get some missed packets when I put it down in the grass. 

The 3rd number 6 is the number of missed packets in the video (you always miss a few on starting up the radio before it binds)

https://youtu.be/hwEDn_gewgE

brycedjohnson, 2018-04-20:
You can see the twitch happening on receiver tab with D16.  One/Some of the channels will go to the min.  That might explain how people have had quads arm on on glitches.  I bet those people have the arm switch in bottom of the mode range?

[image]

turboed13b, 2018-04-21:
That matek hex is the wrong one. Need the rx version.

Rennkopter, 2018-04-21:
Hello to all,
I'm new here and also struggling with the Matek F411-ONE frsky_x D16 problem. Hopefully there will be a solution to this problem soon!
I've uploaded a maidenflight video where you can see this twitches.

https://youtu.be/u9kTNcXV2eo

With frsky_d D8 mode no problems at all.

o7 fly safe, rennkopterpilots

brycedjohnson, 2018-04-21:
@turboed13b Oops sorry about that, here is the updated hex:

[betaflight_3.4.0_MATEKF411RX.hex.zip](https://github.com/betaflight/betaflight/files/1934783/betaflight_3.4.0_MATEKF411RX.hex.zip)

bullcrypto, 2018-04-21:
Currently, running Matek 411RX BF3.3.1 D8 CH1-8 Bind no issues close range, will try this 3.4.0 build later today as I was having the same issues as the rest.

turboed13b, 2018-04-21:
@brycedjohnson Ok in d8 mode I actually get a little bit more range with that hex file. I was able to fly in my basement although it did have a lot of packet loss and twitching was still there.

I will post the video later today.

brycedjohnson, 2018-04-21:
I might have gotten a fix for the D16 issues with #5741

Hexes if you want to test it out. D16 is just this fix. Combo includes my partial Dshot commands fix.
[D16_dshotfix.zip](https://github.com/betaflight/betaflight/files/1935060/D16_dshotfix.zip)

turboed13b, 2018-04-22:
D8 mode matek411 debug.

https://www.youtube.com/watch?v=B75NS59Nx2c&feature=youtu.be

turboed13b, 2018-04-22:
I just tried the recent hexes. The matek range just sucks in d16 I cant really fly in it. Crazybee in d16 it seems like the twitches are gone but it still has that lockout problem. It just drops from the sky no osd shows up either like a normal failsafe.

rubenix, 2018-04-22:
I thought I was doing something wrong as I cannot enable turtle mode on my Snapper7 quad.
Crazybee flight controller with spi-rx; Board on BF 3.3 & BLHeli to 16.67
Arm & Flip over crach on different switches 
Dshot600 fine and can get motor beeps on a switch
As soon as I enable turtle mode on a switch I cannot arm at all...Not on a "normal arm" nor with "turtle mode" activated"
Just suscribed here to get noticed as soon as problem got fixed ;P

turboed13b, 2018-04-22:
@rubenix Use the most recent 3.4 hex on here. It fixes all those problems.

githubDLG, 2018-04-23:
@turboed13b   ` It just drops from the sky no osd shows up either like a normal failsafe.`

We have the same issue when in testing crazybee.

evnngsky, 2018-04-23:
@turboed13b I'd applied your hex(d16 only) to my F411RX, And I couldn't find serious problems.
But It lost and recovered telemetry, repeatedly.

turboed13b, 2018-04-23:
@evnngsky d16 mode doesn't work correctly yet. D8 mode is the only choice right now.

[71 more replies]

PR #5590 (2018-03-31): Fixed DMA conflicts for CRAZYBEEF3 targets.
Fixes #5576.