betaflight/betaflight #5577: Cant set dshotbeacon SRF3, no dhsot command output. Cli working 3.3.0

conuthead, 2018-03-30:
Hi
Using a sp racing f3. Dshot works fine. When I try to set dshot beacon in configuration tab the selection does not stay. It slides back to off position.  If set FOAC to a switch, no dshot command is sent to the esc. Using a fresh clean flash of betaflight 3.30, 3.25, or 3.20. 

If i go to cli and type "dshotprog 255 <command#>" It sens out the dshot command. I am using an oscilloscope to verify the output is not there as I though it was an ESC problem at first. Is this a known bug issue for this FC? 

Proshot shows same behavior. dshotprog works but can't get fc to send command with switch nor beacon.

conuthead, 2018-03-31:
I there any workaround where i can set and AUX switch to send a dshot command manually?  I would like to get this working. Seems so simple to send dshot commands but the FOAC mode nor BEEPER does not send anything.  It seems dshotprog sends out the same command repeated 10 times and works normally?

etracer65, 2018-04-01:
As requested in the issue creation instructions:

> 5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

conuthead, 2018-04-02:
Ok sorry, here is the diff. I flashed a clean install, only thing i did was disable sensors i dont want an turn on arm and FOAC on the switchs aux 0 and aux 1.
'''# Betaflight / SPRACINGF3 (SRF3) 3.3.0 Mar  2 2018 / 04:02:50 (177472b4f) MSP API: 1.37

feature -TELEMETRY
feature -RSSI_ADC
feature -TRANSPONDER
aux 0 0 1 1575 2100 0
aux 1 35 0 1575 2100 0
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set motor_pwm_protocol = DSHOT600
profile 0

rateprofile 0
'''

The beacon work and sends out dshot commands. The beacon has to be set to beacon number 2. Beacon 1 will not save settings. Dshot FOAC still sends no signal. I am testing this on the ground, is this my problem?

conuthead, 2018-04-03:
The issue was all mine. The dshot command was not sent on the time I expected to see it. The behavior that I see when I use the FOAC switch a dshot command of 21 is sent upon rearming to send to the esc to reverse the motors. When using the motors normally a dshot command of 20 is sent at the beginning of arming. I had incorrectly assumed the dshot command would be sent when the switch FOAC was activated but I understand now the logic for doing it this way.

closed 2018-04-03 by conuthead

richard-scott, 2018-04-03:
@conuthead - Not being able to enable and Dshot Beacon to No. 1 via the Configurator is a reported [issue](https://github.com/betaflight/betaflight-configurator/issues/975). You are not the only one having issues with this.