betaflight/betaflight #5579: BLHELI:  Connection failed, ensure that you have selected the correct port or restart your device.

Steven-Joblin, 2018-03-30:
I am running BetaFlight 3.3.0 on my Eachine Auroa 100.
I opened BetaFlight and then connected (via COM4 at 115200 baud). Everything in BetaFlight works great.
I then hit the disconnect button, but kept the quad plugged in via the USB cable.

I opened BLHELI and hit the Connect button (via COM4 at 115200 baud).

I get the following:
2018-03-29 @ 20:58:57 -- Running - OS: Windows, Chrome: 65.0.3325.181, Configurator: 1.1.0
2018-03-29 @ 20:59:00 -- Serial port successfully opened with ID: 1
2018-03-29 @ 20:59:01 -- MultiWii API version received - 1.37.0
2018-03-29 @ 20:59:01 -- Flight controller info, identifier: BTFL, version: 3.3.0
2018-03-29 @ 20:59:02 -- Running firmware released on: Mar 2 2018 03:56:08
2018-03-29 @ 20:59:03 -- Board: OMNI, version: 0
2018-03-29 @ 20:59:04 -- Unique device ID received - 0x31003c4234571620363434
2018-03-29 @ 20:59:10 -- Connection failed, ensure that you have selected the correct port or restart your device.

I tried multiple times and even tried running the ImpulseRC_Driver_Fixer program.

I read in another issue to try re-flashing Betaflight but not paste back my settings that I had saved via the DUMP command.

I reflashed BetaFlight 3.3.0 and sure enough BLHeli connected and work perfectly. As soon as I copied back all my settings (that I had saved via the DUMP command on CLI), sure enough BLHeli had the same problem. What setting am I changing in Betaflight that is causing this to happen?

Here is my DIFF from BetaFlight:
```
# diff
# Betaflight / OMNIBUS (OMNI) 3.3.0 Mar  2 2018 / 03:56:08 (177472b4f) MSP API: 1.37
name Aurora100

feature -RX_PPM
feature RX_SERIAL
feature MOTOR_STOP
feature LED_STRIP
beeper -ON_USB
map TAER1234
serial 0 1 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
led 0 0,8::AI:2
led 1 7,8::C:1
led 2 8,8::C:1
led 3 15,8::AI:2
mode_color 6 0 2
mode_color 6 1 4
mode_color 7 0 0
aux 0 0 0 1775 1900 0
aux 1 1 1 1725 1950 0
aux 2 2 1 1400 1600 0
aux 3 27 2 1050 1275 0
aux 4 13 3 1100 1225 0
set moron_threshold = 32
set acc_trim_pitch = 7
set acc_trim_roll = 35
set min_check = 1010
set rssi_scale = 30
set max_aux_channels = 6
set serialrx_provider = SPEK2048
set min_throttle = 1050
set dshot_idle_value = 1500
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 400
set failsafe_delay = 10
set failsafe_kill_switch = ON
set vbat_warning_cell_voltage = 29
set osd_units = IMPERIAL
set osd_rssi_pos = 1399
set osd_flymode_pos = 2554
set osd_throttle_pos = 129
set osd_vtx_channel_pos = 200
set osd_crosshairs = 2048
set osd_mah_drawn_pos = 2112
set osd_craft_name_pos = 480
set osd_gps_speed_pos = 66
set osd_gps_sats_pos = 1410
set osd_altitude_pos = 161
profile 0

set setpoint_relax_ratio = 30
set dterm_setpoint_weight = 200
set d_pitch = 22
set p_roll = 43
set d_roll = 20
set d_level = 100
rateprofile 0
# 
```

flint723, 2018-03-30:
You are saying it works in 3.2 and not 3.3?  May be a slight typo about version in your post.

Steven-Joblin, 2018-03-30:
Correct, that is what I am saying.  That said, I just went back and tried to recreate the scenario and was able to get it to work.  I am going to close this issue as I think it is a driver related issue that is happening with my PC... I am getting inconsistent behaviors and sometimes I need to use the  Impulse Driver Fixer and then I can get it to work so I don't think it is an issue with BetaFlight or an issue with BLHeli.

closed 2018-03-30 by Steven-Joblin

claytonbakerjr, 2018-03-30:
@Steven-Joblin 
Steve, food for thought even though this is closed.
In my past experience I found that in some cases not unplugging
the USB cable can cause connection issues when going back and forth
between BF and BLHeli.
I got into the habit of totally disconnecting and then reconnecting and since
then I have been about 98% successful with connections.
Being an IT guy I feel that the OS does not always sever the com port connection
properly and by disconnecting the USB it resets.
As per the Driver Fixer, lately since v3.2 I am finding that when I flash a board up to 3.2
or 3.3 I almost have to run the driver fixer each time. I also seen something from Joshua Bardwell
where he mentions something very similar. Once the board is successfully flashed I never have
to touch the Driver Fixer again unless I flash to a newer release.
In any case I hope you work this out :)