betaflight/betaflight #5588: Still struggling with Date-and-Time RTC, etc

JonnyPhenomenon, 2018-03-31:
Cliffnotes version: I want the clock to appear in my DVR recordings and blackbox logs. 
(thought I had it, but its not working) 

Long version:

so I am trying to get my betaflight f3 flight controller to have its RTC clock get updated from the Taranis (X9d+), and I am coming up empty.  Ive been trying to figure this out on my own, and I just don't understand that it is I am missing.

I have updated the taranis firmware, running BF 3.3 on the FC, got LUA scripts working so I can control my irc tramp settings over uart 3, via smartport on uart1, so thats all working fine.  

Telemetry data shows up on the LUA script pages on the Taranis, 
But if I go to the CLI, I do not see any clock data when I run the "status" command in the CLI. Oh, and all the telemetry data dissapears on the taranis until I exit the CLI, which reboots the FC anyway.

I tried adding the RTC to the OSD so the clock is shown during flight (for testing) and nothing appears.  I added it to the post flight Stats page as well, and I get the message "NO RTC" at the top of the list.  maybe this is a clue? Does the BFF3 NOT have a rtc? - I mean, I know it would just get reset at every battery swap, but it knows perfectly well how much time has passed, so I think it must have an rtc... 

Ive searched and searched, and evferything I read says this should be working.  
commits made in #4618 add the rtc to the stats page.  in that commit, it says #ifdef USE_RTC_TIME  printRtcDateTime(&buff[0]); (Or else, print NO RTC) 
so I looked for references to USE_RTC_TIME in the source, and see that in the targets folder the "common_fc_pre.h" file defines USE_RTC_TIME, while a small handful of other targets have specifically undefined it.  the betaflight target.c file includes the platform.h file, which includes the common_fc_pre.h file, so this leads me to think that the betaflightf3 target I am using should have USE_RTC_TIME enabled, am I right? 

I am not an idiot, but there must be something glaringly obvious that I am missing...  can anybody help me out here?
 
here are some Other links/issues etc that I have read, which tell me its supposed to be working, but it isnt. - and they dont explain how...
#5412 @wuhao19920619 said it wasnt working, and @DieHertz said the rtc clock value is sent over telemetry from the transmitter.
#5520 adds time to logs, when there is a gps connection
Pull #4289 supposeedly makes BF aware of date and time by LUA script, https://github.com/betaflight/betaflight/pull/4289 which needed to be added to the lua scripts, betaflight/betaflight-tx-lua-scripts#46 

So looking at the bf-tx-lua scripts that I have loaded, I see the MSP_SET_RTC command should be getting sent by the background.lua as soon as the telemetry connection is available... something like "MSP_SET_RTC, 2018 03 30 22 25 00" 

regardless, it doesnt seem to be working...

you get the idea...

Arakon, 2018-03-31:
As far as I know, the RTC functionality is not enabled for F3 controllers due to lack of space.

mikeller, 2018-03-31:
@Arakon: It's enabled for most of the F3s (including BETAFLIGHTF3), so that's a negatory.

JonnyPhenomenon, 2018-03-31:
OK, so mikeller's comment gives some weight to what I had found, that says USE_RTC_TIME is defined on the betaflightf3 target... but it doesnt explain why it isnt working.  

Note: telemetry from the FC is populating on the taranis, so my LUA screens show me pid values and VTX settings, and whatnot, and I can change them, and save them. but the RTC doesnt show up in the OSD, nor during the stats display on disarm. 

this is a betaflight f3, with version 3.3.0 release. 
the taranis X9d+ is using the latest OpenTX 2.2.1 nightly, (I had tried it with the "non nightly" release as well, and it didnt work with that either. ) 
I am using the betaflight LUA 1.0.1 

interestingly enough, I also have a betaflight **F4** FC with a crossfire rx connected, which DOES update the rtc from my taranis, but does NOT populate the LUA pages.  Go figure!

JonnyPhenomenon, 2018-04-02:
is there a way to start building/compiling the source for the betaflightf3 target and export a list of all of the variables that have been defined? 

I don't compile much. I just setup the building environment, and I was trying to poke around.  I tried make -p BETAFLIGHTF3; and the resulting stream of text didnt give me anything I could understand...  

on a side note: I tried joining the betaflight.slack.com site, and couldnt get connected.  the http://www.betaflight.ch site is not responding...

tonyjobson, 2019-01-22:
I'm having the same problem. If i connect my FC to a usb on the computer and go to the OSD page then it displays the right time.
When I was on OpenTX 2.2.2 if i powered the quad up first and then the Taranis is seemed to work, but if the Taranis was already on then it didn't work (Issue 6504 suggested this)

Running a telemetry LUA script sometimes fixes it, but not always.

I'd really like to get to the bottom of this. RTC makes organising my DVR footage so much easier.

nnsense, 2020-11-12:
Hi, same here, it works while connected to a PC running betaflight, it doesn't when connected to the same pc while betaflight is _not_ running. Apparently it just reads the computer's clock while using BF.

klutvott123, 2022-04-21:
previously the lua scripts would just set the time once and didn't care if it succeeded or not. The rtc routine has been improved in the new lua scripts so that it will keep trying to set the rtc until it gets a response from the FC that lets it know it succeeded so it should work with the latest lua version.

closed 2022-04-21 by klutvott123