betaflight/betaflight #5591: ibus serial rx & telemetry

haloweenhamster, 2018-03-31:
Hi I'm using an sp racing f3 and BF 3.2.5 / configurator 10.2.0 I have Ibus connected to uart tx3, serial rx and ibus telemetry enabled on uart 3 and everything works

updating to BF 3.3 if I try to enable ibus telemetry after selecting ibus receiver the ports page returns to default

if I do a diff in 3.3 after tryingto setup ibus its blank as ports no longer enabled and receiver type goes blank as theres no serial rx to use

```# diff
# Betaflight / SPRACINGF3 (SRF3) 3.2.5 Feb 11 2018 / 01:12:56 (6e69ff00c) MSP API: 1.36

feature -RX_PPM
feature -RSSI_ADC
feature -TRANSPONDER
feature RX_SERIAL
feature MOTOR_STOP
feature ANTI_GRAVITY
serial 2 4160 115200 57600 0 115200
aux 0 0 1 1700 2100
aux 1 1 2 900 1300
aux 2 2 2 1300 1700
aux 3 13 3 1700 2100
aux 4 15 3 1300 2100
aux 5 27 5 1700 2100
aux 6 28 3 1700 2100
aux 7 36 5 900 1300
rxfail 3 h
rxfail 4 s 1000
set rssi_channel = 5
set serialrx_provider = IBUS
set failsafe_kill_switch = ON
set failsafe_throttle_low_delay = 20
set current_meter = NONE
profile 0

rateprofile 0```

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

both slack links have page not responding http://185.87.186.121 / http://185.87.186.121:3000
no response on facebook

McGiverGim, 2018-03-31:
The iBus telemetry has been removed from F3 for space constraints. You need to customize your own firmware if you want to enable it.

varkokonyi, 2018-03-31:
@haloweenhamster If you tell me what I can remove, I will build a hex for you which includes ibus

haloweenhamster, 2018-03-31:
Thanks but I'm a noob so not sure what can be removed or what the advantage of 3.3 is, I'll just stick to 3.2.5 until I know what I'm doing.
I thought it was a glitch, I didn't realize it was a removed feature due to space

McGiverGim, 2018-03-31:
You can remove all the rx protocols but the flysky one.

varkokonyi, 2018-03-31:
I removed every RX protocol and added ibus telemetry, please try this @haloweenhamster 
https://drive.google.com/drive/folders/1EOKCee3ivQMly1TV68n7b4E4O3_SxeI5?usp=sharing

haloweenhamster, 2018-03-31:
Thanks, flashing it now
I hope its useful to others
I feel bad that you have spent time and effort to customize a firmware for me but I am very grateful

varkokonyi, 2018-03-31:
It didn't take more than 10 minutes, I'm happy to help. Let me know if it works

haloweenhamster, 2018-03-31:
Will do
Its all setup fine, I'll fly it tomorrow if there's a break in the rain
Thank again

McGiverGim, 2018-03-31:
Remember to put this command in CLI:
`set ibus_sensor=1,2,3,0,0,0,0,0,0,0,0,0,0,0,0`
If not you will not receive nothing in 3.3 (there was a bug).

haloweenhamster, 2018-03-31:
Thanks I've entered that
Plugged in a battery and checked my transmitter
All channels working in receiver tab & telemetry displayed
Did notice that rssi which is set as aux 1 with a reading of 1990 only gave 16% on setup tab, 98% on old firmware 
is that something that I need to setup / has been removed / glitch?
Thanks

haloweenhamster, 2018-04-01:
```
# set rssi
rssi_channel = 5
Allowed range: 0 - 18

rssi_src_frame_errors = ON
Allowed values: OFF, ON

rssi_scale = 41
Allowed range: 1 - 255

rssi_invert = OFF
Allowed values: OFF, ON

osd_rssi_alarm = 20
Allowed range: 0 - 100

osd_rssi_pos = 234
Allowed range: 0 - 3071

osd_stat_min_rssi = ON
Allowed values: OFF, ON

# set ibus
ibus_sensor = 1,2,3,4,5,6,21,22,0,0,0,0,0,0,0
Array length: 15
```
Thanks, I'm using a Flysky I6 with flyplus 1.7.5 firmware, I tried rssi_invert = on and rssi stayed at 84% with aux 1 going from 1990 down to 1510 & failsafe set to 1000, also is there any way to invert arm output? tx has it reversed
thank you for all of your assistance

McGiverGim, 2018-04-01:
You only need to select the correct arm range in the modes tab.
Don't forget to close this issue, as it has been solved. 😉

haloweenhamster, 2018-04-01:
Will do I was just hoping to get answers to the last 2 questions, slightly off topic to the original issue but related as its telemetry

The arm input works fine as set in modes tab, its the arm output, telemetry says armed when disarmed and unarmed when armed

Rssi invert on only reads 84%, invert off reads 18% with channel at 1990 16% with failsafe / channel at 1000

varkokonyi, 2018-04-01:
Maybe try adjusting `rssi_scale` . Just a guess...

closed 2018-04-01 by haloweenhamster

haloweenhamster, 2018-04-03:
@varkokonyi I was wondering if you could do a quick change for me

ibus_shared.c
```
case IBUS_SENSOR_TYPE_ARMED:
            value.uint16 = ARMING_FLAG(ARMED) ? 0 : 1;
            break;
```

Swap it to 1 : 0;

I'm hoping that will fix the ibus telemetry 
Thanks

varkokonyi, 2018-04-03:
Okay, I can do that. I'll do it today

mikeller, 2018-04-03:
@haloweenhamster, @varkokonyi: Please share your findings. If this is a bug it should be fixed in the firmware.

varkokonyi, 2018-04-03:
Dona @haloweenhamster

haloweenhamster, 2018-04-03:
Yeah that's fixed it, cleanflight has the same issue & probably butterflight

Thank you for all of your help and time

varkokonyi, 2018-04-03:
Then I'll submit a PR from this

McGiverGim, 2018-04-03:
Good job! I suppose this only affects modded TX. This sensor is not available for standard TX.

haloweenhamster, 2018-04-03:
I'm using a modded i6, not sure what's available on other tx's like the evo or any running multimodules

McGiverGim, 2018-04-03:
I think this sensor (21) is not used too much... :P iNav has the same problem:

https://github.com/iNavFlight/inav/blob/c66c28e7dc1e3c1ec8d1475f840691011ca712ba/src/main/telemetry/ibus_shared.c#L177-L178

PR #5613 (2018-04-03): Fix arming-releated IBUS telemetry bug
Fixes #5591 

@haloweenhamster discovered that these should be swapped