betaflight/betaflight #5592: ITerm reset to zero - mc falls from the sky

TheIsotopes, 2018-03-31:
using latest 3.3.1 builds ... during a flip, iterm is reset to zero on all axis (this does not change anymore) and my copter falls from the sky ... can anyone tell me whats wrong

here an bb-log ... you can see it in flight 1 @ 00:25.072 and in flight 2 @ 02:47.315
[BTFL_BLACKBOX_LOG_Mark1_20180325_161000.TXT](https://github.com/betaflight/betaflight/files/1865257/BTFL_BLACKBOX_LOG_Mark1_20180325_161000.TXT)

and here my settings
```
# name
name Mark1

# feature
feature -SOFTSERIAL
feature -TELEMETRY
feature -OSD
feature LED_STRIP
feature AIRMODE
feature DYNAMIC_FILTER

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW

# map
map TAER1234

# led
led 0 9,2::CW:6
led 1 9,3::CW:6
led 2 9,4::CW:6
led 3 9,5::CW:6
led 4 6,5::CW:6
led 5 6,4::CW:6
led 6 6,3::CW:6
led 7 6,2::CW:6

# aux
aux 0 0 0 1600 2000 0
aux 1 13 1 1000 1200 0
aux 2 26 0 1600 1800 0
aux 3 31 0 1200 1400 0
aux 4 36 1 1800 2000 0

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rc_interp_ch = RPYT
set max_aux_channels = 2
set serialrx_provider = SUMD
set blackbox_device = SPIFLASH
set dshot_idle_value = 550
set motor_pwm_protocol = DSHOT300
set pid_process_denom = 1

# profile
set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set setpoint_relax_ratio = 50
set dterm_setpoint_weight = 100
set p_roll = 44

# rateprofile
set roll_rc_rate = 145
set pitch_rc_rate = 145
set yaw_rc_rate = 120
set roll_expo = 20
set pitch_expo = 20
set roll_srate = 75
set pitch_srate = 75
```

etracer65, 2018-03-31:
In both logs motor 1 stopped producing thrust.  Most likely an ESC de-sync or possibly some problem with that motor or ESC (since it was the same motor both times).  If you look at the motor traces you'll see that motor 1 is being commanded to full throttle because that corner is not making thrust and the flight controller is telling it to push harder and harder.

TheIsotopes, 2018-04-01:
you seem to be right ... i have looked at other logs and there too motor 1 is being commanded to full throttle ... i'll take a look at motor and esc ... imho it's not a bug in bf and i closed now

closed 2018-04-01 by TheIsotopes