betaflight/betaflight #5597: Bug in throttle RC input handling

ctzsnooze, 2018-04-01:
I was logging @joelucid's throttle boost PR and noticed a disconcerting 'glitch' upwards in the motors trace at the onset of any throttle input that starts at zero.

It can be easily replicated, just log your quad with props off, air mode off, do some quick or slow throttle blips up from zero throttle.

You'll see in the motors trace a tiny blip up that happens at the time of the very first throttle step above zero.   It is sometimes a fraction of the RC input step.  It can be quite large if the input is quick.

Some images from my logs:

[image]
[image]
[image]

I guess we don't see it in flight, since it only happens when coming up from zero throttle.  If we do that after a flip or roll or similar zero-throttle manoeuvre, we probably wouldn't notice.

So although this definitely should not happen, it might not be of any importance to fix.

DieHertz, 2018-04-01:
I'll look into it, could you provide a diff as well?

ctzsnooze, 2018-04-01:
Thanks!

`# diff all

# version
# Betaflight / FF_PIKOF4 (PIK4) 3.4.0 Mar 26 2018 / 11:56:08 (4b20ab5) MSP API: 1.39

# reset configuration to default settings
defaults nosave

# name

# resources
resource MOTOR 1 B00
resource MOTOR 2 A02
resource MOTOR 3 A03

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature TELEMETRY
feature LED_STRIP
feature ANTI_GRAVITY

# beeper

# map
map TAER1234

# serial
serial 5 32 115200 57600 0 115200

# led
led 0 0,0::CW:2
led 1 1,0::CW:6
led 2 2,0::CW:6
led 3 3,0::CW:2

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0
aux 1 26 1 1300 2100 0
aux 2 28 0 1700 2100 0
aux 3 35 2 1300 2100 0

# adjrange
adjrange 0 0 3 900 2100 12 3

# rxrange

# vtx

# rxfail

# master
set align_gyro = CW0
set gyro_hardware_lpf = EXPERIMENTAL
set gyro_lowpass_hz = 150
set gyro_lowpass2_hz = 300
set gyro_lowpass2_order = 2
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set moron_threshold = 64
set align_acc = CW0
set acc_hardware = NONE
set min_check = 1020
set rc_interp_ch = RPYT
set max_aux_channels = 8
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set min_throttle = 1060
set min_command = 975
set dshot_idle_value = 420
set motor_pwm_protocol = DSHOT300
set current_meter = ADC
set yaw_motors_reversed = ON
set small_angle = 180
set auto_disarm_delay = 0
set pid_process_denom = 1
set frsky_vfas_precision = 1
set report_cell_voltage = ON
set osd_warnings = 0
set osd_vbat_pos = 2092
set osd_rssi_pos = 2088
set osd_tim_1_pos = 2102
set osd_tim_2_pos = 2081
set osd_flymode_pos = 2381
set osd_throttle_pos = 2273
set osd_vtx_channel_pos = 2425
set osd_ah_pos = 2248
set osd_current_pos = 2433
set osd_mah_drawn_pos = 2401
set osd_craft_name_pos = 2410
set osd_gps_speed_pos = 2266
set osd_gps_lon_pos = 2130
set osd_gps_lat_pos = 2113
set osd_gps_sats_pos = 2099
set osd_home_dir_pos = 2350
set osd_home_dist_pos = 2351
set osd_compass_bar_pos = 2314
set osd_altitude_pos = 2295
set osd_pid_roll_pos = 2471
set osd_pid_pitch_pos = 2503
set osd_pid_yaw_pos = 2535
set osd_debug_pos = 2049
set osd_power_pos = 2369
set osd_pidrate_profile_pos = 2393
set osd_avg_cell_voltage_pos = 2124
set osd_pit_ang_pos = 2305
set osd_rol_ang_pos = 2337
set osd_battery_usage_pos = 2440
set osd_disarmed_pos = 2186
set osd_nheading_pos = 2359
set osd_nvario_pos = 2327
set osd_esc_tmp_pos = 2130
set osd_esc_rpm_pos = 2131
set task_statistics = OFF
set pwr_on_arm_grace = 0

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 90
set dterm_lowpass2_hz = 180
set dterm_notch_hz = 0
set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set anti_gravity_threshold = 500
set anti_gravity_gain = 5000
set setpoint_relax_ratio = 0
set dterm_setpoint_weight = 100
set iterm_rotation = ON
set iterm_windup = 30
set p_pitch = 37
set i_pitch = 65
set d_pitch = 20
set p_roll = 37
set i_roll = 60
set d_roll = 20
set p_yaw = 55
set i_yaw = 62
set d_level = 100

# profile
profile 1

set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 90
set dterm_lowpass2_hz = 180
set dterm_notch_hz = 0
set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set anti_gravity_threshold = 500
set anti_gravity_gain = 5000
set setpoint_relax_ratio = 0
set dterm_setpoint_weight = 100
set iterm_windup = 30
set p_pitch = 36
set i_pitch = 65
set d_pitch = 17
set p_roll = 36
set i_roll = 60
set d_roll = 17
set p_yaw = 55
set i_yaw = 62
set d_level = 100

# profile
profile 2

set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 90
set dterm_lowpass2_hz = 180
set dterm_notch_hz = 0
set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set anti_gravity_threshold = 500
set anti_gravity_gain = 5000
set setpoint_relax_ratio = 1
set dterm_setpoint_weight = 100
set iterm_windup = 30
set p_pitch = 36
set i_pitch = 65
set d_pitch = 17
set p_roll = 36
set i_roll = 60
set d_roll = 17
set p_yaw = 55
set i_yaw = 62
set d_level = 100

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 130
set pitch_rc_rate = 130
set yaw_rc_rate = 130
set tpa_rate = 0

# rateprofile
rateprofile 1

set roll_rc_rate = 155
set pitch_rc_rate = 155
set yaw_rc_rate = 155
set roll_srate = 66
set pitch_srate = 66
set yaw_srate = 66
set tpa_rate = 0

# rateprofile
rateprofile 2

set roll_rc_rate = 190
set pitch_rc_rate = 190
set yaw_rc_rate = 190
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 55
set tpa_rate = 0

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# `

mikeller, 2018-04-03:
`aux 3 35 2 1300 2100 0`

`src/main/interface/msp_box.c:    { BOXFLIPOVERAFTERCRASH, "FLIP OVER AFTER CRASH", 35 },`

Could it be that what you are seeing is an artefact in the log that is produced by the `SPIN_DIRECTION_NORMAL` / `SPIN_DIRECTION_REVERSED` command being sent for crash flip?

ctzsnooze, 2018-04-03:
I wasn't flicking the switch for turtle mode, and I wasn't in turtle mode.  Those were just normal throttle up blips with the quad sitting on the desk.  It should be easy to replicate if it's real.

ctzsnooze, 2018-04-03:
Ah OK and YES!  You're right!

I've tested again on a FF_PIKOF4 with R-XSR.  This one gets the glitch only when smoothing on throttle is enabled, not otherwise.

But if I set runaway_takeoff_prevention = OFF, the glitch (and the little blue LED flash on first throttle up) both disappear.

So this is a minor artefact related to runaway takeoff prevention that occurs only when RC smoothing is on.

There is a related small issue in these logs also.  There are small spikes going through in to the motors, occurring at the 9ms step point, only when throttle RC smoothing is on.  I've seen this on a couple of Furious (FF_xxx) boards now.  Changing smoothing to auto, or 7ms, or 10ms makes no difference.

[image]
no smoothing on throttle, runaway protection on

[image]
smoothing on, runaway protection on

[image]
smoothing on, runaway protection off

ctzsnooze, 2018-04-03:
Also, runaway takeoff check should that happen every throttle up?  Or only for throttle ups within X seconds of boot up?  With smoothing on, the LED flashes every single throttle up.

closed 2018-05-05 by ctzsnooze