betaflight/betaflight #5601: Arming Motors only after 10 - 15 secs

Yawness, 2018-04-01:
Hey Guys,

so i flashed my SPRacingF3 to Betaflight 3.3.0 and got my copter fly. Everything is perfect, instead of the arming. So my biggest issue is, that after i connect the copter and my remote, i have to wait up to 10 - 15 secs. till the beeper tells me, i can arm now. I already searched and found CLI-Commands like the moron threshold, arming angle, the time value after connecting to ar. everything is set to high or lowest, but the time will not be any shorter. CLI-Status says, its the calibration. Which value tells me something about the calib? Does anybody of you have had the same problem?
It would be so fckn awesome if i could get rid of this problem. Its driving me crazy!

My CLI-Diff:

# diff
# Betaflight / SPRACINGF3 (SRF3) 3.3.0 Mar  2 2018 / 04:02:50 (177472b4f) MSP API: 1.37
feature -RX_PPM
feature -TELEMETRY
feature -TRANSPONDER
feature RX_SERIAL
feature OSD
serial 2 64 115200 57600 0 115200
aux 0 0 0 1700 2100 0
aux 1 1 1 1700 2100 0
aux 2 2 1 1300 1700 0
set gyro_lpf = 188HZ
set gyro_sync_denom = 1
set moron_threshold = 150
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1100
set max_check = 2000
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set small_angle = 90
set pid_process_denom = 1
set osd_warnings = 63
set osd_vbat_pos = 2552
set osd_rssi_pos = 2107
set osd_mah_drawn_pos = 2528
set osd_warnings_pos = 2048
set osd_battery_usage_pos = 2065
set pwr_on_arm_grace = 0
profile 0
rateprofile 0
# 

My CLI-Dump:

# dump
# Betaflight / SPRACINGF3 (SRF3) 3.3.0 Mar  2 2018 / 04:02:50 (177472b4f) MSP API: 1.37
name -
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
[... 548 lines cut ...]
set i_yaw = 45
set d_yaw = 20
set p_level = 50
set i_level = 50
set d_level = 75
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650
#

Arakon, 2018-04-01:
You're running 8k on gyro on a flight controller with i2c gyro.. that's massively overloading the gyro. Also, for some reason you have a hardware filter active (gyro_lpf). Disable that and set it to 4k and it should work properly.. it's surprising that it even flies decently.

Yawness, 2018-04-01:
Thank you man for giving me the right advise!
Just did one flight with that filter active and i was not feeling a difference. Everything was fine. I had 1000 Loop Cycles with 5-10% CPU load. No Erros or something else. How can i get such information to not fck up anything in my future builds?

I followed youre adivse, but there is no difference. The Gyro still takes up to 15 secs. till im ready to arm. I even went back to 2.6
I compared the dumps from CF and BF, but i cannot find a difference at gyro values.

Here is the CF Dump:

set debug_mode = NONE
set emf_avoidance = OFF
set i2c_highspeed = ON
set mid_rc = 1500
set min_check = 1120
set max_check = 1880
set rssi_channel = 0
set rssi_scale = 30
set rssi_ppm_invert = OFF
set rc_smoothing = OFF
set rx_min_usec = 885
set rx_max_usec = 2115
set serialrx_provider = SBUS
set sbus_inversion = ON
set spektrum_sat_bind = 0
set input_filtering_mode = OFF
set min_throttle = 1050
set max_throttle = 2000
set min_command = 1000
set motor_pwm_rate = 400
set servo_center_pulse = 1500
set servo_pwm_rate = 50
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set retarded_arm = OFF
set disarm_kill_switch = ON
set auto_disarm_delay = 15
set max_arm_angle = 25
set fixedwing_althold_dir = 1
set reboot_character = 82
set gps_provider = NMEA
set gps_sbas_mode = AUTO
set gps_auto_config = ON
set gps_auto_baud = OFF
set telemetry_switch = OFF
set telemetry_inversion = ON
set telemetry_send_cells = ON
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set hott_alarm_sound_interval = 5
set ibus_report_cell_voltage = OFF
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 36
set vbat_warning_cell_voltage = 36
set amperage_meter_scale = 0
set amperage_meter_offset = 0
set multiwii_amperage_meter_output = OFF
set amperage_meter_source = VIRTUAL
set vbat_hysteresis = 2
set align_gyro = DEFAULT
set align_acc = DEFAULT
set align_mag = DEFAULT
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = -180
set small_angle = 25
set max_angle_inclination = 380
set pid_process_denom = 1
set gyro_sync = 1
set gyro_sample_hz = 2000
set gyro_lpf = OFF
set gyro_lowpass_level = NORMAL
set gyro_lowpass_hz = 90
set gyro_notch_hz = 130
set gyro_notch_cutoff_hz = 130
set moron_threshold = 32
set imu_dcm_kp = 2500
set imu_dcm_ki = 0
set pid_at_min_throttle = ON
set yaw_motor_direction = 1
set yaw_jump_prevention_limit = 200
set tri_unarmed_servo = ON
set servo_lowpass_freq =  400.000
set servo_lowpass_enable = OFF
set failsafe_delay = 10
set failsafe_off_delay = 200
set failsafe_throttle = 1000
set failsafe_kill_switch = OFF
set failsafe_throttle_low_delay = 100
set failsafe_procedure = 1
set acc_hardware = 0
set baro_hardware = 0
set mag_hardware = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = SPIFLASH
set magzero_x = 0
set magzero_y = 0
set magzero_z = 0

set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83
set gps_wp_radius = 200
set nav_controls_heading = ON
set nav_speed_min = 100
set nav_speed_max = 300
set nav_slew_rate = 30
set alt_hold_deadband = 40
set alt_hold_fast_change = ON
set deadband = 0
set yaw_deadband = 10
set yaw_control_direction = 1
set 3d_deadband_throttle = 50
set throttle_correction_value = 0
set throttle_correction_angle = 800
set default_rate_profile = 0
set gimbal_mode = NORMAL
set acc_cut_hz = 15
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff =  5.000
set acc_unarmedcal = ON
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf =  0.600
set baro_cf_vel =  0.985
set baro_cf_alt =  0.965
set mag_declination = 0
set pid_controller = LUX
set p_pitch = 35
set i_pitch = 20
set d_pitch = 25
set p_roll = 35
set i_roll = 20
set d_roll = 25
set p_yaw = 85
set i_yaw = 45
set d_yaw = 0
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 45
set i_level = 10
set d_level = 75
set p_vel = 120
set i_vel = 45
set d_vel = 1
set pid_delta_method = MEASUREMENT
set yaw_p_limit = 500
set yaw_lpf_hz = 0
set dterm_lowpass_level = HIGH
set dterm_lowpass_hz = 100
set horizon_tilt_effect = 75
set horizon_tilt_mode = SAFE
set gtune_loP_rll = 10
set gtune_loP_ptch = 10
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set rc_yaw_expo = 0
set thr_mid = 0
set thr_expo = 25
set roll_rate = 85
set pitch_rate = 85
set yaw_rate = 85
set tpa_rate = 50
set tpa_breakpoint = 1500