betaflight/betaflight #5605: Arm Switch Not working, but Stick arming

VAL72, 2018-04-01:
I have a Strange behavior of arming feature in BF3.3.0/omnibusF4sd fs-x6b/ibus tricopter+gps+mag with arming Switch. IT won't Spin the motors, but warnings LED flashing Yellow/blue. Servo is normal working. Erasing the arming Switch Feature and simply arming with Stick command...all okay. No Errors, No warnings, nice flight.
Reconfiguring aux1 to arm Switch causes Same Errors as before. UAV Not Spinning motors and warning LED flashing yellow/blue. Beeper 3 Times short beep Error Message. Seems to make No Sense.
By the way, i checked Up all Standard issues with arming Problems, but all seems fine. Respecting IT Arms with standard Stick command, there i can Not imagine any config failure.
What is the big difference between arming Switch and Stick command?

mikeller, 2018-04-01:
@VAL72: Can you post a `diff` of your configuration please, like you were asked to when submitting this issue:
> 5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/);

VAL72, 2018-04-02:
so, i actually updated the first issue message. hope, now it is a better understanding what happend in my controller.

mikeller, 2018-04-02:
Thanks. Your min_check is quite low. Could it be that it's only reached when you pull down the throttle stick, but not when the throttle is low but you don't touch it? Try increasing it and try again.

VAL72, 2018-04-02:
Hi, i know what you mean, but  using ibus it quiet stable. The ranges are exactly 1000 to 2000 all channels.
For your reserch i did change the min_check to 1100. But no goal. When arming with the switch. The Arm LED registers arming. servo ok, but no motors. warning LED yello flasching one time every...second?

etracer65, 2018-04-02:
@VAL72 I'm not really sure what you're going for here.  You have `feature MOTOR_STOP` enabled and airmode disabled so it's normal that the motors don't spin immediately after arming.  You have to raise the throttle above the `min_check` value to make the motors spin.

To test if there are other conditions preventing you from arming, with props removed go into the configurator and the CLI tab.  Type `status` and look for the arming disabled flags.  "CLI" and "MSP" are normal but others would indicate a reason arming would be prevented.

VAL72, 2018-04-02:
@etracer65 i hope, you are able to help me to solve this issue "not arming /not spinning Motors with "arming Switch". The essential request i did was: "what is the big difference between stick arming and switch arming". These two questions should help to find my solution.
Ok. i know this motor_stop. I told you, the uav is running well with stick arming, but not with switch arming feature! of cause i did the throttle above min_check, because i was running the uav with stick arming( not with switch arming feature).
Thanks, know the status command.
See this:
'''# status
System Uptime: 32 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=168MHz, Vref=3.29V, Core temp=36degC, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=QMC5883
SD card: None inserted
Stack size: 2048, Stack address: 0x1000fff0
I2C Errors: 0, config size: 2283, max available config: 16384
CPU:9%, cycle time: 125, GYRO rate: 8000, RX rate: 130, System rate: 9
Arming disable flags: CLI MSP

#'''
So, following your instructions, there should be no problem, but definitely there is no motor running with switch arming feature! And of cause i testet with more throttle than min_check.
As i told first, the uav is well running with stick arming. only configuring switch arming is setting the uav from good to bad guy :-(. So what is your suggestion in that case?

jflyper, 2018-04-02:
I have loaded the second diff (the one with ARM mode switch) onto my keychain Omnibus F4 V3 (OMNIBUSF4SD target) flashed with 3.3.0.

I have changed receiver configuration to MSP RX and enabled motor test.

I can arm without a problem using AUX1.
Bare board, nothing other than USB connected nor wired.

Can you try using the MSP RX and see how it works?

VAL72, 2018-04-03:
@jflyper thanks for your try. I have omnibus F4 V2 very similar. i am not comfortable with RX_MSP. But i tryed it now first time. Effect was, i can switch the AUX1 with RX_MSP, see it changing in receivers tab and also in the modes tab, but not changing color in the Modes TAB. The stick control (AETR) not following the RX_MSP when checking in the receivers tab! In CLI "Status" is reportet RXLOSS ! I don't understand that, because it is RX_MSP on USB. How can i lose it?
With original X6B Receiver using ibus on uart1 all 10 used channels are following in the receivers tab and no problem so far. So, maybe RX_MSP is not my favourite testing feature at this time.
In the meanwhile i am thinking about the fact, the yaw servo is working after aux1 "arming" in normal setup. the arming status led is getting the right color now (now i set it to black, if armed for better visibility with warning overlay). Warning overlay is flashing yellow color approx. one time per second.
Is it possible, the uav is "partial" arming and blocking the motor start or simply blocking or freezing the dshot600 protocoll?
But why does it only happen with configured AUX arming switch? Without configured arming switch with simple stick arming is all fine working. (but for sure it is more safe, to use an arming switch, to immediately freeze the uav in case of dangerous situations)

jflyper, 2018-04-03:
@VAL72 I'm afraid that your configuration is somehow mangled badly.

Would you please test with the RX_MSP again, with the following procedure?

(1) Flash the FC with v3.3.0.

(2) Goto CLI and issue "defaults".

(3) Goto CLI and paste your diff for AUX switch above. Issue `save` to save and reboot.

(4) Goto Configuration tab and change your receiver to RX_MSP.

(5) Do Save & Reboot.

(6) Goto Motor tab, turn on "I understand the risks" switch.

(7) Goto Receiver tab, hit "Control sticks" and "Enable controls".

(8) Goto Mode tab, manipulate your AUX1 slider.

jflyper, 2018-04-04:
@VAL72 
I think you are not following my instructions. We were trying to establish a common configuration that works.

Since we can not establish the common starting ground for trouble shooting, and the issue is not a software problem, I will close this now. Please use other channels of support.

EDIT
I even loaded my TRI with your configuration, and it arms with AUX switch without a problem.

closed 2018-04-04 by jflyper

jflyper, 2018-04-04:
EDIT2
Btw, you seem to be oriented toward navigation functions, which BF is not good at all. I would suggest you take the iNav route.

VAL72, 2018-04-04:
@jflyper thanks at all. Maybe netter i Order a new Controller. Maybe IT IS a Hardware Problem at all