betaflight/betaflight #5610: Feature Request: Option to erase internal flash every startup

flint723, 2018-04-02:
If it is not possible to auto erase (loop record) when the flash is full (#5404), what about an option to auto erase the flash every time the quad is powered on? For some users this might be nice as there would always be a log of the last flight, in the case of something going wrong.  

Only bad thing is the 20 second delay on every start up, but probably no way around that.  

Another scenario where it might not be good, is if the fc actually rebooted in flight.  Then you might lose the log from before the reboot.

I know you can do it in OSD already, but this would be for the person who just flies, doesn't download logs, but wants blackbox working every flight.

Just curious if anyone's interested in a feature like this, or if it's even possible.

mikeller, 2018-04-02:
This sounds like a pretty useless feature: If the flash was erased every time on bootup, then how would you download the log of the last flight, especially if power was lost during a crash? The erase would start as soon as you plugged in USB, and the data would be gone by the time you got ready to download.

But if you really want this feature, enable the BLACKBOX_ERASE mode for the full range of an AUX channel, and your flight controller will erase the log on every startup.

flint723, 2018-04-02:
True.... did not think that the USB power would cause an erase.

I don't need the feature myself but have seen plenty of cases where people wanted a blackbox from a crash but their flash was full of previous flights.

Thanks for the reply.

mikeller, 2018-04-02:
Yeah, I guess we could use the USB detection (which is really a 'no battery connected' detection), and only trigger the erase if no battery was connected. It's still somewhat dodgy though, and I don't know if people would want to risk losing valuable logs due to battery detection not working properly.

flint723, 2018-04-02:
You're right that's too dodgy.  On some of my boards with ESC telemetry as sensor, they read 22.2v in configurator with no battery plugged in.

mikeller, 2018-04-02:
It's one of the shortcomings of current board designs that they aren't properly using an USB detect pin, but rely on battery voltage to detect that they are plugged into USB.

flint723, 2018-04-03:
Seems like this is not really an option here.

Other logging systems I've seen in RC have the ability to sense when they are connected by usb, but as you say they designed their own hardware.

Pretty amazing how good and reliable BF logging already is vs some I have used in other areas of RC.

closed 2018-04-03 by flint723