betaflight/betaflight #5611: Documentation for INFLIGHT_ACC_CAL

TwoToneEddy, 2018-04-02:
Hi,
This is more of a documentation issue i'd like to raise (and potentially rectify). I can't seem to find any documentation for this feature. Does any documentation exist? I've had a look in the code, specifically sensors/acceleration.c at the method performInflightAccelerationCalibration() and i'm not really sure what's going on. Can someone either point me in the direction of the documentation or give me abit of info so I can document it?
Many thanks
Lee

mikeller, 2018-04-02:
@TwoToneEddy: Looking at the code, it looks like the `INFLIGHT_ACC_CAL` feature makes accelerometer calibration available on a stick command: `THR_LO + YAW_LO + PIT_HI + ROL_HI`. This is different from the 'normal' acc calibration when unarmed (`THR_HI + YAW_LO + PIT_LO + ROL_CE`), as in it's also supposed to work when armed.
Not sure if / how this is supposed to work, since entering the stick command will force the craft into an edge flip, which is probably not what you want when calibrating the accelerometer.
Also, note that this functionality is very old, and should probably be removed.

McGiverGim, 2018-04-03:
I think this is to trim the copter in the air, maybe: http://ncopters.blogspot.com.es/2012/04/howto-inflight-acc-calibration-in.html

mikeller, 2018-04-03:
Good find, @McGiverGim! But still not sure what the value of this is (since you can simply arm on the ground), and if it still works with the current code.

McGiverGim, 2018-04-03:
A more detailed info is here: http://www.multiwii.com/forum/viewtopic.php?f=7&t=893

All of this comes because a perfect flat in the earth, can drift when hovering in the air. This makes a perfect hover in the air and applies it as zero position to the accelerometer.

I don't know if this is being used or not now, but I'm sure that if we remove it some users will appear ;)

mikeller, 2018-04-03:
@McGiverGim: My comment about removing was for the case that it isn't working any more. If it is working, fine, but if it isn't, then it's just using up space for no benefit at all and it should be fixed (if there still is demand for it), or removed. The last changes to it that I could find (quickly) were in 2012...

McGiverGim, 2018-04-03:
A analytics system (or other similar system) in the configurator will be useful for this, to know what features are being used and what not.

mikeller, 2018-04-03:
@McGiverGim: Totally agreed. Data was being collected and sent off to Google Analytics way back when we forked from the cleanflight configurator. We didn't want to collect user's 'private' data and removed the instrumentation. In hindsight this wasn't a good move.

mikeller, 2018-04-12:
@menachg: See https://discourse-cdn-sjc2.com/standard16/uploads/micro_motor_warehouse/original/2X/a/a63ab1e85c1b9358c301a8b32b893dafb764aa0c.png

menachg, 2018-04-13:
yea i did exactly that.
but it never goes into trim mode in the first place when throttle all the way up(and i checked my endpoints it goes to upper limit)
it doesnt start beeping or anything

mikeller, 2018-04-13:
What is your `max_check` set to?

menachg, 2018-04-13:
2000 and throttle goes above thag and more than tx deadband
So it should register its in the top

mikeller, 2018-04-13:
Try setting that to 1950 and see if that fixes it.

menachg, 2018-04-13:
Ok so apparently it does see im in full throtle but what is suppose to change acc_trim
I now realize changes rates profile:
Throtle up+
Ele up=rateprofile 1
Ele down=rateprofile 2
Ail left=rateprofile 4
Ail right=rateprofile 3
What is the new command for acc trim?

mikeller, 2018-04-15:
@menachg: The commands for ACC trim are still the same:
- if you are in ACRO mode (which does not use ACC anyway), the commands are used to change rate profiles;
- if you are in ANGLE or HORIZON mode, the commands are used for ACC trim.

I know this is quite unintuitive, but there are only so many stick combinations that can be used. I think using any combination that is throttle down / yaw center to change flight related settings is dangerous - this is the default position for the throttle / yaw stick, so any accidental movement of the roll / pitch stick will cause an unintended change.

menachg, 2018-04-15:
yew.. i just realized that
it took me a look in src code to realize this
well the way i programmed my tx when disarmed it defaults to rate mode
i changed it and it works like a charm now
thanks for the patience anyways :)

closed 2018-05-22 by stale[bot]