betaflight/betaflight #5614: BF 3.3.0 (3.4.0) SP Racing F3 Current sensor issue

severteka, 2018-04-03:
I use SPRF3 board. Before updating I used  BF 3.2.4. The current sensor worked correctly (since version 2.9). I use the current sensor Allegro ASC758-LCB-100U. After updating to version 3.3 BF, the value of current sensor is incorrect, about +17A. I tried BF 3.4, but the current sensor all the same works not correctly. 
Link to video:
https://drive.google.com/file/d/1SH6ByP00XSnzJ_LBaCpJHupbSKgGMWLT/view?usp=sharing

mikeller, 2018-04-03:
@severteka: The way of how the calibration is applied was changed from 3.2 to 3.3, you'll have to recalibrate the current sensor.

severteka, 2018-04-03:
I could not find anywhere information, that in BF 3.3 procedure of calibration of the current sensor differs. Give me, please, the link to new method of calibration.

I tried BF3.3.1, where is written:
-Increased the range of ibata_offset to accommodate certain Hall effect sensors (#5467);

only at ibata_offset =-16000 I could receive the acceptable value. But it is strange. Earlier ibata_offset value was in mV, and now in "parrots" :) .

Perhaps only I have such problem?

mikeller, 2018-04-04:
#4743 is the pull request changing the current sensor calibration.

@Dimand: Can I ask you to submit a pull request to update 
https://github.com/betaflight/betaflight/blob/master/docs/Battery.md
with the new process to calibrate the current sensor?

Dimand, 2018-04-04:
@mikeller: no worries, I will do a little write up tonight after work.