betaflight/betaflight #5617: VTX "Pit mode" on a switch is broken

AndersHoglund, 2018-04-03:
PR #4627 introduced VTX Pitmode as an RC mode on a Tx switch. Unfortunately there are a number of problems with this. A user on RCG notified me on the problems. 

1. When testing with a Tramp VTX, I can only get the VTX to go into but not out of Pitmode. It is stuck once set. This can be fixed by a small patch:

```
$ git diff
diff --git a/src/main/io/vtx.c b/src/main/io/vtx.c
index 465cc9c..d1b5962 100644
--- a/src/main/io/vtx.c
+++ b/src/main/io/vtx.c
@@ -176,11 +176,11 @@ static bool vtxProcessPitMode(vtxDevice_t *vtxDevice)
                     vtxCommonSetPitMode(vtxDevice, true);
                     return true;
                 }
-            } else {
-                if (pitOnOff) {
-                    vtxCommonSetPitMode(vtxDevice, false);
-                    return true;
-                }
+            }
+        } else {
+            if (pitOnOff) {
+                vtxCommonSetPitMode(vtxDevice, false);
+                return true;
             }
         }
     }
```
 
2. As there now are at least three (3) different ways to control Pitmode, via CMS, via Spektrum Tx VTX Setup menu and this RC Mode switch, there are also a competition and conflict between those three. With the patch above the RC mode switch will take precedence and override any other menu selection, even if the RC Pitmode switch is not set up at all. So the patch above are nor real solution.  What we are missing is a function similar to `IS_RC_MODE_ACTIVE(BOXVTXPITMODE)`, but only returning if the RC mode has been setup or not, not that it is activated. We would need distinguish three states, first of all if the RC mode is setup at all, then if the mode is active or not. Seems the original code was done with the assumption this was the case. Maybe there are such functions? I have not found any though. This would not fully solve the conflicts, you can not use all three ways to control Pitmode anyhow, but at least you could avoid activating the dominating RC Mode switch. Not the best solution, best would be a totally new system, allowing multiple input. Not sure what that would look like though....

vkolotov, 2018-08-24:
I've got Akk ultimate VTX that I can switch to PIT mode via OSD, however PIT mode does not work on mode switch (though a radio switch).

xamboozi, 2018-09-19:
I have the AKK X2 ultimate connected to a holybro F4 V2 and cannot enable or disable pit mode via a radio switch or with a LUA script. The only way I've been able to disable pit mode is by changing the output power.

pierreh, 2019-08-22:
I've got a AKK FX2-Dominator and I experience the same thing. I believe though it behaves the same as the TBS Unify. See https://www.youtube.com/watch?v=dCWngIk8MW4 and https://github.com/betaflight/betaflight/wiki/Unify-Smartaudio#switching-between-freestyle-and-race

haslinghuis, 2022-05-08:
Closing as inactive

closed 2022-05-08 by haslinghuis