betaflight/betaflight #5622: Matek F722 3.4 BARO, GPS, MAG problem

fpoullet, 2018-04-04:
Hello !

i have Matek 722 std with GPS M8N with MAG.
In defaults mode I2c error 6 (no GPS and MAG actived)
if i active GPS + MAG i2c error not stop and CPU 100% after 30s.

This problem is with always release !

jflyper, 2018-04-04:
Most I2C compass issues are caused by one or more of the followings, and not software problems.

- Incorrect wiring, including SDA/SCL mismatch. (This is most likely your problem.)
- Inappropriate I2C conditioning including proper pull-up that matches with the speed and length of the bus.
- Incorrect configuration, including settings of configuration variable `mag_bustype` and `mag_i2c_device`.
- Unsupported compass device including random variants collectively called QMC5883.

Please use other channels of support such as Betaflight thread in RCG, starting with providing CLI `diff` output.

closed 2018-04-04 by jflyper

jflyper, 2018-04-04:
[image]

jflyper, 2018-04-04:
```
# diff
# Betaflight / MATEKF722 (MKF7) 3.3.1 Apr  3 2018 / 09:50:01 (4b7eaf79d) MSP API: 1.37
feature GPS
serial 3 2 115200 115200 0 115200
set gps_provider = UBLOX
```