betaflight/betaflight #5624: Reverse Prop Orientation Gyro Crash Issue

SpoonFPV, 2018-04-04:
I don't fly reverse prop orientation, I'm only reporting what I observed on two separate individuals's quads at a recent race.  When using reverse prop orientation there seemed to be an issue when something was hit at high speed where the quad would spin wildly, climb and never recover.  The pilots reported using a later release of betaflight 3.2

I know this used to be an issue with all quads, that was solved, but seems to be something that happens to users running reversed props.  I believe it's caused when the gyro maxes out.  Unfortunately neither of them were my quads, so I don't know the exact configuration, or have blackbox data to show the fault.  I can only report that it's extremely dangerous and probably should be looked into.

claytonbakerjr, 2018-04-04:
Interesting that you posted this. I too experienced this on several occasions after 3.2 I also run reverse props. The only option to get it to stop was to kill the motors and re-arm. I had it happen mostly on clipping gates at higher speeds. It would spin and climb and throttle had zero effect in all cases. I was going to get blackbox logs but weather has been too ugly to fly the last few weeks.

SpoonFPV, 2018-04-04:
Yeah, back in the day...(like two years ago) it used to happen every time you clipped a gate, and it was just accepted, but betaflight did some magic to fix it that seems to be broken on reverse props

etracer65, 2018-04-04:
Try to capture a blackbox log of the event otherwise there's not much anybody can do to research or remedy.

Also make sure you have `set gyro_overflow_detect = ALL` in the CLI.  ICM-206xx gyros are vulnerable to overflow/inversion events at high rates of spin and the `gyro_overflow_detect` will prevent a flyaway.  There's no relationship to prop rotation other than the prop direction on impact may contribute to addition spin making overflow more likely.

richard-scott, 2018-04-05:
Does it do the same in 3.3?

varkokonyi, 2018-04-05:
No, because that feature prevents it

richard-scott, 2018-04-05:
@varkokonyi - yes, that was my point. The users need to test the same setup on 3.3 and see if it still is an issue. As you say, we expect that issue to not be there anymore.

claytonbakerjr, 2018-04-05:
@richard-scott 
@varkokonyi 
In my case this also happens on 3.3, I originally noticed it in 3.2 then shortly afterwards
I updated to 3.3 on all quads. I was out flying the 3.3 builds about 8 days ago and this
happened twice during that day.
As @etracer65 has commented I too will provide BB logs once the
weather in my area allows for more flying. I've been down with a bug and draggin my butt
out in the cold is the last thing I want to do :)

closed 2018-05-12 by stale[bot]