betaflight/betaflight #5625: KakuteF4 V2 blackbox logs not opening

trendab, 2018-04-04:
I can't open the blackbox logs from my KakuteF4 V2 board, they freeze the blackbox explorer.

Logging at 1khz on 8/8. Tried Betaflight and Cleanflight BB explorer 3.0.0 RC1 as well as the butterflight explorer built into the configurator - all freeze when I try opening BB logs from this FC. They work fine with log files from other FCs.
I posted this as an issue on the BB explorer page - mikeller said it may be a firmware issue and I should post here. 
Diff and log file below.
Thanks
[BTFL_BLACKBOX_LOG_Dragon_#2_20180404_191311.zip](https://github.com/betaflight/betaflight/files/1877327/BTFL_BLACKBOX_LOG_Dragon_.2_20180404_191311.zip)
```# version
# Betaflight / KAKUTEF4V2 (KTV2) 3.3.0 Mar  2 2018 / 03:48:43 (177472b4f) MSP API: 1.37

# name
name Dragon #2

# resources
resource MOTOR 1 B01
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 A03

# mixer

# servo

# servo mix

# feature
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map

# serial

# led

# color

# mode_color

# aux
aux 0 0 0 1850 2100 0
aux 1 13 3 1850 2100 0
aux 2 26 2 1850 2100 0
aux 3 35 2 1300 1700 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h

# master
set gyro_sync_denom = 4
set gyro_lowpass_hz = 100
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set gyro_stage2_lowpass_hz = 250
set gyro_use_32khz = ON
set min_check = 1010
set max_check = 1990
set rssi_channel = 6
set blackbox_device = SPIFLASH
set min_throttle = 1010
set dshot_idle_value = 350
set use_unsynced_pwm = ON
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 32000
set failsafe_delay = 5
set bat_capacity = 1100
set vbat_max_cell_voltage = 45
set current_meter = ADC
set ibata_scale = 333
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set pid_process_denom = 1
set osd_rssi_alarm = 42
set osd_cap_alarm = 1000
set osd_tim2 = 1025
set osd_vbat_pos = 2400
set osd_rssi_pos = 2106
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2432
set osd_flymode_pos = 333
set osd_throttle_pos = 2424
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2454
set osd_mah_drawn_pos = 2369
set osd_craft_name_pos = 2090
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 2135
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2281
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 2056
set osd_disarmed_pos = 2250
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 377
set osd_esc_rpm_pos = 407
set osd_core_temp_pos = 2081
set osd_stat_max_spd = OFF
set osd_stat_rtc_date_time = ON
set debug_mode = NOTCH
set displayport_max7456_wht = 3

# profile
profile 0

set dterm_lowpass = 90
set vbat_pid_gain = ON
set anti_gravity_gain = 3000
set dterm_setpoint_weight = 100

# rateprofile
rateprofile 0

set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 78```

flint723, 2018-04-05:
Does logging at 2khz make any difference?  I can't open this log either.

trendab, 2018-04-05:
Did not try 2khz, but I tried 500mhz, that did not work, neither.

McGiverGim, 2018-04-06:
If it helps, is related with the `P interval`:
```
H P interval:0/1
```
I think the zero is not correct, at least the explorer does not use it correctly. I changed this 0/1 by 1/2 in the two occurrences of this element, and the log is opened without problem.
I don't understand what represents this value, so I don't know what will be the correct value.
Can someone explain what represents this?

McGiverGim, 2018-04-06:
It has something to do with the blackbox rate:

https://github.com/betaflight/betaflight/blob/c8a646ebc3d35b9a6ef1de36190808869fefafe1/src/main/blackbox/blackbox.c#L1199

flint723, 2018-04-06:
In the past I have seen 1.5khz logs fail to open in this manner (not sure if it was 3.3 or 3.4) and recently another user had a log at 1.5khz that wouldn't open.  I have this log, will attach it.

[LOG00008 (2).zip](https://github.com/betaflight/betaflight/files/1884736/LOG00008.2.zip)

McGiverGim, 2018-04-06:
Same problem: `H P interval:0/1`

I've found the description of this: https://github.com/betaflight/betaflight/issues/2954 but I'm not too sure if I can understand enough to solve the problem.

flint723, 2018-04-06:
It seems editing the BB file is kind of volatile (microsoft notepad breaks it, sublime text sort of breaks it but not totally)

McGiverGim, 2018-04-06:
I've taking a look at the code, and I think the problem has something to do with rounding some values.

With a looptime of 125, the I interval is 256, and the P interval denom is 1, not zero. But with a looptime of 126, that is what your log shows, the I interval is 253 and results in a denom of 0, creating this problem. What I don't know if why your looptime is 126 and not 125.

McGiverGim, 2018-04-06:
I need to go out, but if someone is interested, the problem is here:

https://github.com/betaflight/betaflight/blob/c8a646ebc3d35b9a6ef1de36190808869fefafe1/src/main/blackbox/blackbox.c#L1698-L1706

Depending of the values in the blackbox of `P denom` and `I interval` the first function gives a value < 1, being and integer is rounded to zero, producing all this problem. Maybe we must multiply numerator and denominator by `blackboxIInterval` to be sure they remain as >= 1.

McGiverGim, 2018-04-07:
If I'm not wrong the `P interval` is the number of gyro loop cycles that must pass before we capture a blackbox frame (I or P). 
If that is true, @trendab in your Blackbox file you must replace the two occurrences of `P interval` by `1/7`. I've do it and all seems ok. @flint723 in your Blackbox file you must use `1/5`.
You can change this values with Notepad++, for example.
I will try to make a "first version" PR today to talk about this to see if my thoughts are right.

McGiverGim, 2018-04-07:
@trendab @flint723 I've built firmware versions with a patch for the Blackbox. Could you test it?

https://drive.google.com/open?id=1O8ZnNd2fA60nedPm9UcUuo0juSEUAmDM

trendab, 2018-04-07:
@McGiverGim  Thank you, Sir, it works for me! Not entirely sure what you did, but it did the trick.
Thanks, again!

McGiverGim, 2018-04-07:
Happy to hear it :)
I don't know if the solution will work for everybody, I've pushed a PR and I will discuss the details with the others developers to get to a final conclusion.

flint723, 2018-04-07:
The log I posted wasn't mine, I've told him about this firmware but not heard back if he tested it.

McGiverGim, 2018-04-07:
Sorry, I didn't notice that. I'm out, I can try to build it for you in some hours.

McGiverGim, 2018-04-08:
@flint723 I've added a firmware for you https://drive.google.com/open?id=1O8ZnNd2fA60nedPm9UcUuo0juSEUAmDM

flint723, 2018-04-08:
Thanks, I've tested it just now, definitely seems to have worked.
[1.5khz logs before and after firmware fix.zip](https://github.com/betaflight/betaflight/files/1886861/1.5khz.logs.before.and.after.firmware.fix.zip)

closed 2018-04-09 by mikeller

McGiverGim, 2018-04-09:
PR merged, it will be available in 3.3.2 or current master ;)

It will need the a new release of the Blackbox Explorer because the official change is slightly different than what I build for yours.