betaflight/betaflight #5632: Blackbox not logged, matek f405-osd bf 3.31

chriswils, 2018-04-05:
Upgrading Matek f405-OSD from 3.2.2 to 3.3.1, none of the four boards I have now do any blackbox logging. I have blackbox set in mode on AUX2, confirmed working with 3.2.2. The post-flight screen says `blackbox: failed`. Tried formatting the card, use another card (which I have already used for logging previously), still not working. Not even the LOGS directory is created.

# Betaflight / MATEKF405 (MKF4) 3.3.1 Apr  3 2018 / 09:48:52 (4b7eaf79d) MSP API: 1.37

# name
name WLGRD

# resources

# mixer

# servo

# servo mix

# feature
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB

# map
map TAER1234

# serial
serial 3 32 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100 0
aux 1 13 2 1700 2100 0
aux 2 26 1 1300 1700 0
aux 3 28 0 1800 2100 0
aux 4 35 1 1775 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1020
set rssi_channel = 16
set blackbox_p_ratio = 64
set motor_pwm_protocol = DSHOT600
set vbat_scale = 111
set beeper_dshot_beacon_tone = 5
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 2
set osd_warnings = 63
set osd_cap_alarm = 1000
set osd_tim1 = 769
set osd_vbat_pos = 2519
set osd_rssi_pos = 2106
set osd_tim_1_pos = 54
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 2496
set osd_craft_name_pos = 2507
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set vtx_band = 5
set vtx_channel = 5
set vtx_freq = 5806

chriswils, 2018-04-07:
Previous versions are working with my older sdcards, but not this one. Also tried `sdcard_dma = on`, which made status go from `failing` to `initializing fs`, but not all the way. Replaced with newer, larger SD and that works.

closed 2018-04-07 by chriswils

mikeller, 2018-04-07:
SD card support on current flight controllers is somewhat brittle, partly due to SD cards seemingly being picky about what speeds they work at when used in SPI mode.
Betaflight now has support for SDIO attached SD cards, which should work a bit better. But this will only be available for new targets.