betaflight/betaflight #5636: RunCam Split v1 with January 2018 firmware can't be controlled by Betaflight

andrejpodzimek, 2018-04-05:
This is a follow-up on #4840. As described there, RunCam stopped working with Betaflight for a few weeks/months, likely because Betaflight updates had been ahead of the anticipated RunCam firmware update. Then the issue was partially fixed for a short while; RunCam [could be activated](https://github.com/betaflight/betaflight/issues/4840#issuecomment-355828587), although it took multiple channel changes (usually 2-3) to start recording. The latest camera firmware was supposed to fix that, IIUC.

I've installed the new January 2018 firmware into my RunCam Split v1. The camera makes videos and connects to the mobile app, but there's no way to control it from Betaflight (tried 3.3.0, 3.3.1). The camera doesn't respond to the "camera power button" channel _at_ _all_. No beeps, no recording.

The problem has been around in one form or other for more than 6 months now. :disappointed: I wasn't expecting the support for a mid-2017 camera to be ditched that quickly. The camera controls worked back in August / September 2017. That said, would it please be possible to bring back that functionality? I'm more than happy to build custom Betaflight on my own, as long as my v1 cameras work again.

etracer65, 2018-04-06:
@andrejpodzimek It might help if you post a `diff` of your settings.  Enclose it in three back ticks (``\`) to make a code block.

@azolyoung Any thoughts?

azolyoung, 2018-04-07:
@andrejpodzimek

Hi, I'v flashed bf3.3.1 to my matek f405, worked as before..
Could you try again with following steps, and make sure your split has connected some spare uart to your FC, but  the split is power up with other power source.
1. Power up your Split
2. Wait 2 seconds,
3. Power up your FC,
4. Pull you stick that assoicated with camera button to trigger action of camera

If this steps is work for you, I think I know the reason about the issue

ghostface, 2018-04-07:
Same issue with betaflight 3.3.1 and runcam split mini

McGiverGim, 2018-04-08:
@azolyoung one question about the control of the Split. It can be added as modes to the modes tab, but, can it be controlled with the "FPV Camera Control" ( https://github.com/betaflight/betaflight/wiki/FPV-Camera-Control-(Joystick-Emulation))  like your new runcam control adapter (https://shop.runcam.com/runcam-control-adapter/) does?

Sometimes we need only to enter the OSD, and don't want (or don't have) free channels to assign in the modes tab.

Regards!

azolyoung, 2018-05-14:
Hi @McGiverGim , sorry for the late reply, I missed this mail in past days until today got your another notify:(
Yes, we got the responses from our customers, and we can reproduce problem in our quads, It's on the way will be resolved

McGiverGim, 2018-05-14:
Thanks for your support! 😊

lian, 2018-05-16:
same issue here (jan 2018 firmware, split mini). looking forward for a fix since its an awesome feature to have!

rav-rav, 2018-05-28:
I got a split now, too and ran into this issue.
Soft restart of the FC seems to make it work.

It does work when connected via USB and using CLI by just typing "save". And it also works when rebooting through the OSD menu.

open OSD -> save, reboot -> commands work

No idea where to look into the code, though.

kozlyuk, 2018-06-17:
I have the same problem with both RC Split 2 and V2.0.0 firmware.
Anybody know decision?

azolyoung, 2018-06-23:
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_OMNIBUSF4SD.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_OMNIBUSF4.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_MATEKF405.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_CLRACINGF4.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_BETAFLIGHTF4.hex

Hi, I've submit a PR(#6196) to BF to solve this problem, I was test the PR in my quad that also has this problem, and it work fine now. There are some firmwares that solve this problem. If you have time, please help test it. Thank you!

explorethetruth, 2018-06-23:
@azolyoung if you compile a build for matek722 target I will also certainly test it.

azolyoung, 2018-06-25:
@explorethetruth , oh, yes, matek722 has built

https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_MATEKF722.hex

explorethetruth, 2018-06-25:
@azolyoung I tested it right now. The problem is still present. The camera only responds after the flight controller is rebooted without turning off the RunCam power.
However, on 3.3.3 after activating any button binded to the camera while drone is disarmed, the flight controller rebooted itself, and everything starts working correctly. But if you take off without these actions, you will not be able to control the camera during the flight.

andrejpodzimek, 2018-06-26:
I also have these FC reboots after the first change in the camera power channel. Subsequent channel flips seem to work. Testes with 3 different F7s.

azolyoung, 2018-06-26:
Hi @andrejpodzimek  @explorethetruth , 

I'm sorry, in previous version still has bug in the initializing:(, and the problem is same with you met, need reboot FC after rcsplit initialized
the following are the new firmwares that build from #6213, it fixed the problem that need reboot FC
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_OMNIBUSF4SD_PR6213.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_OMNIBUSF4_PR6213.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_MATEKF722_PR6213.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_MATEKF405_PR6213.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_CLRACINGF4_PR6213.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_BETAFLIGHTF4_PR6213.hex

explorethetruth, 2018-06-26:
@azolyoung nothing changed in #6213 

> The camera only responds after the flight controller is rebooted without turning off the RunCam power.

I'm not an advanced programmer (and did not particularly study aspects of the RunCam Device Protocol). But maybe it makes sense to just add a small delay before initialization, because in my opinion the problem lies in the fact that the flight controller starts up faster than the camera.

azolyoung, 2018-06-27:
Hi, @explorethetruth, you are right, what I'm doing is something like you said, in the #6213, It's try to initialize rcsplit every 1000ms, and it will try 4 times, If it fails all four times, it is necessary to manually restart the flight controller so that the rcsplit serial communication can work normally. Because the different hardware conditions are different, my local test: The total number of attempts is 4 times and each interval of 1000ms can work well. So now the problem may be to find the most reliable number of attempts and intervals.

If you still need to restart the flight controller manually, I have built a firmware that tries 5 times every 2000ms. I would like to ask you to help test it again.
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_MATEKF722_retry_5times_interval2000ms.hex

explorethetruth, 2018-06-27:
@azolyoung ok, that's what I found out. Something in my 'diff all' dump prevents the flight controller from initializing the camera. I reconfigured my quad without dumps and the camera started working correctly.
I also rechecked it on the firmware without an increased number of initialization attempts (the one you applied earlier) - everything works correctly too. So you do not need to increase it. The problem was clearly somewhere in my 'diff all' dump. Here is it [Google Drive](https://drive.google.com/file/d/1slueLYURp_e56aGOLEuyShdP5ZpYFO8N/view?usp=sharing) (if you are interested).

ghostface, 2018-06-27:
@explorethetruth maybe post a diff all from after the manual reconfiguring so we can compare it to the one where it doesn't work?

explorethetruth, 2018-06-27:
@ghostface here is a new config, with it the camera is working fine [Google Drive](https://drive.google.com/file/d/1DiDllVpkCMl5sHBWBKL6uTpNQQ1FW-aS/view?usp=sharing)

azolyoung, 2018-06-28:
Hi @explorethetruth, Thanks! I add two cli commands to the #6213, so If there still some guys met you case, they can change these settings to make rcsplit sereial work, no need new firmware release.
```
rcdevice_init_dev_attempts = 2
Allowed range: 0 - 10
Default value: 4

rcdevice_init_dev_attempt_interval = 1000
Allowed range: 500 - 5000
```

explorethetruth, 2018-06-28:
@azolyoung I think this is a great option. I will definitely try different values. Thank you!

azolyoung, 2018-07-03:
@andrejpodzimek, Hi, could you try again with 3.4RC5 to confirm is this issue has fixed in your quad? Thanks!

closed 2018-08-09 by stale[bot]